openwrt/openwrt #15237: WS-AP3825i cannot boot Openwrt snapshot images in the master branch.

vochong, 2024-04-22:
### Describe the bug

Since the last few months, none of the OpenWrt master snapshot images can boot in WS-AP3825i. The images are downloaded from OpenWrt firmware selector download site. The console boot log is shown below.

There is no problem with OpenWrt 23.05.x images.

```
root@OpenWrt:/tmp# sysupgrade -n openwrt-master-mpc85xx-p1020-extreme-networks_w
s-ap3825i-squashfs-sysupgrade.bin 
Wed Mar 20 16:37:10 UTC 2024 upgrade: Commencing upgrade. Closing all shell sessions.
Watchdog handover: fd=3
- watchdog -
Watchdog does not have CARDRESET support
ipsec_starter[5817]: charon stopped after 400 ms
ipsec_starter[5817]: ipsec starter stopped
Wed Mar 20 16:37:13 UTC 2024 upgrade: Sending TERM to remaining processes ...
Wed Mar 20 16:37:14 UTC 2024 upgrade: Sending signal TERM to vsftpd (2682)
Wed Mar 20 16:37:14 UTC 2024 upgrade: Sending signal TERM to S95done (4812)
Wed Mar 20 16:37:14 UTC 2024 upgrade: Sending signal TERM to mount_root (4829)
Wed Mar 20 16:37:18 UTC 2024 upgrade: Sending KILL to remaining processes ...
Wed Mar 20 16:37:18 UTC 2024 upgrade: Sending signal KILL to mount_root (4829)
[   52.862281] jffs2_build_filesystem(): erasing all blocks after the end marker... 
[  301.195266] done.
[  301.204701] jffs2: notice: (4829) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[  301.220237] jffs2: fork failed for JFFS2 garbage collect thread: 4
[  308.671881] stage2 (6143): drop_caches: 3
Wed Mar 20 16:37:26 UTC 2024 upgrade: Switching to ramdisk...
Wed Mar 20 16:37:28 UTC 2024 upgrade: Performing system upgrade...
[  311.381053] do_stage2 (6143): drop_caches: 3
Unlocking firmware ...

Writing from <stdin> to firmware ...     
Wed Mar 20 16:38:25 UTC 2024 upgrade: Upgrade completed
Wed Mar 20 16:38:26 UTC 2024 upgrade: Rebooting system...
umount: can't unmount /dev: Resource busy
[  368.617516] reboot: Restarting system

U-Boot 2010.12.6 (Feb 26 2014 - 12:51:09) (primary)

Power LED ON
CPU0:  P1020E, Version: 1.1, (0x80ec0011)
Core:  E500, Version: 5.1, (0x80212051)
Clock Configuration:
       CPU0:800  MHz, CPU1:800  MHz, 
       CCB:400  MHz,
       DDR:333.333 MHz (666.667 MT/s data rate) (Asynchronous), LBC:25   MHz
L1:    D-cache 32 kB enabled
       I-cache 32 kB enabled
Board: P1020RDB CPLD: V15.15 PCBA: V15.0
rom_loc: nor upper bank
SD/MMC : 4-bit Mode
eSPI : Enabled
I2C:   ready
SPI:   ready
DRAM:  Configuring DDR for 666.667 MT/s data rate
DDR: 256 MiB (DDR3, 32-bit, CL=6, ECC off)
Testing DRAM from 0x00010000 to 0x000fffff
DRAM test phase 1:
DRAM test phase 2:
DRAM test passed.
FLASH: 64 MiB
L2:    256 KB enabled
MMC:  FSL_ESDHC: 0
PCIe1: Root Complex of mini PCIe SLOT, x1, regs @ 0xffe0a000
  01:00.0     - 168c:003c - Network controller
PCIe1: Bus 00 - 01
PCIe2: Root Complex of PCIe SLOT, x1, regs @ 0xffe09000
  03:00.0     - 168c:0033 - Network controller
PCIe2: Bus 02 - 03
In:    serial
Out:   serial
Err:   serial
Net:   eTSEC1, eTSEC3
### JFFS2 loading '/home/config/shadow'
rescan: First time in use
Scanning JFFS2 FS:  done.
find_inode failed for name=home
load: Failed to find inode
### JFFS2 LOAD ERROR<0> for /home/config/shadow!
Hit 'd' for diagnostics, or any other key to stop autoboot:  0 
## Booting kernel from FIT Image at ec000000 ...
   Using 'config-1' configuration
   Trying 'kernel-1' kernel subimage
     Description:  POWERPC OpenWrt Linux-6.1.86
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0xec0000ec
     Data Size:    3584452 Bytes = 3.4 MiB
     Architecture: PowerPC
     OS:           Linux
     Load Address: 0x01000000
     Entry Point:  0x01000000
     Hash algo:    crc32
     Hash value:   a28ffeac
     Hash algo:    sha1
     Hash value:   fcfae71d80f3655e37dac5d9d95e97afab9f5425
   Verifying Hash Integrity ... crc32+ sha1+ OK
## Flattened Device Tree from FIT Image at ec000000
   Using 'config-1' configuration
   Trying 'fdt-1' FDT blob subimage
     Description:  POWERPC OpenWrt extreme-networks_ws-ap3825i device tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0xec36b400
     Data Size:    12574 Bytes = 12.3 KiB
     Architecture: PowerPC
     Hash algo:    crc32
     Hash value:   95b2a45c
     Hash algo:    sha1
     Hash value:   0fc127433f710e00617e9ce2ba65c36fd7ded240
   Verifying Hash Integrity ... crc32+ sha1+ OK
   Booting using the fdt blob at 0xec36b400
   Loading Kernel Image ... OK
OK
   Loading Device Tree to 09ff9000, end 09fff11d ... OK
ft_fixup_l2cache: FDT_ERR_NOTFOUND                              <=== ALWAYS GOT STUCK HERE

```

### OpenWrt version

r25996-e0363233c9

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mpc85xx/p1020

### Device

Extreme Networks WS-AP3825i

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-04-23:
Cause this looks suspicously like:
https://github.com/openwrt/openwrt/issues/9779

robimarko, 2024-04-23:
@blocktrron @Hurricos

Hurricos, 2024-04-23:
I'll take a look at this. #9779 should no longer be relevant as we use simpleboot (thanks to @blocktrron) which hands the dtb directly through. 

I don't know if I recall ever booting 6.1.6 -- I think the mpc85xx kernel bump might be the issue.

Hurricos, 2024-04-23:
@vochong says

> Since the last few months

The 6.1 bump was just a month ago; do you know more exactly when this started? A bisect will also catch it.

Hurricos, 2024-04-23:
Diff between snapshot and 23.05.3 boot:

```diff
--- /tmp/a	2024-04-23 17:45:34.390964962 -0400
+++ /tmp/b	2024-04-23 17:51:03.320550029 -0400
@@ -1,19 +1,19 @@
 ## Booting kernel from FIT Image at 02000000 ...
    Using 'config-1' configuration
    Trying 'kernel-1' kernel subimage
-     Description:  POWERPC OpenWrt Linux-6.1.86
+     Description:  POWERPC OpenWrt Linux-5.15.150
      Type:         Kernel Image
      Compression:  uncompressed
      Data Start:   0x020000ec
-     Data Size:    7140936 Bytes = 6.8 MiB
+     Data Size:    6986816 Bytes = 6.7 MiB
      Architecture: PowerPC
      OS:           Linux
      Load Address: 0x01000000
      Entry Point:  0x01000000
      Hash algo:    crc32
-     Hash value:   20cf95fb
+     Hash value:   bfa4e683
      Hash algo:    sha1
-     Hash value:   3df8f1bc19daaf3b3f38992f716353f2d96ac541
+     Hash value:   4a555b135c14b2a2c4d35553cca72f22c86fd47a
    Verifying Hash Integrity ... crc32+ sha1+ OK
 ## Flattened Device Tree from FIT Image at 02000000
    Using 'config-1' configuration
@@ -21,16 +21,18 @@
      Description:  POWERPC OpenWrt extreme-networks_ws-ap3825i device tree blob
      Type:         Flat Device Tree
      Compression:  uncompressed
-     Data Start:   0x026cf884
-     Data Size:    12574 Bytes = 12.3 KiB
+     Data Start:   0x026a9e7c
+     Data Size:    12541 Bytes = 12.2 KiB
      Architecture: PowerPC
      Hash algo:    crc32
-     Hash value:   95b2a45c
+     Hash value:   ecfc0d58
      Hash algo:    sha1
-     Hash value:   0fc127433f710e00617e9ce2ba65c36fd7ded240
+     Hash value:   6d81d68bcfbf0113b69bb11ac38d7761ddf8959f
    Verifying Hash Integrity ... crc32+ sha1+ OK
-   Booting using the fdt blob at 0x26cf884
+   Booting using the fdt blob at 0x26a9e7c
    Loading Kernel Image ... OK
 OK
-   Loading Device Tree to 09ff9000, end 09fff11d ... OK
+   Loading Device Tree to 09ff9000, end 09fff0fc ... OK
 ft_fixup_l2cache: FDT_ERR_NOTFOUND
+
+zImage starting: loaded at 0x01000000 (sp: 0x016aafa0)
```

Hurricos, 2024-04-24:
Nice! So, you're saying you can reproduce a boot failure on commit 7dbe27e3 using kernel 5.15? Can you show the output of the boot when you built 7dbe27e3?

I'm upgrading my build machine at the moment, but I will start my git bisect bad at commit 7dbe27e3.

vochong, 2024-04-24:
@Hurricos 
Here is the console log when booting the image without the commit 2733831e1cdf0e37e2ba353b2dcd8f4c014c2831 and all later commits.

```
U-Boot 2010.12.6 (Feb 26 2014 - 12:51:09) (primary)

Power LED ON
CPU0:  P1020E, Version: 1.1, (0x80ec0011)
Core:  E500, Version: 5.1, (0x80212051)
Clock Configuration:
       CPU0:800  MHz, CPU1:800  MHz, 
       CCB:400  MHz,
       DDR:333.333 MHz (666.667 MT/s data rate) (Asynchronous), LBC:25   MHz
L1:    D-cache 32 kB enabled
       I-cache 32 kB enabled
Board: P1020RDB CPLD: V15.15 PCBA: V15.0
rom_loc: nor upper bank
SD/MMC : 4-bit Mode
eSPI : Enabled
I2C:   ready
SPI:   ready
DRAM:  Configuring DDR for 666.667 MT/s data rate
DDR: 256 MiB (DDR3, 32-bit, CL=6, ECC off)
Testing DRAM from 0x00010000 to 0x000fffff
DRAM test phase 1:
DRAM test phase 2:
DRAM test passed.
FLASH: 64 MiB
L2:    256 KB enabled
MMC:  FSL_ESDHC: 0
PCIe1: Root Complex of mini PCIe SLOT, x1, regs @ 0xffe0a000
  01:00.0     - 168c:003c - Network controller
PCIe1: Bus 00 - 01
PCIe2: Root Complex of PCIe SLOT, x1, regs @ 0xffe09000
  03:00.0     - 168c:0033 - Network controller
PCIe2: Bus 02 - 03
In:    serial
Out:   serial
Err:   serial
Net:   eTSEC1, eTSEC3
### JFFS2 loading '/home/config/shadow'
rescan: First time in use
Scanning JFFS2 FS: .  Unknown node type: e008 len 48 offset 0x3d24f18
Unknown node type: e009 len 28 offset 0x3d24f48
Unknown node type: e009 len 28 offset 0x3d24f64
Unknown node type: e008 len 76 offset 0x3d24f80
Unknown node type: e009 len 28 offset 0x3d24fcc
Unknown node type: e008 len 48 offset 0x3d24fe8
Unknown node type: e009 len 28 offset 0x3d25018
Unknown node type: e008 len 76 offset 0x3d25584
Unknown node type: e009 len 28 offset 0x3d255d0
Unknown node type: e009 len 28 offset 0x3d255ec
Unknown node type: e008 len 76 offset 0x3d261c0
Unknown node type: e009 len 28 offset 0x3d2620c
Unknown node type: e008 len 76 offset 0x3d26580
Unknown node type: e009 len 28 offset 0x3d265cc
. done.
find_inode failed for name=home
load: Failed to find inode
### JFFS2 LOAD ERROR<0> for /home/config/shadow!
Hit 'd' for diagnostics, or any other key to stop autoboot:  0 
## Booting kernel from FIT Image at ec000000 ...
   Using 'config-1' configuration
   Trying 'kernel-1' kernel subimage
     Description:  POWERPC OpenWrt Linux-5.15.150
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0xec0000ec
     Data Size:    3260624 Bytes = 3.1 MiB
     Architecture: PowerPC
     OS:           Linux
     Load Address: 0x01000000
     Entry Point:  0x01000000
     Hash algo:    crc32
     Hash value:   1ce99a7f
     Hash algo:    sha1
     Hash value:   d2b903cd2ee6c0a6d7e827d569d5cc841fefffbc
   Verifying Hash Integrity ... crc32+ sha1+ OK
## Flattened Device Tree from FIT Image at ec000000
   Using 'config-1' configuration
   Trying 'fdt-1' FDT blob subimage
     Description:  POWERPC OpenWrt extreme-networks_ws-ap3825i device tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0xec31c30c
     Data Size:    12574 Bytes = 12.3 KiB
     Architecture: PowerPC
     Hash algo:    crc32
     Hash value:   95b2a45c
     Hash algo:    sha1
     Hash value:   0fc127433f710e00617e9ce2ba65c36fd7ded240
   Verifying Hash Integrity ... crc32+ sha1+ OK
   Booting using the fdt blob at 0xec31c30c
   Loading Kernel Image ... OK
OK
   Loading Device Tree to 09ff9000, end 09fff11d ... OK
ft_fixup_l2cache: FDT_ERR_NOTFOUND
```

Hurricos, 2024-04-24:
Bisecting going well so far -- the problem commit is in this range:
```console
[labby@hashtag openwrt]$ git log --oneline 4443470a575b6fd775ff7c9d5eb40024db003213..97f542238a23a1c8461b40dfb26d9213826d8e78 | grep mpc85xx
102009f3ea mpc85xx: p1020: convert Aerohive AP330/AP350 to simpleImage
a99dfd195e mpc85xx: add 6.1 testing kernel
6a8b831593 mpc85xx: p1010: change wrapper address of simple image devices
894ff58067 mpc85xx: kernel: refresh 6.1 config
44ed2dc91c mpc85xx: refresh 6.1 patches
7b01179e62 mpc85xx: copy patches 5.15 to 6.1
123daef93c mpc85xx: copy config 5.15 to 6.1
3361ff9572 mpc85xx: Get rid of 'of_flat_dt_is_compatible'
```

Exciting! Looks like it's to do with one of the other SimpleImage touches.

Hurricos, 2024-04-24:
Lol. oops.

```diff
diff --git a/target/linux/mpc85xx/patches-5.15/107-powerpc-85xx-add-ws-ap3825i-support.patch b/target/linux/mpc85xx/patches-5.15/107-powerpc-85xx-add-ws-ap3825i-support.patch
index bd731a2344..51da0f4527 100644
--- a/target/linux/mpc85xx/patches-5.15/107-powerpc-85xx-add-ws-ap3825i-support.patch
+++ b/target/linux/mpc85xx/patches-5.15/107-powerpc-85xx-add-ws-ap3825i-support.patch
@@ -62,4 +62,4 @@ WS-AP3825i AP.
 +simpleboot-ws-ap3825i|\
  simpleboot-tl-wdr4900-v1)
      platformo="$object/fixed-head.o $object/simpleboot.o"
-     link_address='0x1000000'
+     link_address='0x1500000'
```

Looks like this diff was needed in commit 6a8b831593895ff9c8a687de7f7d0d24cdef8b3f (@CHKDSK88):

```diff
diff --git a/target/linux/mpc85xx/image/p1020.mk b/target/linux/mpc85xx/image/p1020.mk
index d3d7f93ddd..bbf674bf30 100644
--- a/target/linux/mpc85xx/image/p1020.mk
+++ b/target/linux/mpc85xx/image/p1020.mk
@@ -74,8 +74,8 @@ define Device/extreme-networks_ws-ap3825i
   DEVICE_PACKAGES := kmod-ath10k-ct ath10k-firmware-qca988x-ct
   BLOCKSIZE := 128k
   KERNEL_NAME := simpleImage.ws-ap3825i
-  KERNEL_ENTRY := 0x1000000
-  KERNEL_LOADADDR := 0x1000000
+  KERNEL_ENTRY := 0x1500000
+  KERNEL_LOADADDR := 0x1500000
   KERNEL = kernel-bin | fit none $(KDIR)/image-$$(DEVICE_DTS).dtb
   IMAGES := sysupgrade.bin
   IMAGE/sysupgrade.bin := append-kernel | append-rootfs | pad-rootfs | append-metadata
```

Hurricos, 2024-04-24:
`git bisect` confirms:

```console
# first bad commit: [6a8b831593895ff9c8a687de7f7d0d24cdef8b3f] mpc85xx: p1010: change wrapper address of simple image devices
```

I'm going to apply the patch I described above, then recompile and restest. If it works, I'll reapply it atop master and submit a PR.

Hurricos, 2024-04-24:
Aaand, there it is:

```console
## Booting kernel from FIT Image at 02000000 ...
   Using 'config-1' configuration
   Trying 'kernel-1' kernel subimage
     Description:  POWERPC OpenWrt Linux-5.15.150
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x020000ec
     Data Size:    7041640 Bytes = 6.7 MiB
     Architecture: PowerPC
     OS:           Linux
     Load Address: 0x01500000
     Entry Point:  0x01500000
     Hash algo:    crc32
     Hash value:   e52c726d
     Hash algo:    sha1
     Hash value:   6fa3408f0ba0e596b5be46831946b54c1d2f5c5a
   Verifying Hash Integrity ... crc32+ sha1+ OK
## Flattened Device Tree from FIT Image at 02000000
   Using 'config-1' configuration
   Trying 'fdt-1' FDT blob subimage
     Description:  POWERPC OpenWrt extreme-networks_ws-ap3825i device tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x026b74a4
     Data Size:    12574 Bytes = 12.3 KiB
     Architecture: PowerPC
     Hash algo:    crc32
     Hash value:   95b2a45c
     Hash algo:    sha1
     Hash value:   0fc127433f710e00617e9ce2ba65c36fd7ded240
   Verifying Hash Integrity ... crc32+ sha1+ OK
   Booting using the fdt blob at 0x26b74a4
   Loading Kernel Image ... OK
OK
   Loading Device Tree to 09ff9000, end 09fff11d ... OK
ft_fixup_l2cache: FDT_ERR_NOTFOUND

zImage starting: loaded at 0x01500000 (sp: 0x01bb8fa0)
Allocating 0xd05b40 bytes for kernel...
Decompressing (0x00000000 <- 0x01514000:0x01bb7268)...
Done! Decompressed 0xcccb80 bytes
```

Reapplying diff to master ...

Hurricos, 2024-04-24:
#15261 fixes this.

stintel, 2024-04-25:
84a48ce400b2c7b0779f51e83c68de5f8ec58ffd

closed 2024-04-25 by stintel

Hurricos, 2024-04-25:
Thank you! @vochong please come back in 24 to the next snapshot build to test.

vochong, 2024-04-25:
Thanks a lot @Hurricos.

I've built my custom image with the approved commit and it works great.