openwrt/openwrt #15238: 23.05.x MR33 and MR52 - Static IP config on Lan breaks networking

antonical, 2024-04-22:
### Describe the bug

See this article for details. https://forum.openwrt.org/t/cisco-meraki-mr52-no-dns-after-upgrade-to-23-05-3/194709/32 Essentially when configuring a static IP address on the Lan interface bridge results in broken networking. No DNS resolver respomds and software cannot be updated even though the network config is correct. Switching to DHCP for the Lan bridge restores networking and dns works agian and also software updating works again.

### OpenWrt version

23.05.x

### OpenWrt release

https://us06web.zoom.us/j/81417471905?pwd=xPkbnvMkAy4WXcvQKBRijPIpTE8iXO.1

### OpenWrt target/subtarget

ipq40xx/generic

### Device

MR33 and MR52

### Image kind

Official downloaded image

### Steps to reproduce

Flash MR33 or MR 52 to new firmware do not keep config. Configure Static IP on the Lan bridge. No dns resolver respomds and you cannot get to the internet. Switch to DHCP on teh Lan bridge and all networking is restored.

### Actual behaviour

Broken networking

### Expected behaviour

Normal networking shoudl be active.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

antonical, 2024-04-22:
Sorry I tried 23.05.2 to 23.05.5 from the Openwrt firmware download site for each device MR33 and MR52

The version running on the MR33 is OpenWrt 23.05.3 r23809-234f1a2efa / LuCI openwrt-23.05 branch git-24.086.45142-09d5a3

spamsool, 2024-04-23:
On my MR33 after upgrading from 21.02 to 23.05.x I get random MAC address on LAN and br-lan interface upon every boot on 23.05.x
I had to set MAC address to static, otherwise it kept getting different IP every time.

[1.562240] ipqess-edma c080000.ethernet: generated random MAC address 12:a2:ab:ca:33:02

antonical, 2024-04-23:
Yes I also get random MAC on every reboot.

antonical, 2024-04-25:
Please remove the invalid label on this bug report as the missing information has now been completed.

jow-, 2024-04-25:
Please provide the output of `uci show network` and `ifstatus lan` in the broken state.

antonical, 2024-04-25:
> Please provide the output of `uci show network` and `ifstatus lan` in the broken state.

I will have to pull it and reconfigure it as it is now in DHCP mode. We went through a lot of this initial diag on this thread 

https://forum.openwrt.org/t/cisco-meraki-mr52-no-dns-after-upgrade-to-23-05-3/194709/32

With many iterations and networking info provided. There is nothing untoward in any of the configs they all look correct. We have 12 of these devices configured with static IP's these two devices a MR52 and an MR33 were upgraded to 23.05.[0,1,2,3] and none of those builds would work with a static IP config on the lan bridge. On DHCP a random mac is provided every reboot and we have had 2 random lockups whilst testing requiring hard reboots to recover. 

Cheers

rkboni, 2024-07-23:
FWIW, I see the random-MAC-address-on-boot on a MR-52 install across both 23.05.3 and just-upgraded 23.05.4, but networking otherwise is just fine... I have a static IP on `br-lan` and everything works fine.

The only hickup I had was that the default route got unset at some point and I couldn't access the system via WireGuard (because the system is on the 192.168.1.0/24 net and the WG IPs are 10.10.0.0/24).  Manually adding that fixed it, and it's been fine through subsequent reboots & upgrades.

antonical, 2024-10-23:
I am just coming back to this now that 23.05.5 is released for the MR52. It has the same issues. 

OOTB with a static IP configured on the default LAN interface I cannot get to the network. I can connect into the MR52 but can't get out from it. This is the same on 23.05.1/2/3/4/5 and the latest snapshot build.  23.05.0 works as expected. 

Hope this helps someone else.

rkboni, 2024-10-24:
I suspect the following change needs to be back-ported: https://github.com/openwrt/openwrt/commit/d264d3a60ecc1 and the following change needs to be un-done: https://github.com/openwrt/openwrt/pull/15410 (possibly due to some Kernel updates switching the interfaces back so that `eth0` is the PoE port as it should be).

I'm trying to address the first, and if I get that sorted out (mainly, if I can figure out how to get the right GitHub actions configured and `.config` set up to auto-build via GHA) and tested will also look at the latter.

rkboni, 2024-10-24:
As for MR33, I suspect the twin change in https://github.com/openwrt/openwrt/pull/16623 should address that if it was also caused by the ole random MAC on reboot issue.

rkboni, 2024-10-24:
@antonical try the build in https://github.com/rkb-org/openwrt/releases/tag/meraki-c4a15f7, that should fix both the MAC addresses (I've tested this locally as well), and the LEDs being backwards (haven't had a chance to test this, but it's just a revert of the original LED swap).