openwrt/openwrt #15250: emmc board stuck on boot

Smartype, 2024-04-24:
### Describe the bug

mediatek-filogic-unielec_u7981-01-emmc, with 23.05 it stuck on boot. while main branch is working.

branch: openwrt-23.05
commit id: 675d9a2d4141a9251e6feb0dd948b7f6533553ae

### OpenWrt version

r23837-675d9a2d41

### OpenWrt release

23.05-SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

unielec_u7981-01-emmc

### Image kind

Self-built image

### Steps to reproduce

Choose target/device. build and then flash sys upgrade with uboot tftp.
Then it boots and stuck.

### Actual behaviour

Kernel stucks.

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.153 (eric@exands-R630) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23837-675d9a2d41) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Sun Apr 21 21:26:53 2024
[    0.000000] Machine model: Unielec U7981-01 (EMMC)
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000004fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
[    0.000000]   node   0: [mem 0x0000000043030000-0x0000000047c7ffff]
[    0.000000]   node   0: [mem 0x0000000047c80000-0x0000000047ffffff]
[    0.000000]   node   0: [mem 0x0000000048000000-0x000000004fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000004fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 17 pages/cpu s32664 r8192 d28776 u69632
[    0.000000] pcpu-alloc: s32664 r8192 d28776 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64512
[    0.000000] Kernel command line: root=/dev/mmcblk0p6 kernel=/dev/mmcblk0p5 console=ttyS0,115200 factory=/dev/mmcblk0p3
[    0.000000] Unknown kernel command line parameters "kernel=/dev/mmcblk0p5 factory=/dev/mmcblk0p3", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 240884K/262144K available (8128K kernel code, 890K rwdata, 2232K rodata, 448K init, 285K bss, 21260K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 640 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x000000000c080000
[    0.000000] arch_timer: cp15 timer(s) running at 13.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x2ff89eacb, max_idle_ns: 440795202429 ns
[    0.000000] sched_clock: 56 bits at 13MHz, resolution 76ns, wraps every 4398046511101ns
[    0.000129] Calibrating delay loop (skipped), value calculated using timer frequency.. 26.00 BogoMIPS (lpj=130000)
[    0.000139] pid_max: default: 32768 minimum: 301
[    0.000422] Mount-cache hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.000432] Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.001723] rcu: Hierarchical SRCU implementation.
[    0.002073] smp: Bringing up secondary CPUs ...
[    0.002397] Detected VIPT I-cache on CPU1
[    0.002422] GICv3: CPU1: found redistributor 1 region 0:0x000000000c0a0000
[    0.002451] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.002516] smp: Brought up 1 node, 2 CPUs
[    0.002530] SMP: Total of 2 processors activated.
[    0.002535] CPU features: detected: 32-bit EL0 Support
[    0.002539] CPU features: detected: CRC32 instructions
[    0.002565] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.002628] CPU: All CPU(s) started at EL2
[    0.002639] alternatives: patching kernel code
[    0.005951] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.005973] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.006115] pinctrl core: initialized pinctrl subsystem
[    0.006767] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.007110] DMA: preallocated 128 KiB GFP_KERNEL pool for atomic allocations
[    0.007138] DMA: preallocated 128 KiB GFP_KERNEL|GFP_DMA pool for atomic allocations
[    0.007168] DMA: preallocated 128 KiB GFP_KERNEL|GFP_DMA32 pool for atomic allocations
[    0.007508] thermal_sys: Registered thermal governor 'fair_share'
[    0.007512] thermal_sys: Registered thermal governor 'bang_bang'
[    0.007516] thermal_sys: Registered thermal governor 'step_wise'
[    0.007521] thermal_sys: Registered thermal governor 'user_space'
[    0.007721] ASID allocator initialised with 65536 entries
[    0.008192] pstore: Registered ramoops as persistent store backend
[    0.008198] ramoops: using 0x10000@0x42ff0000, ecc: 0
[    0.018680] cryptd: max_cpu_qlen set to 1000
[    0.020967] SCSI subsystem initialized
[    0.021111] libata version 3.00 loaded.
[    0.022267] clocksource: Switched to clocksource arch_sys_counter
[    0.022942] NET: Registered PF_INET protocol family
[    0.023039] IP idents hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.023412] tcp_listen_portaddr_hash hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.023432] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.023441] TCP established hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.023460] TCP bind hash table entries: 2048 (order: 3, 32768 bytes, linear)
[    0.023488] TCP: Hash tables configured (established 2048 bind 2048)
[    0.023557] UDP hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.023574] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.023682] NET: Registered PF_UNIX/PF_LOCAL protocol family
[    0.023712] PCI: CLS 0 bytes, default 64
[    0.026217] workingset: timestamp_bits=46 max_order=16 bucket_order=0
[    0.029815] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.029831] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.064508] pwm-mediatek 10048000.pwm: clock: top fail: -517
[    0.073677] Serial: 8250/16550 driver, 3 ports, IRQ sharing disabled
[    0.074570] printk: console [ttyS0] disabled
[    0.094742] 11002000.serial: ttyS0 at MMIO 0x11002000 (irq = 18, base_baud = 2500000) is a ST16650V2
[    0.727898] printk: console [ttyS0] enabled
[    0.732879] cacheinfo: Unable to detect cache hierarchy for CPU 0
[    0.741328] loop: module loaded
---- no more logs 

### Expected behaviour

It boots to system init.

### Additional info

This issue does not exist in main branch.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Smartype, 2024-04-24:
> Invalid Release reported. `23.05-SNAPSHOT`
> Is this from a clean repository?

As this board is not in v23.05.3 tag, so I use openwrt-23.05 branch instead.

mrkiko, 2024-04-24:
Maybe @dangowrt can help?
May commit e1831321de15ba4a8401cbf8c2c713939de968c1 help?

dangowrt, 2024-04-24:
Could be missing pinctrl is the cause, or a missing clock. I don't have that hardware, it was UniElec themselves which submitted the patches. Probably what ever is needed for the GL-iNet MT2500 or X3000 is needed for that board as well.

Smartype, 2024-04-24:
> Could be missing pinctrl is the cause, or a missing clock. I don't have that hardware, it was UniElec themselves which submitted the patches. Probably what ever is needed for the GL-iNet MT2500 or X3000 is needed for that board as well.

Thanks, but with main branch, this dts works without any issue.

Smartype, 2024-04-24:
> Maybe @dangowrt can help?
> May commit [e183132](https://github.com/openwrt/openwrt/commit/e1831321de15ba4a8401cbf8c2c713939de968c1) help?

Thanks, I will back port that commit to 23.05 and give it a try.

dangowrt, 2024-04-24:
On Wed, Apr 24, 2024 at 03:20:07AM -0700, Eric Day wrote:
> Thanks, but with main branch, this dts works without any issue.

Also for the openwrt-23.05 branch it was UniElec themselves requesting
the backport via a pull-request:
https://github.com/openwrt/openwrt/pull/14991

The PR was hence obviously untested.