openwrt/openwrt #15265: Raspberry Pi 5's openwrt peripherals and 2.5 G network card do not work.

ykla, 2024-04-25:
### Describe the bug

Raspberry Pi 5's openwrt peripherals and 2.5 G network card do not work.

The Raspberry Pi 5's fan doesn't work, as well as the rtc clock doesn't work, and the USB Gigabit NIC doesn't work. Only the HDMI, WIFI and Ethernet cards work.

```
root@OpenWrt:~# cat /boot/config.txt 
################################################################################
# Bootloader configuration - config.txt
################################################################################

################################################################################
# For overclocking and various other settings, see:
# https://www.raspberrypi.com/documentation/computers/config_txt.html
################################################################################

# OpenWrt config
include distroconfig.txt

[all]
# Place your custom settings here.
dtparam=i2c1=on
dtoverlay=i2c-rtc,ds1307
dtparam=spi=on
dtparam=i2s=on
```

```
root@OpenWrt:~# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
root@OpenWrt:~# 
```
```
root@OpenWrt:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux 6.1.86 xhci-hcd xHCI Host Controller
Bus 002 Device 001: ID 1d6b:0003 Linux 6.1.86 xhci-hcd xHCI Host Controller
Bus 003 Device 001: ID 1d6b:0002 Linux 6.1.86 xhci-hcd xHCI Host Controller
Bus 004 Device 001: ID 1d6b:0003 Linux 6.1.86 xhci-hcd xHCI Host Controller
Bus 004 Device 002: ID 0b95:1790 ASIX AX88179A
root@OpenWrt:~# dmesg | grep usb
[    0.677554] usbcore: registered new interface driver usbfs
[    0.683065] usbcore: registered new interface driver hub
[    0.688410] usbcore: registered new device driver usb
[    0.693513] usb_phy_generic phy: supply vcc not found, using dummy regulator
[    1.146158] usbcore: registered new interface driver uas
[    1.151506] usbcore: registered new interface driver usb-storage
[    1.799444] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
[    1.808098] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.815704] usb usb1: Product: xHCI Host Controller
[    1.820949] usb usb1: Manufacturer: Linux 6.1.86 xhci-hcd
[    1.826715] usb usb1: SerialNumber: xhci-hcd.0
[    1.840215] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.01
[    1.848877] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.856400] usb usb2: Product: xHCI Host Controller
[    1.861562] usb usb2: Manufacturer: Linux 6.1.86 xhci-hcd
[    1.867239] usb usb2: SerialNumber: xhci-hcd.0
[    1.928424] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
[    1.936986] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.944499] usb usb3: Product: xHCI Host Controller
[    1.949653] usb usb3: Manufacturer: Linux 6.1.86 xhci-hcd
[    1.955328] usb usb3: SerialNumber: xhci-hcd.1
[... 388 lines cut ...]
```

### OpenWrt version

r26009-6ca8305598

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2712

### Device

Raspberry Pi 5 Model B Rev 1.0

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Noltari, 2024-04-25:
@ykla ideally you should split this into different issues...

1. Regarding the RTC, did you install the I2C driver?
`opkg install kmod-i2c-bcm2835`
(Reboot after installing it)

2. I'm using the official RPi 5 fan and it works OOB, without any config needed on its specific new fan connector.
Which fan are you using?
Maybe it's a different one and it requires some configuration.

3. Which 2.5G ethernet card are you trying?
Is it connected by PCIe? If it does, then you need to enable pcie in the config:
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#enable-pcie
`dtparam=pciex1`

4. I'm using a TP-Link UE300 without issues, but I haven't tried any AX88179A-based card...

ykla, 2024-04-25:
> @ykla ideally you should split this into different issues...
>
> 1. Regarding the RTC, did you install the I2C driver?
> `opkg install kmod-i2c-bcm2835`
> (Reboot after installing it)
>
> 2. I'm using the [...]

Just in 

 <https://openwrt.org/toh/raspberry_pi_foundation/raspberry_pi>

Sorry ,not 2.5 NetCard, I am using 1G USB to Rj45 device AX88179A.

I just installed kmod-i2c-bcm2835 from <https://openwrt.org/toh/raspberry_pi_foundation/raspberry_pi>

Noltari, 2024-04-25:
> > @ykla ideally you should split this into different issues...
> > 
> > 1. Regarding the RTC, did you install the I2C driver?
> >    `opkg install kmod-i2c-bcm2835`
> >    (Reboot after installing it)
> > 2. I'm using the official RPi 5 fan and it works OOB, without any config needed on its specific new fan connector.
> >    Which fan are you using?
> >    Maybe it's a different one and it requires some configuration.
> > 3. Which 2.5G ethernet card are you trying?
> >    Is it connected by PCIe? If it does, then you need to enable pcie in the config:
> >    https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#enable-pcie
> >    `dtparam=pciex1`
> > 4. I'm using a TP-Link UE300 without issues, but I haven't tried any AX88179A-based card...
> 
> Just in
> 
> https://openwrt.org/toh/raspberry_pi_foundation/raspberry_pi
> 
> Sorry ,not 2.5 NetCard, I am using 1G USB to Rj45 device AX88179A.
> 
> I just installed kmod-i2c-bcm2835 from https://openwrt.org/toh/raspberry_pi_foundation/raspberry_pi before I reported issues.

Can you try adding `dtparam=i2c_arm=on` to your config?

ykla, 2024-04-25:
> > > @ykla ideally you should split this into different issues...
> > > 
> > > 1. Regarding the RTC, did you install the I2C driver?
> > >    `opkg install kmod-i2c-bcm2835`
> > >    (Reboot after installing it)
> > > 2. I'm using the official RPi 5 fan and it works OOB, without any config needed on its specific new fan connector.
> > >    Which fan are you using?
> > >    Maybe it's a different one and it requires some configuration.
> > > 3. Which 2.5G ethernet card are you trying?
> > >    Is it connected by PCIe? If it does, then you need to enable pcie in the config:
> > >    https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#enable-pcie
> > >    `dtparam=pciex1`
> > > 4. I'm using a TP-Link UE300 without issues, but I haven't tried any AX88179A-based card...
> > 
> > 
> > Just in
> > https://openwrt.org/toh/raspberry_pi_foundation/raspberry_pi
> > Sorry ,not 2.5 NetCard, I am using 1G USB to Rj45 device AX88179A.
> > I just installed kmod-i2c-bcm2835 from https://openwrt.org/toh/raspberry_pi_foundation/raspberry_pi before I reported issues.
> 
> Can you try adding `dtparam=i2c_arm=on` and removing `dtparam=i2c1=on` from your config?

Replacing removing dtparam=i2c1=on with dtparam=i2c_arm=on seems to have no effect as before.

Noltari, 2024-04-25:
Ok, I think I've some AX88179A adapter laying around, so I will test it when I have time.
Anyhow, this is not the best adaptar to use on the RPi with OpenWrt, it's usually better to use Realtek-based USB adapters.

Noltari, 2024-08-07:
Sorry for the delay but I've just tested an AmazonBasics AX88179A adapter on a RPi 4B and it's working correctly on latest main branch:
```
[   72.631827] usb 2-1: new SuperSpeed USB device number 2 using xhci_hcd
[   72.668858] usb 2-1: New USB device found, idVendor=0b95, idProduct=1790, bcdDevice= 1.00
[   72.677063] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   72.684216] usb 2-1: Product: AX88179
[   72.687878] usb 2-1: Manufacturer: ASIX Elec. Corp.
[   72.692767] usb 2-1: SerialNumber: 000050B61AF82E
[   74.210097] ax88179_178a 2-1:1.0 eth1: register 'ax88179_178a' at usb-0000:01:00.0-1, ASIX AX88179 USB 3.0 Gigabit Ethernet, 00:50:b6:1a:f8:2e
[  102.541645] br-lan: port 2(eth1) entered blocking state
[  102.546914] br-lan: port 2(eth1) entered disabled state
[  102.552195] ax88179_178a 2-1:1.0 eth1: entered allmulticast mode
[  102.558411] ax88179_178a 2-1:1.0 eth1: entered promiscuous mode
[  110.871460] bcmgenet fd580000.ethernet eth0: Link is Down
[  110.877163] br-lan: port 1(eth0) entered disabled state
[  110.883694] bcmgenet fd580000.ethernet eth0: Link is Down
[  110.889555] bcmgenet fd580000.ethernet eth0: Link is Down
[  114.324860] ax88179_178a 2-1:1.0 eth1: ax88179 - Link status is: 1
[  114.332638] br-lan: port 2(eth1) entered blocking state
[  114.337878] br-lan: port 2(eth1) entered forwarding state
```

```
[    9.241927] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    9.251268] bcmgenet fd580000.ethernet eth0: Link is Down
[    9.256737] br-lan: port 1(eth0) entered blocking state
[    9.262011] br-lan: port 1(eth0) entered disabled state
[    9.267257] bcmgenet fd580000.ethernet eth0: entered allmulticast mode
[    9.273974] bcmgenet fd580000.ethernet eth0: entered promiscuous mode
[    9.295888] br-lan: port 2(eth1) entered blocking state
[    9.301136] br-lan: port 2(eth1) entered disabled state
[    9.306427] ax88179_178a 2-1:1.0 eth1: entered allmulticast mode
[    9.312695] ax88179_178a 2-1:1.0 eth1: entered promiscuous mode
[    9.322040] bcmgenet fd580000.ethernet eth0: Link is Down
[    9.328365] bcmgenet fd580000.ethernet eth0: Link is Down
[    9.334205] bcmgenet fd580000.ethernet eth0: Link is Down
[    9.340240] bcmgenet fd580000.ethernet eth0: Link is Down
[   11.988405] ax88179_178a 2-1:1.0 eth1: ax88179 - Link status is: 1
[   11.996426] br-lan: port 2(eth1) entered blocking state
[   12.001685] br-lan: port 2(eth1) entered forwarding state
[   98.137618] ax88179_178a 2-1:1.0 eth1: ax88179 - Link status is: 0
[   98.144048] br-lan: port 2(eth1) entered disabled state
[  101.911493] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[  101.919611] br-lan: port 1(eth0) entered blocking state
[  101.924857] br-lan: port 1(eth0) entered forwarding state
```

Therefore I will close this since everything reported here is working.

closed 2024-08-07 by Noltari

alessandromrc, 2024-11-03:
> Sorry for the delay but I've just tested an AmazonBasics AX88179A adapter on a RPi 4B and it's working correctly on latest main branch:
>
> ```
> [   72.631827] usb 2-1: new SuperSpeed USB device number [...]

Hi, it would be interesting to know which kernel driver you are using to make the "AX88179A" adapter work... it seems to not be working on the latest snapshot for the pi5.

Thanks.