openwrt/openwrt #15270: [22.03] Host issue compiling sed on MacOS

BKPepe, 2024-04-25:
### Describe the bug

1. My MacOS was recently updated to 14.4.1
2. I wanted to start with the clean build_dir, but unfortunately, it was not great idea, though.
3. I can not compile sed as tool for MacOS

### OpenWrt version

22.03.06

### OpenWrt release

22.03.06

### OpenWrt target/subtarget

mvebu/cortex-a9

### Device

Turris Omnia

### Image kind

Official downloaded image

### Steps to reproduce

1. Have the latest MacOS version
2. Clone OpenWrt repository and checkout openwrt-22.03 branch
3. Compile tools

### Actual behaviour

```
mv -f $depbase.Tpo $depbase.Po
depbase=`echo lib/strerror.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
gcc -DHAVE_CONFIG_H -I.  -I. -I./lib -I./lib -I./sed -I/Volumes/OpenWrt/omnia-hbl/build/staging_dir/host/include   -O2 -I/Volumes/OpenWrt/omnia-hbl/build/staging_dir/host/include  -MT lib/strerror.o -MD -MP -MF $depbase.Tpo -c -o lib/strerror.o lib/strerror.c &&\
mv -f $depbase.Tpo $depbase.Po
lib/obstack.c:351:31: error: incompatible function pointer types initializing 'void (*)(void) __attribute__((noreturn))' with an expression of type 'void (void)' [-Wincompatible-function-pointer-types]
__attribute_noreturn__ void (*obstack_alloc_failed_handler) (void)
                              ^
1 error generated.
make[5]: *** [Makefile:2781: lib/obstack.o] Error 1
make[5]: *** Waiting for unfinished jobs....
lib/strerror.c:59:7: warning: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Wdeprecated-declarations]
      sprintf (buf, fmt, n);
      ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/stdio.h:180:1: note: 'sprintf' has been explicitly marked deprecated here
__deprecated_msg("This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.")
^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/sys/cdefs.h:218:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
1 warning generated.
make[5]: Leaving directory '/Volumes/OpenWrt/omnia-hbl/build/build_dir/host/sed-4.8'
make[4]: *** [Makefile:3105: all-recursive] Error 1
make[4]: Leaving directory '/Volumes/OpenWrt/omnia-hbl/build/build_dir/host/sed-4.8'
make[3]: *** [Makefile:2321: all] Error 2
make[3]: Leaving directory '/Volumes/OpenWrt/omnia-hbl/build/build_dir/host/sed-4.8'
make[2]: *** [Makefile:50: /Volumes/OpenWrt/omnia-hbl/build/build_dir/host/sed-4.8/.built] Error 2
make[2]: Leaving directory '/Volumes/OpenWrt/omnia-hbl/build/tools/sed'
time: tools/sed/compile#51.32#30.94#86.02
    ERROR: tools/sed failed to build.
make[1]: *** [tools/Makefile:166: tools/sed/compile] Error 1
make[1]: Leaving directory '/Volumes/OpenWrt/omnia-hbl/build'
```

### Expected behaviour

Sed works and can be compiled

### Additional info

I backported sed's update from master branch and it went alright.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-04-27:
Fixed via https://github.com/openwrt/openwrt/pull/15289

closed 2024-04-27 by robimarko

robimarko, 2024-04-27:
@BKPepe Does gdb fail now?

BKPepe, 2024-04-27:
I'll check it out later today.

BKPepe, 2024-04-28:
I can confirm your findings:
- toolchain/gdb now fails. 
Error log:
```
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 15] for this enumeration type [-Wenum-constexpr-conversion]
    integer_for_size<sizeof (T), static_cast<bool>(T (-1) < T (0))>::type
                                                   ^
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 1] for this enumeration type [-Wenum-constexpr-conversion]
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 3] for this enumeration type [-Wenum-constexpr-conversion]
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 3] for this enumeration type [-Wenum-constexpr-conversion]
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 3] for this enumeration type [-Wenum-constexpr-conversion]
5 errors generated.
make[5]: *** [Makefile:1641: ada-lang.o] Error 1
```

robimarko, 2024-04-28:
Yeah, same error here

BKPepe, 2024-04-28:
I found this:
https://sourceware.org/bugzilla/show_bug.cgi?id=30423 which leads to https://sourceware.org/git/?p=binutils-gdb.git;a=commit;h=ae61525fcf456ab395d55c45492a106d1275873a , let me try this one.

robimarko, 2024-04-28:
That fixes that error, but then you get:
```
make[5]: Entering directory '/Volumes/OpenWrt/openwrt/build_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/gdb-11.2/gdbsupport'
make  all-am
make[6]: Entering directory '/Volumes/OpenWrt/openwrt/build_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/gdb-11.2/gdbsupport'
g++ -std=gnu++11 -DHAVE_CONFIG_H -I.  -I./../include -I./../gdb -I../gnulib/import -I./../gnulib/import -I.. -I./.. -I./../intl -I../bfd -I./../bfd  -I/Volumes/OpenWrt/openwrt/staging_dir/host/include  -Wall -Wpointer-arith -Wno-unused -Wunused-value -Wunused-function -Wno-switch -Wno-char-subscripts -Wempty-body -Wunused-but-set-parameter -Wunused-but-set-variable -Wno-sign-compare -Wno-mismatched-tags -Wno-error=deprecated-register -Wsuggest-override -Wdeprecated-copy -Wdeprecated-copy-dtor -Wredundant-move -Wmissing-declarations -Wmissing-prototypes -Wformat -Wformat-nonliteral   -MT agent.o -MD -MP -MF .deps/agent.Tpo -c -o agent.o agent.cc
In file included from agent.cc:21:
In file included from ./../gdb/target/target.h:24:
In file included from ./../gdb/target/wait.h:23:
../gdbsupport/enum-flags.h:94:3: error: unknown type name 'DIAGNOSTIC_PUSH'
  DIAGNOSTIC_PUSH
  ^
../gdbsupport/enum-flags.h:95:46: error: expected ';' at end of declaration list
  DIAGNOSTIC_IGNORE_ENUM_CONSTEXPR_CONVERSION
                                             ^
                                             ;
../gdbsupport/enum-flags.h:99:3: error: unknown type name 'DIAGNOSTIC_POP'
  DIAGNOSTIC_POP
  ^
../gdbsupport/enum-flags.h:100:1: error: expected member name or ';' after declaration specifiers
};
^
4 errors generated.
make[6]: *** [Makefile:505: agent.o] Error 1
```

BKPepe, 2024-04-28:
You seems to be a lot faster than me. I just currently get into it compiling it with backported patch.

robimarko, 2024-04-28:
I have about an hour more on Scaleway until I can kill of the M2 machine to stop paying, so gotta be fast

BKPepe, 2024-04-28:
If not mistaken, we miss this commit: https://sourceware.org/git/?p=binutils-gdb.git;a=commit;f=include/diagnostics.h;h=f4906a9a7441ef9c2758513420568994a1e7fed3. More rather this file: [gdb/common/diagnostics.h](https://sourceware.org/git?p=binutils-gdb.git;a=blob;f=gdb/common/diagnostics.h;h=5a63bfd8d4ee949e98453a756d55b11d39e55ee0;hb=f4906a9a7441ef9c2758513420568994a1e7fed3)

Working on it.

robimarko, 2024-04-28:
I mean, that is from 2017, 22.03 is old but that no older than that

robimarko, 2024-04-28:
Hehe, the backport of:
https://sourceware.org/git/?p=binutils-gdb.git;a=blobdiff;f=gdbsupport/enum-flags.h;h=41ac7838f0609ad69f693bcf1614ed22f0ed3989;hp=700037f612603e7ca49d734aef5e3ccfeb196539;hb=ae61525fcf456ab395d55c45492a106d1275873a;hpb=f173b2fc9d03d9e56376ed9ad4c06e82949060dd

To fix the original error introduces this one

BKPepe, 2024-04-28:
> If not mistaken, we miss this commit: https://sourceware.org/git/?p=binutils-gdb.git;a=commit;f=include/diagnostics.h;h=f4906a9a7441ef9c2758513420568994a1e7fed3. More rather this file: [...]

My bad... that file is already included. It was just moved to different path. Nowadays it is in [include](https://sourceware.org/git?p=binutils-gdb.git;a=tree;f=include;hb=refs/heads/master) / [diagnostics.h](https://sourceware.org/git?p=binutils-gdb.git;a=blob_plain;f=include/diagnostics.h;hb=refs/heads/master)

BKPepe, 2024-04-28:
I tried to backport the gdb update (gdb 12.1) from OpenWrt 23.05, but it fails with the issue, which we are having, though about invalid range. I think, I am gonna to try to compile 23.05 from scratch to see if it is there or not.

BKPepe, 2024-04-28:
gdb toolchain fails as well on OpenWrt 23.05. Most likely, there should be different issue for that (like followup issue).

BKPepe, 2024-04-28:
> That fixes that error, but then you get:
>
> ```
> make[5]: Entering directory '/Volumes/OpenWrt/openwrt/build_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-11.2.0_musl_eabi/gdb-11.2/gdbsupport'
> make  all-am [...]

Interestingly, I backported the patch for OpenWrt 23.05 and it does not introduce any new errors neither this one. It was successfully compiled, though.

robimarko, 2024-04-29:
That means that a prerequisite is missing for the GDB in 22.03 as well

BKPepe, 2024-04-29:
Yeah, still can not figure out what is missing for OpenWrt 22.03, when the patch works for OpenWrt 23.05. 🤷 Need to investigate it more.

PR #15289 (2024-04-27): [22.03] tools: fix compilation on macOS 14
Current multiple tools in 22.03 will fail to compile when using macOS 14 with: 
```
depbase=`echo lib/obstack.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
gcc -DHAVE_CONFIG_H -I.  -I. -I./lib -I./lib -I./sed -I/Volumes/OpenWrt/openwrt/staging_dir/host/include   -O2 -I/Volumes/OpenWrt/openwrt/staging_dir/host/include  -MT lib/obstack.o -MD -MP -MF $depbase.Tpo -c -o lib/obstack.o lib/obstack.c &&\
mv -f $depbase.Tpo $depbase.Po
lib/obstack.c:351:31: error: incompatible function pointer types initializing 'void (*)(void) __attribute__((noreturn))' with an ex [... cut]
commit 4a6911fe (2024-04-27): tools/sed: fix compilation on macOS 14

Current sed 4.8 in 22.03 will fail to compile when using macOS 14 with:
depbase=`echo lib/obstack.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
gcc -DHAVE_CONFIG_H -I.  -I. -I./lib -I./lib -I./sed -I/Volumes/OpenWrt/openwrt/staging_dir/host/include   -O2 -I/Volumes/OpenWrt/openwrt/staging_dir/host/include  -MT lib/obstack.o -MD -MP -MF $depbase.Tpo -c -o lib/obstack.o lib/obstack.c &&\
mv -f $depbase.Tpo $depbase.Po
lib/obstack.c:351:31: error: incompatible function pointer types initializing 'void (*)(void) __attribute__((noreturn))' with an expression of typ [... cut]
commit 8d65f020 (2024-04-27): tools/cpio: fix compilation on macOS 14

Current cpio 2.13 in 22.03 will fail to compile when using macOS 14 with:
gcc -DHAVE_CONFIG_H -I. -I..   -I/Volumes/OpenWrt/openwrt/staging_dir/host/include   -O2 -I/Volumes/OpenWrt/openwrt/staging_dir/host/include  -MT obstack.o -MD -MP -MF .deps/obstack.Tpo -c -o obstack.o obstack.c
obstack.c:351:31: error: incompatible function pointer types initializing 'void (*)(void) __attribute__((noreturn))' with an expression of type 'void (void)' [-Wincompatible-function-pointer-types]
__attribute_noreturn__ void (*obstack_alloc_failed_handler) (void)
         [... cut]
commit eede9b1b (2024-04-27): tools/coreutils: fix compilation on macOS 14

Current coreutils 8.32 in 22.03 will fail to compile when using macOS 14 with:
depbase=`echo lib/obstack.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
gcc  -I. -I./lib  -Ilib -I./lib -Isrc -I./src -I/Volumes/OpenWrt/openwrt/staging_dir/host/include    -O2 -I/Volumes/OpenWrt/openwrt/staging_dir/host/include  -MT lib/obstack.o -MD -MP -MF $depbase.Tpo -c -o lib/obstack.o lib/obstack.c &&\
mv -f $depbase.Tpo $depbase.Po
lib/obstack.c:351:31: error: incompatible function pointer types initializing 'void (*)(void) __attribute__((noreturn))' with an expression [... cut]