openwrt/openwrt #15273: Wan is not working on latest Snapshot on Mi 4 A Gigabit

khanjui, 2024-04-26:
### Describe the bug

It connects but does not receives any RX. Possibly related to latest MT7530 switch commits. Snapshot from a day ago is working fine. @dangowrt 

### OpenWrt version

...

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 4A Gigabit Edition

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

9uyone, 2024-04-26:
The same on Redmi AC2100. Tried yesterday

liumingan, 2024-04-26:
The same on Phicomm K2P

ivanich, 2024-04-26:
Happened the same, I suggest to stop accept patches from Daniel Golle, it's been 5-th time I got my device bricked after hist commits.

namiltd, 2024-04-26:
Are you able to compile it yourself and check exactly after which compilation it stopped working?

DragonBluep, 2024-04-26:
> Happened the same, I suggest to stop accept patches from Daniel Golle, it's been 5-th time I got my device bricked after his commits.

I have to say, this is inevitable in the process of software development. Always new features, new bugs and then new fixes. In the past few years, Daniel has contributed a lot of outstanding code to this project.

namiltd, 2024-04-26:
Same problem: https://github.com/openwrt/openwrt/issues/15279

dangowrt, 2024-04-26:
Thank you for reporting this problem.

Probably the cause is one of the MT7530 related commits. While there are a few different version of the Mi 4 A (which are different in exactly the aspect of which port is used as WAN port and how), the Redmi AC2100 is a unique device and uses PHY-muxing of port 0 of the switch as WAN port via eth1. I'm trying to build for MT7621 myself now and test on a similar board.  

I suspect that either of those two commit is the culprit:
d40691a5fb generic: 6.1, 6.6: mt7530: import pending patches
ac8bfe316b generic: 6.1, 6.6: mt7530: import accepted patches

closed 2024-04-27 by openwrt-bot