openwrt/openwrt #15280: [BANANAPI BPI-R4] kernel 6.1 image compilation error

drozdi70, 2024-04-26:
### Describe the bug

While creating new image for 6.1 kernel:

  MODPOST /home/openwrt/work/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-2024.04.03~1e336a85/Module.symvers
ERROR: modpost: "relay_reset" [/home/openwrt/work/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-2024.04.03~1e336a85/mt7996/mt7996e.ko] undefined!
ERROR: modpost: "relay_open" [/home/openwrt/work/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-2024.04.03~1e336a85/mt7996/mt7996e.ko] undefined!
ERROR: modpost: "relay_switch_subbuf" [/home/openwrt/work/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-2024.04.03~1e336a85/mt7996/mt7996e.ko] undefined!
ERROR: modpost: "relay_flush" [/home/openwrt/work/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-2024.04.03~1e336a85/mt7996/mt7996e.ko] undefined!
ERROR: modpost: "relay_file_operations" [/home/openwrt/work/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-2024.04.03~1e336a85/mt7996/mt7996e.ko] undefined!
make[5]: *** [scripts/Makefile.modpost:127: /home/openwrt/work/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-2024.04.03~1e336a85/Module.symvers] Error 1
make[4]: *** [Makefile:1962: modpost] Error 2
make[4]: Leaving directory '/home/openwrt/work/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.86'
make[3]: *** [Makefile:638: /home/openwrt/work/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-2024.04.03~1e336a85/.built] Error 2
make[3]: Leaving directory '/home/openwrt/work/openwrt/package/kernel/mt76'
time: package/kernel/mt76/compile#66.80#10.45#77.96
    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:129: package/kernel/mt76/compile] Error 1
make[2]: Leaving directory '/home/openwrt/work/openwrt'
make[1]: *** [package/Makefile:123: /home/openwrt/work/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt/work/openwrt'
make: *** [/home/openwrt/work/openwrt/include/toplevel.mk:233: world] Error 2

### OpenWrt version

snapshot

### OpenWrt release

snaphot

### OpenWrt target/subtarget

BANANAPI

### Device

BPI-R4

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

drozdi70, 2024-04-26:
Yea clean from the scratchTemat: Re: [openwrt/openwrt] [BANANAPI BPI-R4] kernel 6.1 image compilation error (Issue #15280)Data: 2024-04-26 14:14Nadawca: "github-actions[bot]" ***@***.***&gt;Adresat: "openwrt/openwrt" ***@***.***&gt;; DW: "DroZDi" ***@***.***&gt;; "Author" ***@***.***&gt;; 

Invalid Target/Subtarget reported. BANANAPI
Is this from a supported device?

drozdi70, 2024-04-26:
Yea from the scratch

ynezz, 2024-04-27:
It would be really helpful if you could actually read the _complete_ text and update the issue accordingly, thanks!

[image]

drozdi70, 2024-04-28:
root@BPI-R4:~# . /etc/openwrt_release && echo $DISRIB_REVISION

root@BPI-R4:~# . /etc/openwrt_release && echo $DISRIB_RELEASE

root@BPI-R4:~# cat /tmp/sysinfo/model
Bananapi BPI-R4
root@BPI-R4:~# . /etc/openwrt_release && echo $DISRIB_TARGET

this is a snapshot image compiliation and for kernel 6.1

drozdi70, 2024-04-28:
openwrt@orangepi5-plus:~/work/openwrt$ ./scripts/diffconfig.sh
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r4=y
CONFIG_ATH11K_THERMAL=y
CONFIG_ATH_USER_REGD=y
# CONFIG_KERNEL_WERROR is not set
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_ATH_DFS=y
CONFIG_PACKAGE_ar=y
CONFIG_PACKAGE_ath11k-firmware-qcn9074=y
CONFIG_PACKAGE_binutils=y
CONFIG_PACKAGE_blkid=y
CONFIG_PACKAGE_cfdisk=y
CONFIG_PACKAGE_chat=y
CONFIG_PACKAGE_comgt=y
CONFIG_PACKAGE_comgt-ncm=y
CONFIG_PACKAGE_ethtool-full=y
CONFIG_PACKAGE_f2fs-tools=y
CONFIG_PACKAGE_fdisk=y
CONFIG_PACKAGE_findfs=y
CONFIG_PACKAGE_ipcs=y
CONFIG_PACKAGE_kmod-asn1-decoder=y
CONFIG_PACKAGE_kmod-asn1-encoder=y
CONFIG_PACKAGE_kmod-ata-core=y
CONFIG_PACKAGE_kmod-ath=y
CONFIG_PACKAGE_kmod-ath11k=y
CONFIG_PACKAGE_kmod-crypto-cbc=y
CONFIG_PACKAGE_kmod-crypto-michael-mic=y
CONFIG_PACKAGE_kmod-crypto-sha3=y
CONFIG_PACKAGE_kmod-dax=y
CONFIG_PACKAGE_kmod-dm=y
CONFIG_PACKAGE_kmod-eeprom-at25=y
CONFIG_PACKAGE_kmod-gpio-pca953x=y
CONFIG_PACKAGE_kmod-gpio-pcf857x=y
CONFIG_PACKAGE_kmod-hwmon-adt7475=y
CONFIG_PACKAGE_kmod-hwmon-drivetemp=y
CONFIG_PACKAGE_kmod-hwmon-gpiofan=y
CONFIG_PACKAGE_kmod-hwmon-vid=y
CONFIG_PACKAGE_kmod-i2c-algo-bit=y
CONFIG_PACKAGE_kmod-i2c-gpio=y
CONFIG_PACKAGE_kmod-i2c-mux-gpio=y
CONFIG_PACKAGE_kmod-i2c-mux-pca9541=y
CONFIG_PACKAGE_kmod-i2c-mux-reg=y
CONFIG_PACKAGE_kmod-i2c-smbus=y
CONFIG_PACKAGE_kmod-iio-core=y
CONFIG_PACKAGE_kmod-iwlwifi=y
CONFIG_PACKAGE_kmod-keys-encrypted=y
CONFIG_PACKAGE_kmod-keys-trusted=y
CONFIG_PACKAGE_kmod-mii=y
CONFIG_PACKAGE_kmod-mmc=y
CONFIG_PACKAGE_kmod-mt76-sdio=y
CONFIG_PACKAGE_kmod-mt76-usb=y
# CONFIG_PACKAGE_kmod-mt7915e is not set
CONFIG_PACKAGE_kmod-mt7921-common=y
CONFIG_PACKAGE_kmod-mt7921-firmware=y
CONFIG_PACKAGE_kmod-mt7921e=y
CONFIG_PACKAGE_kmod-mt7921s=y
CONFIG_PACKAGE_kmod-mt7921u=y
CONFIG_PACKAGE_kmod-mt7922-firmware=y
CONFIG_PACKAGE_kmod-mt7925-common=y
CONFIG_PACKAGE_kmod-mt7925e=y
CONFIG_PACKAGE_kmod-mt7925u=y
CONFIG_PACKAGE_kmod-mt792x-common=y
CONFIG_PACKAGE_kmod-mt792x-usb=y
CONFIG_PACKAGE_kmod-mt7981-firmware=y
CONFIG_PACKAGE_kmod-mt7986-firmware=y
CONFIG_PACKAGE_kmod-mtk-t7xx=y
CONFIG_PACKAGE_kmod-mux-core=y
CONFIG_PACKAGE_kmod-mux-gpio=y
CONFIG_PACKAGE_kmod-nvme=y
CONFIG_PACKAGE_kmod-oid-registry=y
CONFIG_PACKAGE_kmod-pps=y
CONFIG_PACKAGE_kmod-ptp=y
CONFIG_PACKAGE_kmod-random-core=y
CONFIG_PACKAGE_kmod-scsi-core=y
CONFIG_PACKAGE_kmod-spi-bitbang=y
CONFIG_PACKAGE_kmod-spi-dev=y
CONFIG_PACKAGE_kmod-spi-gpio=y
CONFIG_PACKAGE_kmod-tpm=y
CONFIG_PACKAGE_kmod-tun=y
CONFIG_PACKAGE_kmod-usb-acm=y
CONFIG_PACKAGE_kmod-usb-net=y
CONFIG_PACKAGE_kmod-usb-net-cdc-ether=y
CONFIG_PACKAGE_kmod-usb-net-cdc-ncm=y
CONFIG_PACKAGE_kmod-usb-net-huawei-cdc-ncm=y
CONFIG_PACKAGE_kmod-usb-net-qmi-wwan=y
CONFIG_PACKAGE_kmod-usb-net-rndis=y
CONFIG_PACKAGE_kmod-usb-serial=y
CONFIG_PACKAGE_kmod-usb-serial-option=y
CONFIG_PACKAGE_kmod-usb-serial-qualcomm=y
CONFIG_PACKAGE_kmod-usb-serial-wwan=y
CONFIG_PACKAGE_kmod-usb-wdm=y
CONFIG_PACKAGE_kmod-wwan=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libbfd=y
CONFIG_PACKAGE_libbz2=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libctf=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_libfdisk=y
CONFIG_PACKAGE_libmount=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libopcodes=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libpcre2=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_libsmartcols=y
CONFIG_PACKAGE_libstdcpp=y
CONFIG_PACKAGE_libsysfs=y
CONFIG_PACKAGE_libusb-1.0=y
CONFIG_PACKAGE_lsblk=y
CONFIG_PACKAGE_mt7921bt-firmware=y
CONFIG_PACKAGE_mt7922bt-firmware=y
CONFIG_PACKAGE_mt7981-wo-firmware=y
CONFIG_PACKAGE_mt7986-wo-firmware=y
CONFIG_PACKAGE_mt7988-2p5g-phy-firmware=y
CONFIG_PACKAGE_objdump=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-rpcsys=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_spidev-test=y
CONFIG_PACKAGE_sysfsutils=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_uqmi=y
CONFIG_PACKAGE_usb-modeswitch=y
CONFIG_PACKAGE_wireless-tools=y
CONFIG_PACKAGE_wwan=y
CONFIG_PACKAGE_zlib=y
CONFIG_TARGET_ROOTFS_PARTSIZE=3000

drozdi70, 2024-05-07:
as kenel 6.1 no more in use it can be closed

closed 2024-05-07 by drozdi70