openwrt/openwrt #15281: dropbear: DROPBEAR_MODERN_ONLY has no effect when disabled, sha1 is still missing

M95D, 2024-04-26:
### Describe the bug

DROPBEAR_MODERN_ONLY has no effect when it is disabled. Dropbear doesn't support sha1 key exchange no matter if the option is enabled or not.

### OpenWrt version

r26027

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Turris Omnia

### Image kind

Self-built image

### Steps to reproduce

Disable DROPBEAR_MODERN_ONLY in menuconfig.
Build and write the firmware.
Try to login with a client that only supports sha1.

### Actual behaviour

Fail to login. Dropbear reports `No matching algo kex.` in the log. Client reports "Couldn't agree a key exchange algorithm." or equivalent message.

### Expected behaviour

Login success.

### Additional info

These changes were part of the update by @rockdrilla here: 
https://github.com/openwrt/openwrt/commit/2d9a0be307b534ceb717267c95402d1d707cd2c3
... but not included in the DROPBEAR_MODERN_ONLY selection mechanism here:
https://github.com/openwrt/openwrt/commit/bf900e02c7102601be2e9280231711e70f065877

Thanks.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rockdrilla, 2024-04-26:
@M95D thanks for reporting! SHA1 is almost completely disabled with commit 2d9a0be except RSA-SHA1 signature algorithm. Config option DROPBEAR_MODERN_ONLY disables it too to completely remove SHA1-related functions (and thus reduce binary size).

rockdrilla, 2024-04-26:
@graysky2 may I ask you to correct me if I'm wrong in something?

M95D, 2024-04-26:
What's the purpose of DROPBEAR_MODERN_ONLY, then?
You either support clients that only know SHA1 (including sha1 key exchange) or you don't.

rockdrilla, 2024-04-26:
@M95D purpose of DROPBEAR_MODERN_ONLY is described in corresponding commit message and Kconfig stanza - only modern, strong and fast algorithms, also binary size optimization at last. SHA1 support is almost completely removed due to weak security guarantees in modern world. More compatibility means less security - at least in case of Dropbear.

graysky2, 2024-04-26:
@rockdrilla - I think you're correct.  For more context: https://github.com/mkj/dropbear/issues/138

M95D, 2024-04-26:
You don't get it and I'm too tired to argue. Closing.

closed 2024-04-26 by M95D