openwrt/openwrt #15283: TP-Link TL-WDR4900 v1 (N900) - wifi switch port stops forwarding packets on reception of certain multicast ethernet packets (e.g. MACsec Key Agreement).

linuxtim, 2024-04-26:
### Describe the bug

TP-Link TL-WDR4900 v1 (N900) - wifi switch port stops forwarding packets on reception of certain multicast Ethernet packets (e.g. MACsec Key Agreement).

802.1X MACsec Key Agreement (MKA) EAPOL (Extensible Authentication Protocol over LAN) packets use special multicast addresses - usually `01:80:C2:00:00:03` but sometimes other multicast or unicast addresses.  Reception of these packets on the external Ethernet interface seems to make the internal WDR4900 Ethernet switch stop forwarding packets.  This happens if the device is not participating in any way in the key agreement, just the reception of the packets generated by other "advertisement" packets by other devices during their key agreements etc. on the network causes the problem.

If I get time, I will try and narrow down this bug, but it appears to be present with both the older (22.03) switchdev builds and newer DSA (23.05) builds.

I'll also see if I can find the original OEM firmware to see if it has the same fault.  This issue isn't impacting me directly at the moment because I've been able to filter these packets from the TL-WDR4900 at an upstream Ethernet switch, but it was "interesting" 

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

mpc85xx/p1010

### Device

TP-Link TL-WDR4900 v1

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-04-26:
"certain packet" kind of demands pcap.
Please be more specific - is it wifi port in br-lan, br-lan or the hardware switch without relation to wifi?

linuxtim, 2024-05-11:
The issue appears to be with the hardware switch.  I'll try and debug this further next week.