openwrt/openwrt #15284: Enable Renesas RZ/G2L Platform

JohnXVincent, 2024-04-26:
### Describe the bug

Hi
I tried installing openWrt release 23.05.3 on Renesas Smarc RZ platform https://renesas.info/wiki/RZ-G/RZ-G2L_SMARC
The serial module is not enabled and hence the installation got stuck before starting the kernel.
This needs the following configurations required to be enabled.
CONFIG_SERIAL_SH_SCI=y
#The following are dependent configuration for CONFIG_SERIAL_SH_SCI=y
CONFIG_SERIAL_SH_SCI_NR_UARTS=18
CONFIG_SERIAL_SH_SCI_CONSOLE=y
CONFIG_SERIAL_SH_SCI_EARLYCON=y
CONFIG_SERIAL_SH_SCI_DMA=y

Please support to enable them in the file target/linux/armsr/armv8/config-6.1

Best Regards
John

### OpenWrt version

 23.05.3 

### OpenWrt release

 23.05.3 

### OpenWrt target/subtarget

arm64

### Device

arm64

### Image kind

Official downloaded image

### Steps to reproduce

1. Prepare USB disk with the image https://mirror.autkin.net/openwrt/releases/23.05.3/targets/armsr/armv8/openwrt-23.05.3-armsr-armv8-generic-ext4-combined.img.gz
2. Start RZ/G2L platform. It start the grub menu with options "OpenWrt", "OpenWrt Safe mode"
3. Select OpenWrt, 

### Actual behaviour

Stuck with the following message:
EFI stub: Booting Linux Kernel...
EFI stub: EFI_RNG_PROTOCOL unavailable
EFI stub: Using DTB from configuration table
EFI stub: Exiting boot services...

### Expected behaviour

Boot successful

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-04-27:
That file does not exist
https://github.com/openwrt/openwrt/tree/openwrt-23.05/target/linux/armsr

JohnXVincent, 2024-04-27:
I was referring the configuration file https://github.com/openwrt/openwrt/blob/master/target/linux/armsr/armv8/config-6.1 from master branch.

mcbridematt, 2024-04-30:
@JohnXVincent are you able to test the kernel 6.6 tree at https://github.com/openwrt/openwrt/pull/14896 ? Any changes can be backported the release (23.05) branch easily?

JohnXVincent, 2024-04-30:
@mcbridematt There are some additional settings required for enabling the ethernet driver. Are you able to add them?

CONFIG_MII=y
CONFIG_NET_VENDOR_RENESAS=y
CONFIG_SH_ETH=y
CONFIG_RAVB=y
CONFIG_RENESAS_ETHER_SWITCH=y
CONFIG_MICREL_PHY=y
CONFIG_PHYLIB=y
CONFIG_PHYLINK=y
CONFIG_USB_STORAGE=y
CONFIG_PTP_1588_CLOCK=y
CONFIG_PTP_1588_CLOCK_OPTIONAL=y 
 
Also inittab requires ttySC0  (target/linux/armsr/base-files/etc/inittab)

ttySC0::askfirst:/usr/libexec/login.sh

Please add this line in file inittab to include ttySC0

I'll test kernel 6.6 with these changes. Thanks

commit 531b3f66 (2024-06-16): kernel: armsr: Renesas: RZ: Ethernet module and ttySC0

Ethernet module enable for Renesas RZ platform
inittab fix for ttySC0

Fixes: #15284
commit 299893c9 (2024-07-06): kernel: armsr: Renesas: RZ: Ethernet module and ttySC0

Ethernet module enable for Renesas RZ platform
inittab fix for ttySC0

Fixes: #15284

(cherry picked from commit 531b3f667c40405581a398a9648a0e6c27909a87)
Link: https://github.com/openwrt/openwrt/pull/15808
closed 2024-06-16 by openwrt-bot