openwrt/openwrt #15286: Please add China CPE510 support to safeloader list

brada4, 2024-04-27:
### Describe the bug

As identified at https://forum.openwrt.org/t/openwrt-cannot-be-installed-on-tl-cpe510-in-the-china-region/196073
TL-CPE510(TP-LINK|CN|N300-5):1.0
According to "pharos" image all devices have same flash layout.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

x86/64

### Device

rocketship

### Image kind

Official downloaded image

### Steps to reproduce

tplink-safeloader compat device list needs to be extended to boot on CN version of the device

### Actual behaviour

fw rejected

### Expected behaviour

fw accepted

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

musashino205, 2024-04-27:
This is not a bug. Please don't use the bug tracker for requests...

006w, 2024-04-28:
Thank you very much for your article. I happen to have this TL-CPE510 device produced in China, and attached the original firmware download. I hope someone can make a firmware for me. I will pay attention to this post frequently and help test feedback as soon as possible. First of all Thank you, because I don’t know English. I use translation software for typing. I hope you can forgive me if it’s not fluent.

Official firmware website: https://smb.tp-link.com.cn/service/detail_download_2492.html
Download address: https://service.tp-link.com.cn/download/20161/TL-CPE510%20V1.0%E5%8D%87%E7%BA%A7%E8%BD%AF%E4%BB %B620160108.zip

[image]
[image]

brada4, 2024-04-28:
If you could try compiling firmware and confirm device id€ntifier for tp-safeloader

006w, 2024-04-28:
Thank you for your reply, I'm a war five scum (combat effectiveness is only 5) computer level, English needs translation software, can't compile, thank you for your reply

brada4, 2024-04-28:
Could you ask at work where you got the device if they have a test device with serial and are able to confirm the hardware identifier. Pictures disagree with software, they say 1.1 in chinese, but there is just 1.0 CN hardware supported by vendor, so we dont know.

006w, 2024-04-29:
First of all, thank you for your reply, maybe the version of each country and region is not the same, China is generally like this, the first version is 1.0, continue to come out 1.1, 1.2, 1.3 such a hardware product, his chip is exactly the same, the official firmware without special instructions, that is, the firmware is universal, the only difference may be which version is less resistor, which version is less capacitor, or less a network port, but basically the chip is unchanged, the firmware can be universal, so the official firmware is 1.0, and the version number of my device is 1.1, which is also applicable

006w, 2024-04-29:
Send you a picture of my equipment information, maybe you can't read Chinese, but you can mainly pay attention to the device information, as well as the version number and other useful information
[image]
[image]
[image]
[image]

brada4, 2024-04-29:
Thank you for this, first picture mentions it is 1.0 also uses Chinese market name with TL-
[image]

006w, 2024-04-29:
I saw that the forum said that you can upload the firmware to TEM in advance through SSH, forcibly flash the test, and I will go back after work at night to see if it works

006w, 2024-04-29:
The general version here starts from 1.0,Theoretically1.0-1.9 are minor changes,Save a lamp,Network port,Capacitor,Resistance or something,2.0The version that starts later,Maybe flash,ROM may be halved,There is also a change of CPU or main control,So,You must pay attention to the model and version number when buying equipment here,Otherwise, it is easy to appear that the equipment you buy cannot meet your own requirements

brada4, 2024-04-29:
Can you take out mainboard (it is like drawer inside the plastic shell) and get the picture of flash chips, at least if it is wired the way supported things are wired.
PS so far it seems only printer ink got versioned

006w, 2024-04-29:
Hello, I obtained the CPU information through SSH. Is this OK?

system type             : Atheros AR934x
processor               : 0
cpu model               : MIPS 74Kc V4.12
BogoMIPS                : 279.55
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 32
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0000, 0x0120, 0x00                      00, 0x0000]
ASEs implemented        : mips16 dsp
shadow register sets    : 1
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available

brada4, 2024-04-29:
It is CN 1.0 as in similar issue: 
https://github.com/openwrt/openwrt/issues/12939
But you have to build and test, take your time or wait (possibly long) for somebody else to add device support....
Maybe @lynxis can sched some light...

006w, 2024-04-30:
Thank you very much for your kind reply, I will also be watching his post closely, thank you

brada4, 2024-04-30:
In the meantime you can try to build OpenWRT with modified tplink-safeloader and at least check if it passes updater validation. If you had 10 routers like taken from office walls you would not care much to brick one not having serial console...

006w, 2024-04-30:
Do you mean to use the general firmware upgrade to test whether it will pass the firmware validation, I don't know where to find it, I know a little bit about TTL, indicating that this guy may have to solder the flying wire, in order to connect to the TTL for the programmer firmware flash, where can I find this tplink-safeloader you said?

brada4, 2024-04-30:
> where

In forum post, there is a source file with "suppoted" CPE510

006w, 2024-04-30:
https://forum.openwrt.org/

Thank you for your reply, I didn't find this result in the openwrt forum
[image]
[image]

brada4, 2024-04-30:
We talked in forum:
https://github.com/openwrt/openwrt/blob/373dc11b519516682ab48bf7db501ffffced4e61/tools/firmware-utils/src/tplink-safeloader.c#L164

006w, 2024-04-30:
See this message you sent me,I just don't understand how this is used,Generally I understand the flash,For example, OPENWRT introduced,Brush a bottom package first,And then brush the latest version in the bottom package,Or TTL,Use the flash command,Put the UBOOT or firmware that needs to be flashed into tmp to execute it can be written,I see this you sent,But I didn't download the file,I didn't understand how to use it

brada4, 2024-04-30:
You need to rebuild firmware with changed "safeloader" that adds suspect compatibility string to firmware image and check that OEM upgrade menu does not reject it.

006w, 2024-04-30:
Thank you very much, I see that I still need a linux system, enter some instructions, it is estimated that it cannot be realized, and the compilation is not good, and it may become bricked, thank you for your reply, I'll wait a little longer, is there a great god or an official can give a firmware, thank you brada4 for your reply, you are very enthusiastic

brada4, 2024-04-30:
it is called a virtual machine, take your time.

Neustradamus, 2024-08-30:
@006w: Any progress?

006w, 2024-08-31:
Hallo, niks het gevorder nie

closed 2025-12-29 by brada4