openwrt/openwrt #15304: TUF AX6000 br-lan ports do not work on Snapshot r26060 but on 25.05.3 r23809

magnus1089, 2024-04-27:
### Describe the bug

When running Snapshot r26060 the br-lan ports don't work. ip a repots that lan4@eth0 are UP even when only lan 0 are attached.  When running 23..05.3 r23809 ip a reports that lan0@eth0 are UP and br-lan works. 
WAN port working fine on both version of software.
Also when moving ethernet cable to lan1, lan2, lan3, lan4, lan5 it works ok on 23.05.3 r23809 but not on Snapshot r26060
ping to local br-lan 192.168.1.1 work with both software. But ping to my laptop 192.168.1.10 will not work on Snapshot r26060 on lan1-4 but will work on lan5 but luci will load first page "LuCI - Lua Configuration Interface" but not load any further from laptop and ping will some time fails from laptop.

### OpenWrt version

r26060-a8dde7e5bd

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

ASUS TUF-AX6000

### Image kind

Official downloaded image

### Steps to reproduce

Try to connect using br-lan ports both direction will fail on lan1-lan4 when running Snapshot 26060

### Actual behaviour

I don't know what happens but when attach ethenet cable on lan1 port the serial terminal shows lan4 when running Snapshot 26060. But shows correct lan1 when running 23.05.3 r23809 

### Expected behaviour

br-lan ports should work in snapshot 26060

### Additional info

[Snapshot r26060 bootlog.txt](https://github.com/openwrt/openwrt/files/15140037/Snapshot.r26060.bootlog.txt)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

magnus1089, 2024-04-27:
[23.05.3 r23809 bootlog.txt](https://github.com/openwrt/openwrt/files/15140046/23.05.3.r23809.bootlog.txt)

magnus1089, 2024-04-28:
This change seems to fix the problem.
diff --git a/target/linux/mediatek/dts/mt7986a-asus-tuf-ax6000.dts b/target/linux/mediatek/dts/mt7986a-asus-tuf-ax6000.dts
index 1cdfb5f155..58ed318d62 100644
--- a/target/linux/mediatek/dts/mt7986a-asus-tuf-ax6000.dts
+++ b/target/linux/mediatek/dts/mt7986a-asus-tuf-ax6000.dts
@@ -298,25 +298,25 @@
                port@1 {
                        reg = <4>;
                        label = "lan1";
-                       phy-handle = <&swphy1>;
+                       phy-handle = <&swphy4>;
                };

                port@2 {
                        reg = <3>;
                        label = "lan2";
-                       phy-handle = <&swphy2>;
+                       phy-handle = <&swphy3>;
                };

                port@3 {
                        reg = <2>;
                        label = "lan3";
-                       phy-handle = <&swphy3>;
+                       phy-handle = <&swphy2>;
                };

                port@4 {
                        reg = <1>;
                        label = "lan4";
-                       phy-handle = <&swphy4>;
+                       phy-handle = <&swphy1>;
                };

                port@5 {

magnus1089, 2024-04-28:
Fix:
[diff that fixes issue 15304.txt](https://github.com/openwrt/openwrt/files/15142444/diff.that.fixes.issue.15304.txt)

namiltd, 2024-05-01:
Could you please generate a Pull Request for this patch?

magnus1089, 2024-05-01:
I can try.
> Could you please generate a Pull Request for this patch?

magnus1089, 2024-05-11:
Fixed in pull request #15353 that are now merged.

closed 2024-05-11 by magnus1089