openwrt/openwrt #15308: [BANANAPI BPI-R4] kernel 6.6 image stops

drozdi70, 2024-04-28:
### Describe the bug

after new compilation from a scratch for kernel 6.6 OS disabe CPUs:

[   16.842938] mtk_t7xx 0003:01:00.0: Packet drop on channel 0x1004, port not found
[   16.850347] mtk_t7xx 0003:01:00.0: Packet drop on channel 0x1012, port not found
[   21.360949] wwan wwan0: port wwan0at0 attached
[   21.365510] wwan wwan0: port wwan0mbim0 attached
[   21.746309] mtk_t7xx 0003:01:00.0: Port AT is not opened, drop packets
[   22.060760] mtk_t7xx 0003:01:00.0: Port AT is not opened, drop packets
[   22.067292] mtk_t7xx 0003:01:00.0: Port AT is not opened, drop packets
[   22.073824] mtk_t7xx 0003:01:00.0: Port AT is not opened, drop packets
[   22.899764] mtk_t7xx 0003:01:00.0: Packet drop on channel 0x100a, port not found
[   24.289329] mtk_t7xx 0003:01:00.0: Port AT is not opened, drop packets
[   24.625518] mtk_t7xx 0003:01:00.0: Port AT is not opened, drop packets
[   36.312371] vproc: disabling

### OpenWrt version

r26027-ee4e69cc35

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Self-built image

### Steps to reproduce

New image compiled for kernel 6.6, starting from SD card but after some seconds vproc: disabling

### Actual behaviour

OS stopped

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r4=y
CONFIG_ATH11K_THERMAL=y
CONFIG_ATH_USER_REGD=y
# CONFIG_KERNEL_WERROR is not set
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
CONFIG_LIBQMI_COLLECTION_BASIC=y
CONFIG_LIBQMI_WITH_MBIM_QMUX=y
CONFIG_LIBQMI_WITH_QRTR_GLIB=y
CONFIG_LIBSSH2_OPENSSL=y
CONFIG_LINUX_6_6=y
# CONFIG_LUCI_CSSTIDY is not set
# CONFIG_LUCI_JSMIN is not set
[... 250 lines cut ...]
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_uqmi=y
CONFIG_PACKAGE_usb-modeswitch=y
CONFIG_PACKAGE_usbids=y
CONFIG_PACKAGE_usbutils=y
CONFIG_PACKAGE_vim=y
CONFIG_PACKAGE_wireless-tools=y
CONFIG_PACKAGE_wwan=y
CONFIG_PACKAGE_zlib=y
CONFIG_PACKAGE_zsh=y
CONFIG_PARTED_READLINE=y
CONFIG_PERL_NOCOMMENT=y
CONFIG_PERL_THREADS=y
CONFIG_TARGET_ROOTFS_PARTSIZE=3000
CONFIG_TESTING_KERNEL=y
CONFIG_ZSTD_OPTIMIZE_O3=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

drozdi70, 2024-05-04:
again from the scratch:

 OpenWrt SNAPSHOT, r26171-3f28c422ba

root@OpenWrt:/# vi [   36.312433] vproc: disabling

danpawlik, 2024-05-05:
I'm also doing compilation from scratch and it works normally. (6bb2551)
Just curious, are you using eMMC or sdcard?

drozdi70, 2024-05-05:
yeah now it is ok, problem fixed

closed 2024-05-05 by drozdi70