openwrt/openwrt #15312: multiple definition of `xmalloc'

trippleflux, 2024-04-28:
### Describe the bug

When compiling net/frr , I am getting linker errors : 
```
ccache x86_64-openwrt-linux-gnu-gcc  -I. -Iinclude -Iproviders/common/include -Iproviders/implementations/include  -DAES_ASM -DBSAES_ASM -DCMLL_ASM -DECP_NISTZ256_ASM -DGHASH_ASM -DKECCAK1600_ASM -DMD5_ASM -DOPENSSL_BN_ASM_GF2m -DOPENSSL_BN_ASM_MONT -DOPENSSL_BN_ASM_MONT5 -DOPENSSL_CPUID_OBJ -DOPENSSL_IA32_SSE2 -DPOLY1305_ASM -DRC4_ASM -DSHA1_ASM -DSHA256_ASM -DSHA512_ASM -DVPAES_ASM -DWHIRLPOOL_ASM -DX25519_ASM -fPIC -pthread -m64 -Wa,--noexecstack -Wall -O3 -march=tigerlake -mtune=tigerlake -funsafe-math-optimizations -ffp-contract=fast -ftree-vectorize -fno-caller-saves -fno-plt -march=tigerlake -mtune=tigerlake -funsafe-math-optimizations -ffp-contract=fast -ftree-vectorize -fhonour-copts -fmacro-prefix-map=/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/openssl-openssl-3.1.5-quic1=openssl-openssl-3.1.5-quic1 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -O3 -DPIC -fpic -march=tigerlake -mtune=tigerlake -funsafe-math-optimizations -ffp-contract=fast -ftree-vectorize -fno-caller-saves -fno-plt -march=tigerlake -mtune=tigerlake -funsafe-math-optimizations -ffp-contract=fast -ftree-vectorize -fhonour-copts -fmacro-prefix-map=/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/openssl-openssl-3.1.5-quic1=openssl-openssl-3.1.5-quic1 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -O3 -fpic -fuse-ld=bfd -znow -zrelro -DOPENSSL_USE_NODELETE -DL_ENDIAN -DOPENSSL_PIC -DOPENSSLDIR="\"/etc/ssl\"" -DENGINESDIR="\"/usr/lib/engines-3\"" -DMODULESDIR="\"/usr/lib/ossl-modules\"" -DOPENSSL_BUILDING_OPENSSL -DZLIB -DZLIB_SHARED -DNDEBUG -D_FORTIFY_SOURCE=1 -DPIC -DOPENSSL_PREFER_CHACHA_OVER_GCM -I/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-13.2.0_glibc/usr/include -I/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-13.2.0_glibc/include -MMD -MF crypto/evp/libcrypto-lib-pmeth_lib.d.tmp -MT crypto/evp/libcrypto-lib-pmeth_lib.o -c -o crypto/evp/libcrypto-lib-pmeth_lib.o crypto/evp/pmeth_lib.c
/usr/bin/ld: /home/username/works/openwrt/staging_dir/host/lib/libelf.a(xmalloc.o): in function `xmalloc':
xmalloc.c:(.text+0x0): multiple definition of `xmalloc'; /home/username/works/openwrt/staging_dir/host/lib/libelf.a(libgnu_la-xmalloc.o):xmalloc.c:(.text+0x0): first defined here
/usr/bin/ld: /home/username/works/openwrt/staging_dir/host/lib/libelf.a(xmalloc.o): in function `xcalloc':
xmalloc.c:(.text+0x30): multiple definition of `xcalloc'; /home/username/works/openwrt/staging_dir/host/lib/libelf.a(libgnu_la-xmalloc.o):xmalloc.c:(.text+0x180): first defined here
/usr/bin/ld: /home/username/works/openwrt/staging_dir/host/lib/libelf.a(xmalloc.o): in function `xrealloc':
xmalloc.c:(.text+0x60): multiple definition of `xrealloc'; /home/username/works/openwrt/staging_dir/host/lib/libelf.a(libgnu_la-xmalloc.o):xmalloc.c:(.text+0x40): first defined here
/usr/bin/ld: /home/username/works/openwrt/staging_dir/host/lib/libelf.a(xstrdup.o): in function `xstrdup':
xstrdup.c:(.text+0x0): multiple definition of `xstrdup'; /home/username/works/openwrt/staging_dir/host/lib/libelf.a(libgnu_la-xmalloc.o):xmalloc.c:(.text+0x270): first defined here
cc1: note: someone does not honour COPTS correctly, passed 2 times
```

Does this related with the recent elfutils related upgrade?, because it doesn't happened before these.

### OpenWrt version

6e5edec159e2dc468607a5b9179f722857a38421

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86_64

### Device

generic

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

trippleflux, 2024-04-28:
net/frr need the latest vanilla one from their github : 
```
PKG_NAME:=frr
PKG_VERSION:=10.1

PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL:=https://github.com/FRRouting/frr.git
PKG_MIRROR_HASH:=4cb2fbda16bfd0b04c24a4f7c8adf6eaff0aa1f83cf77a9d4c8e7fc07be51120
PKG_SOURCE_DATE:=20240428
PKG_SOURCE_VERSION:=5f059e0c7d2b6b8bce0a618726ba5d731f2d2ad4
```

And delete "./feeds/packages/net/frr/patches/996-qpb_header_fix.patch"

BKPepe, 2024-04-28:
This should be moved into packages folder as the package frr is there.

trippleflux, 2024-04-28:
Sorry for these, I thought it's elfutils related, but it's self modification frr package ?

trippleflux, 2024-04-28:
Ok It's also happened in vanilla Openwrt frr package.

closed 2024-04-28 by trippleflux