openwrt/openwrt #15314: [22.03, 23.05] MacOS Ventura: toolchain/gdb: fails to compile

BKPepe, 2024-04-28:
### Describe the bug

Since MacOS was updated to Ventura including CI/CD, toolchain gdb can not be compiled on OpenWrt 23.05 and OpenWrt 22.03.

### OpenWrt version

23.05.3 and OpenWrt 22.03.6

### OpenWrt release

23.05.3 and OpenWrt 22.03.6

### OpenWrt target/subtarget

mvebu/cortex-a9

### Device

Turris Omnia

### Image kind

Official downloaded image

### Steps to reproduce

1. Clone OpenWrt repository
2. Checkout any stable branch, which is supported
3. Try to compile tools, toolchain
4. toolchain gdb fails.

### Actual behaviour

toolchain/gdb fails to compile:

```
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 15] for this enumeration type [-Wenum-constexpr-conversion]
    integer_for_size<sizeof (T), static_cast<bool>(T (-1) < T (0))>::type
                                                   ^
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 1] for this enumeration type [-Wenum-constexpr-conversion]
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 3] for this enumeration type [-Wenum-constexpr-conversion]
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 3] for this enumeration type [-Wenum-constexpr-conversion]
4 errors generated.
make[5]: *** [Makefile:2362: ada-exp.o] Error 1
In file included from annotate.c:19:
In f
ile included from ./defs.h:65:
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 15] for this enumeration type [-Wenum-constexpr-conversion]
    integer_for_size<sizeof (T), static_cast<bool>(T (-1) < T (0))>::type
                                                   ^
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 1] for this enumeration type [-Wenum-constexpr-conversion]
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 3] for this enumeration type [-Wenum-constexpr-conversion]
3 errors generated.
make[5]: *** [Makefile:1655: annotate.o] Error 1
In file included from ada-lang.c:21:
In file included from ./defs.h:65:
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 15] for this enumeration type [-Wenum-constexpr-conversion]
    integer_for_size<sizeof (T), static_cast<bool>(T (-1) < T (0))>::type
                                                   ^
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 1] for this enumeration type [-Wenum-constexpr-conversion]
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 3] for this enumeration type [-Wenum-constexpr-conversion]
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 3] for this enumeration type [-Wenum-constexpr-conversion]
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 3] for this enumeration type [-Wenum-constexpr-conversion]
5 errors generated.
make[5]: *** [Makefile:1655: ada-lang.o] Error 1
make[5]: Leaving directory '/Volumes/OpenWrt/omnia-hbd/build/build_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.3.0_musl_eabi/gdb-12.1/gdb'
make[4]: *** [Makefile:11484: all-gdb] Error 2
make[4]: Leaving directory '/Volumes/OpenWrt/omnia-hbd/build/build_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.3.0_musl_eabi/gdb-12.1'
make[3]: *** [Makefile:1030: all] Error 2
make[3]: Leaving directory '/Volumes/OpenWrt/omnia-hbd/build/build_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.3.0_musl_eabi/gdb-12.1'
make[2]: *** [Makefile:75: /Volumes/OpenWrt/omnia-hbd/build/build_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.3.0_musl_eabi/gdb-12.1/.built] Error 2
```

### Expected behaviour

gdb can be compiled on MacOS Ventura

### Additional info

Some details are here: https://github.com/openwrt/openwrt/issues/15270#issuecomment-2081401824

We were able to track it down to this commit: 
https://sourceware.org/git/?p=binutils-gdb.git;a=commit;h=ae61525fcf456ab395d55c45492a106d1275873a , which should fix that, but we backported it, but no luck. It introduced another error.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

olivier-lu, 2024-06-10:
Hi,
I am trying to compile v23.05.3 on macOS 14.5 but I am still getting the error eventhough, I see that patch file at toolchain/gdb/patches/0001-gdbsupport-ignore-Wenum-constexpr-conversion-in-enum.patch

This is the error I am seeing:

```
make[6]: Entering directory '/Volumes/OpenWrt-23.05/openwrt/build_dir/toolchain-arm_cortex-a9_gcc-12.3.0_musl_eabi/gdb-12.1/gdb'
g++ -x c++ -std=gnu++11   -I. -I. -I./config -DLOCALEDIR="\"/Volumes/OpenWrt-23.05/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-12.3.0_musl_eabi/share/locale\"" -DHAVE_CONFIG_H -I./../include/opcode -I./../readline/readline/.. -I./../zlib -I../bfd -I./../bfd -I./../include -I../libdecnumber -I./../libdecnumber -I./../intl -I./../gnulib/import -I../gnulib/import -I./.. -I.. -I./../libbacktrace/ -I../libbacktrace/  -DTUI=1 -I/Volumes/OpenWrt-23.05/openwrt/staging_dir/host/include    -I./.. -pthread  -Wall -Wpointer-arith -Wno-unused -Wunused-value -Wunused-function -Wno-switch -Wno-char-subscripts -Wempty-body -Wunused-but-set-parameter -Wunused-but-set-variable -Wno-sign-compare -Wno-mismatched-tags -Wno-error=deprecated-register -Wsuggest-override -Wdeprecated-copy -Wdeprecated-copy-dtor -Wredundant-move -Wmissing-declarations -Wmissing-prototypes -Wformat -Wformat-nonliteral   -c -o gdb.o -MT gdb.o -MMD -MP -MF ./.deps/gdb.Tpo gdb.c
In file included from gdb.c:19:
In file included from ./defs.h:65:
./../gdbsupport/enum-flags.h:95:52: error: integer value -1 is outside the valid range of values [0, 15] for this enumeration type [-Wenum-constexpr-conversion]
    integer_for_size<sizeof (T), static_cast<bool>(T (-1) < T (0))>::type
                                                   ^
1 error generated.
make[6]: *** [Makefile:1655: gdb.o] Error 1
make[6]: Leaving directory '/Volumes/OpenWrt-23.05/openwrt/build_dir/toolchain-arm_cortex-a9_gcc-12.3.0_musl_eabi/gdb-12.1/gdb'
make[5]: *** [Makefile:11484: all-gdb] Error 2
make[5]: Leaving directory '/Volumes/OpenWrt-23.05/openwrt/build_dir/toolchain-arm_cortex-a9_gcc-12.3.0_musl_eabi/gdb-12.1'
make[4]: *** [Makefile:1030: all] Error 2
make[4]: Leaving directory '/Volumes/OpenWrt-23.05/openwrt/build_dir/toolchain-arm_cortex-a9_gcc-12.3.0_musl_eabi/gdb-12.1'
make[3]: *** [Makefile:75: /Volumes/OpenWrt-23.05/openwrt/build_dir/toolchain-arm_cortex-a9_gcc-12.3.0_musl_eabi/gdb-12.1/.built] Error 2
make[3]: Leaving directory '/Volumes/OpenWrt-23.05/openwrt/toolchain/gdb'
time: toolchain/gdb/compile#1.40#0.46#2.27
    ERROR: toolchain/gdb failed to build.
make[2]: *** [toolchain/Makefile:93: toolchain/gdb/compile] Error 1
make[2]: Leaving directory '/Volumes/OpenWrt-23.05/openwrt'
make[1]: *** [toolchain/Makefile:89: /Volumes/OpenWrt-23.05/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-12.3.0_musl_eabi/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt-23.05/openwrt'
make: *** [/Volumes/OpenWrt-23.05/openwrt/include/toplevel.mk:232: world] Error 2
```

Any idea why the patch didn't work for me?

cichiwskyj, 2025-04-10:
I started today, trying to compile 24.10.0 and it seems to fail on the same issue of toolchain/gdb.