openwrt/openwrt #15327: Bridger doesn't work for this network config

rany2, 2024-04-29:
### Describe the bug

When setting up bridger, the `/sys/kernel/debug/ppe0/bind` is empty for the following `/etc/config/network` but it works if only VLAN filtering is used and no bridges containing `br-vlan` (i.e., there is no bridge called `br-vlan1` which contains `br-vlan.1`).

Unfortunately this is not an option for me because I have a RADIUS server which means that hostapd will always end up creating a new bridge with `br-vlan.X` where X is the VLAN id.

I'm not sure if this is a bug with bridger or the NAT accelerator. The `/etc/config/network` is as follows:

```

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option packet_steering '1'

config interface 'lan'
	option device 'br-vlan1'
	option proto 'static'
	option ipaddr 'x'
	option netmask '255.255.255.0'
	option gateway 'y'
	option delegate '0'

config device
	option name 'wan'
	option mtu '2022'

config device
	option name 'lan1'
	option mtu '2022'

config device
	option name 'lan2'
	option mtu '2022'

config device
	option name 'lan3'
	option mtu '2022'

config interface 'man'
	option proto 'none'
	option device 'br-vlan4093'

config interface 'guest'
	option proto 'none'
	option device 'br-vlan100'

config interface 'iot'
	option proto 'none'
	option device 'br-vlan200'

config device
	option type 'bridge'
	option name 'br-vlan1'
	list ports 'lan1.1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'wan'

config device
	option type 'bridge'
	option name 'br-vlan100'
	list ports 'lan1.100'

config device
	option type 'bridge'
	option name 'br-vlan200'
	list ports 'lan1.200'

config device
	option type 'bridge'
	option name 'br-vlan4093'
	list ports 'lan1.4093'
```

My `/etc/config/firewall` is as follows (there is no difference between enabling flow offloading or not):

```

config defaults
	option input 'DROP'
	option output 'ACCEPT'
	option forward 'DROP'
	option synflood_protect '1'
	option drop_invalid '1'
	option flow_offloading '1'
	option flow_offloading_hw '1'

config zone
	option name 'lan'
	option input 'DROP'
	option output 'ACCEPT'
	option forward 'DROP'
	list network 'lan'

config rule
	option name 'Allow-SSH'
	list proto 'tcp'
	option src '*'
	option dest_port '22'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMP'
	list proto 'icmp'
	option src '*'
	option target 'ACCEPT'

config rule
	option name 'Allow-HTTP'
	list proto 'tcp'
	option src 'lan'
	option dest_port '80'
	option target 'ACCEPT'

config rule
	option name 'Allow-HTTPS'
	list proto 'tcp'
	option src 'lan'
	option dest_port '443'
	option target 'ACCEPT'

config zone
	option name 'mgmt'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'DROP'
	list network 'man'

config zone
	option name 'untrusted'
	option input 'DROP'
	option output 'ACCEPT'
	option forward 'DROP'
	list network 'guest'
	list network 'iot'

```

I have an additional nftables file attached here (`/usr/share/nftables.d/ruleset-post/99-bridge.nft`) but it makes no difference when I delete it:

```
#!/usr/sbin/nft -f

# delete if exists, do not fail if not
# see https://github.com/openwrt/openwrt/issues/11620
add table bridge filter
add chain bridge filter forward
add chain bridge filter forward_guest
add chain bridge filter forward_iot
flush chain bridge filter forward
flush chain bridge filter forward_guest
flush chain bridge filter forward_iot
flush table bridge filter
delete chain bridge filter forward
delete chain bridge filter forward_guest
delete chain bridge filter forward_iot
delete table bridge filter

## VARIABLES ##
define GUEST_WIFI_BRIDGE = br-vlan100
define IOT_WIFI_BRIDGE = br-vlan200
define GUEST_VLAN = lan1.100
define IOT_VLAN = lan1.200

table bridge filter {
	chain forward_guest {
		iifname $GUEST_VLAN accept
		oifname $GUEST_VLAN accept
		drop
	}

	chain forward_iot {
		iifname $IOT_VLAN accept
		oifname $IOT_VLAN accept
		drop
	}

	chain forward {
		type filter hook forward priority filter; policy accept;
		ibrname $GUEST_WIFI_BRIDGE jump forward_guest
		ibrname $IOT_WIFI_BRIDGE jump forward_iot
	}
}
```

CC @nbd168 

### OpenWrt version

r26117-5c833329ce

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Xiaomi Redmi Router AX6S

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rany2, 2024-04-29:
This is also a `/etc/config/wireless` that triggers the bug, but I don't think it's required:

```
config wifi-device 'radio1'
	option type 'mac80211'
	option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option band '5g'
	option htmode 'HE80'
	option channel 'auto'
	option acs_exclude_dfs '1'
	option country '00'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option mode 'ap'
	option encryption 'psk2+ccmp'
	option ssid 'x'
	option wpa_psk_file '/etc/hostapd.wpa_psk'
	option vlan_bridge 'br-vlan'
	option per_sta_vif '1'
```

Obviously `/etc/modules.conf` contains `options mt7915e wed_enable=Y` but I just didn't provide this detail as it seemed obvious:

```
# examples:
# options mod1 option=val
# blacklist mod2
options mt7915e wed_enable=Y
```

gssjshark, 2024-06-26:
Did you ever get this issue resolved?

rany2, 2024-06-26:
Unfortunately not.

WuSiYu, 2024-12-12:
similar issue when setting an AP bridged to a VLAN eth interface