openwrt/openwrt #15328: qmi-wwan driver not found

philoftherock, 2024-04-29:
### Describe the bug

Using a self compiled image for the Gl.iNet E750, the qmi-wwan driver is not working with the built-in Quectel EM060K-GL 4G modem. uqmi commands return not found and /dev/cdc-wdm0 is not present.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

ath79/nand

### Device

GL.iNet GL-E750

### Image kind

Self-built image

### Steps to reproduce

I am using the official repos from github [https://github.com/openwrt/openwrt](url) and I make the build with menuconfig after updating | upgrading the feeds. I include the MCU package for the OLED screen [https://github.com/gl-inet/GL-E750-MCU-instruction](url) the packages listed in the user guide [https://openwrt.org/docs/guide-user/network/wan/wwan/ltedongle](url) so:
kmod-usb-net-qmi-wwan uqmi luci-proto-qmi kmod-usb-serial-option picocom

in my lastest attempt, I also tried to include those packages that are present in the officiel Gl.iNet firmware:
wwan
kmod-usb-net-qmi-wwan
kmod-usb-serial-wwan
uqmi
kmod-usb-acm
kmod-usb-core
kmod-usb-ehci
kmod-usb-net
kmod-usb-net-cdc-ether
kmod-usb-net-cdc-ncm
kmod-usb-net-huawei-cdc-ncm
kmod-usb-net-ipheth
kmod-usb-net-rndis
kmod-usb-ohci
kmod-usb-serial
kmod-usb-serial-option
kmod-usb-wwan
kmod-usb-storage
kmod-usb-uhci
kmod-usb-wdm
kmod-usb-usb2

### Actual behaviour

/dev/cdc-wdm0 is not present.

root@OpenWrt:~# AT+QCFG="usbnet"
`-ash: AT+QCFG=usbnet: not found`

root@OpenWrt:~# cat /sys/kernel/debug/usb/devices
```
T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  1, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.15
S:  Manufacturer=Linux 5.15.150 ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=1b000000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
[... 312 lines cut ...]
### Diffconfig

```text
#!/bin/sh
make ./scripts/config/conf >/dev/null || { make ./scripts/config/conf; exit 1; }
grep \^CONFIG_TARGET_ .config | head -n3 > tmp/.diffconfig.head
grep \^CONFIG_TARGET_DEVICE_ .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_KMODS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_NONSHARED=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_DEVEL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TOOLCHAINOPTS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_BUSYBOX_CUSTOM=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TARGET_PER_DEVICE_ROOTFS=y' .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage1 Config.in >/dev/null
./scripts/kconfig.pl '>+' tmp/.diffconfig.stage1 .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage2 Config.in >/dev/null
./scripts/kconfig.pl '>' tmp/.diffconfig.stage2 .config >> tmp/.diffconfig.head
cat tmp/.diffconfig.head
rm -f tmp/.diffconfig tmp/.diffconfig.head
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-04-30:
luci-proto-qmi is a good starting point, only one of 4 data modes is active at any one time, and you can change them by writing AT commands to picocom, not to command line.
https://openwrt.org/docs/guide-user/network/wan/wwan/ltedongle

philoftherock, 2024-04-30:
> luci-proto-qmi is a good starting point, only one of 4 data modes is active at any one time, and you can change them by writing AT commands to picocom, not to command line. https://openwrt.org/docs/guide-user/network/wan/wwan/ltedongle

How can I control the qmi with picocom if there's no /cdc/cdc-wdm0 ?

brada4, 2024-04-30:
You need too use serial port for serial commands...

philoftherock, 2024-04-30:
yes, I am using a virtual serial picocom. I can't use the command `picocom /dev/cdc-wdm0` because the port not does exist

I was successful with `picocom /dev/ttyUSB3` and I was able to connect with the 3G protocol and I was able to connect to the provider. https://openwrt.org/docs/guide-user/network/wan/wwan/3gdongle

Still isn't working with 4G/LTE though

brada4, 2024-04-30:
You need yo set one of qcfg modes as per lte doc.

philoftherock, 2024-04-30:
yes with the command picocom /dev/ttyUSB3 

```
AT+CGDCONT?
```
returns:
```
+CGDCONT: 1,"IPV4V6","CONNECT","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0
+CGDCONT: 2,"IPV4V6","ims","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0
+CGDCONT: 3,"IPV4V6","sos","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,1
```

Returns: OK

AT+QCFG="usbnet",0
Returns: OK

after reboot, again, the only way to get a connect is with the proto-3g. there are no qmi driver attatched to the modem

root@OpenWrt:~# cat /sys/kernel/debug/usb/devices
```

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  2, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.15
S:  Manufacturer=Linux 5.15.150 ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=1b000000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 4
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=05e3 ProdID=0618 Rev=87.54
S:  Product=USB2.0 Hub
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=100mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   1 Ivl=256ms

T:  Bus=01 Lev=02 Prnt=02 Port=00 Cnt=01 Dev#=  3 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=05e3 ProdID=0752 Rev= 2.33
S:  Manufacturer=Generic
S:  Product=USB Storage
C:* #Ifs= 1 Cfg#= 1 Atr=80 MxPwr= 98mA
I:* If#= 0 Alt= 0 #EPs= 2 Cls=08(stor.) Sub=06 Prot=50 Driver=usb-storage
E:  Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms

T:  Bus=01 Lev=02 Prnt=02 Port=01 Cnt=02 Dev#=  4 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=2c7c ProdID=030b Rev= 5.04
**S:  Manufacturer=Quectel
S:  Product=EM060K-GL
S:  SerialNumber=20e39aaa
C:* #Ifs= 6 Cfg#= 1 Atr=a0 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=30 Driver=option
E:  Ad=01(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 1 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=40 Driver=option
E:  Ad=83(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=82(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 2 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=40 Driver=option
E:  Ad=85(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=84(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 3 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=40 Driver=option
E:  Ad=87(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=86(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 4 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=88(I) Atr=03(Int.) MxPS=   8 Ivl=32ms
E:  Ad=8e(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=0f(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#=12 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=70 Driver=(none)
E:  Ad=89(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms**

```

brada4, 2024-04-30:
usb-modeswitch can do such initialisation
you can try saving parameters with AT&W 
what is the default USB protocol?

philoftherock, 2024-05-01:
I re

closed 2024-05-01 by philoftherock

brada4, 2024-05-01:
Literarily first line of picocom fragment, try to get desired mode after restart, and the network driver will attach.
```
AT+QCFG="usbnet"	# check the current mode
```

philoftherock, 2024-05-03:
I think you misunderstood. I did all those steps with QMI intially and it never worked. Doesn't matter now I got it working with MBIM

brada4, 2024-05-03:
You never shown default mode after reboot.