openwrt/openwrt #15338: mbedtls 3.6.0 error on MBEDTLS_CIPHER_MODE_CFB

peterwillcn, 2024-04-30:
### Describe the bug

mbedtls: [Remove Kconfig options removed from mbedtls 3.6.0](https://github.com/openwrt/openwrt/commit/adc29202c2044ad039f2904b822c6a6fe84ac984)
These options are not available in mbedtls 3.6.0 and selecting them
causes an error.

The last two upgrades resulted in a lot of dependencies on MBEDTLS_CIPHER_MODE_CFB
The software cannot be compiled normally. 

### OpenWrt version

master

### OpenWrt release

master

### OpenWrt target/subtarget

all

### Device

all

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

suyuan168, 2024-04-30:
The upstream shadowsocks has not been updated.
What is the difference between mbedtls 3.6.0  or
2.28.8 and the latest one? Is it compatible?

peterwillcn, 2024-04-30:
> The upstream shadowsocks has not been updated. What is the difference between mbedtls 3.6.0 or 2.28.8 and the latest one? Is it compatible?

Shadowsocks libev has not been maintained for a long time, so you can switch to Shadowsocks rust without relying on mbedtls to indirectly solve this problem

closed 2024-04-30 by peterwillcn