openwrt/openwrt #15340: procd: env LD_PRELOAD gets always overwritten

ThomasCr, 2024-04-30:
### Describe the bug

Hi,

I try to mod the package `sshtunnel` to use it in a `mwan3` environment. To use the correct interface, we need to LD_PRELOAD with the /lib/mwan3/libwrap_mwan3_sockopt.so.1.0 and additional vars like FAMILY=ipv4 DEVICE=wlan4wan2 ....

So I added a new var into my sshtunnel config:

```
root@gate:~# uci show|grep "sshtunnel.srv1.procdEnv"
sshtunnel.srv1.procdEnv='FAMILY=ipv4 DEVICE=wlan4wan2 FWMARK=0x3f00 LD_PRELOAD=/lib/mwan3/libwrap_mwan3_sockopt.so.1.0'

```

and modded my init script like

```
root@gate:~# diff -Nru /etc/init.d/sshtunnel /etc/init.d/sshtunnel.mod
--- /etc/init.d/sshtunnel       2023-04-15 17:49:30.000000000 +0200
+++ /etc/init.d/sshtunnel.mod   2024-04-30 14:40:07.000000000 +0200
@@ -52,7 +52,8 @@
                'ServerAliveInterval:min(0)' \
                'StrictHostKeyChecking:or("yes", "no", "accept-new")' \
                'TCPKeepAlive:or("yes", "no")' \
-               'VerifyHostKeyDNS:or("yes", "no")'
+               'VerifyHostKeyDNS:or("yes", "no")' \
+               'procdEnv:string(1)'
 }

 validate_tunnelR_section() {
@@ -167,6 +168,7 @@
        local ARGS_options=""
        local ARGS_tunnels=""
        local count=0
+       local env=""

        config_foreach validate_tunnelR_section "tunnelR" load_tunnelR
        config_foreach validate_tunnelL_section "tunnelL" load_tunnelL
@@ -181,6 +183,7 @@
        ARGS="$ARGS_options -o ExitOnForwardFailure=yes -o BatchMode=yes -nN $ARGS_tunnels -p $port $user@$hostname"

        procd_open_instance "$server"
+       [ -n "$procdEnv" ] && procd_set_param env $procdEnv
        procd_set_param command "$PROG" $ARGS
        procd_set_param stdout 1
        procd_set_param stderr 1

```

so my environment of the ssh-tunnel is also set like:

```
root@gate:~# tr '\0' '\n' < /proc/$(ps aux | grep 'ssh' | head -n1 | awk '{print $2}')/environ
HOME=/
TERM=linux
PATH=/usr/sbin:/usr/bin:/sbin:/bin
DEVICE=wlan4wan2
FAMILY=ipv4
FWMARK=0x3f00
LD_PRELOAD=/lib/libsetlbf.so
```
We can see, that every var is added - **only** LD_PRELOAD seems to be overwritten.

By the way, [it should be possible to have multiple LD_PRELOAD files](https://stackoverflow.com/questions/8474363/specifying-multiple-files-with-ld-preload), to what ever set the LD_PREOAD should keep maintaining existing vars.

When I am not wrong, it is somewhere located there:

https://github.com/openwrt/procd/blob/946552a7b598a0b88db6101e864679554ec4f221/service/instance.c#L485

But I am not a C guy, so I don't know how to fix it.

### OpenWrt version

r23497-6637af95aa

### OpenWrt release

23.05.0

### OpenWrt target/subtarget

ipq806x/generic

### Device

TP-Link Archer C2600

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ysc3839, 2024-04-30:
A temporary solution is using `patchelf --add-needed` to add that so to executable file.

ThomasCr, 2024-05-04:
@brada4 then LD_PRELOAD=/lib/libsetlbf.so should also not set?

ThomasCr, 2024-05-04:
with patchelf I can modifier the binary (in my case ssh-client) to always load the /lib/mwan3/libwrap_mwan3_sockopt.so.1.0 ?