openwrt/openwrt #15344: [Bananapi BPi-R3 Mini] The USB bus does not see the device when the board is loaded

Kanaka321, 2024-04-30:
### Describe the bug

When downloading OpenWRT, with the m2 keyB module connected, the modem module is not detected over the usb interface bus, the firmware simply does not see it, I tried to download a clean snapshot, the same problem is also the same, another device on another bus (flash drive) is detected at startup. But m2 keyB can be started if you use hotplug downloads (that is, connect the module while OpenWRT is running)
`root@OpenWrt:~# dmesg | grep usb
[    3.384415] usbcore: registered new interface driver usbfs
[    3.389944] usbcore: registered new interface driver hub
[    3.395275] usbcore: registered new device driver usb
[    3.409801] xhci-mtk 11200000.usb: xHCI Host Controller
[    3.415032] xhci-mtk 11200000.usb: new USB bus registered, assigned bus number 1
[    3.425483] xhci-mtk 11200000.usb: hcc params 0x01403f99 hci version 0x110 quirks 0x0000000000210010
[    3.434633] xhci-mtk 11200000.usb: irq 127, io mem 0x11200000
[    3.440472] xhci-mtk 11200000.usb: xHCI Host Controller
[    3.445684] xhci-mtk 11200000.usb: new USB bus registered, assigned bus number 2
[    3.453069] xhci-mtk 11200000.usb: Host supports USB 3.2 Enhanced SuperSpeed
[    3.468508] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[    3.488802] usbcore: registered new interface driver usb-storage
[    3.929177] usb 1-1: new high-speed USB device number 2 using xhci-mtk
[    4.111389] usb-storage 1-1:1.0: USB Mass Storage device detected
[    4.117678] scsi host0: usb-storage 1-1:1.0
[   11.178889] usbcore: registered new interface driver cdc_acm
[   11.210903] usbcore: registered new interface driver usbserial_generic
[   11.217465] usbserial: USB Serial support registered for generic
[   11.224122] usbcore: registered new interface driver cdc_ether
[   11.230753] usbcore: registered new interface driver cdc_ncm
[   14.714261] usbcore: registered new interface driver option
[   14.719905] usbserial: USB Serial support registered for GSM modem (1-port)
[  170.899263] usb 1-1: reset high-speed USB device number 2 using xhci-mtk
[  889.709318] usb 2-1: new SuperSpeed USB device number 2 using xhci-mtk
[  890.839191] usb 2-1: USB disconnect, device number 2
[  902.899429] usb 2-1: new SuperSpeed USB device number 3 using xhci-mtk
[  902.991895] cdc_ncm 2-1:1.6 eth2: register 'cdc_ncm' at usb-11200000.usb-1, CDC NCM (NO ZLP), 00:00:11:12:13:14
[  903.031956] cdc_ncm 2-1:1.8 eth3: register 'cdc_ncm' at usb-11200000.usb-1, CDC NCM (NO ZLP), 00:00:11:12:13:16
[  903.072188] cdc_ncm 2-1:1.10 eth4: register 'cdc_ncm' at usb-11200000.usb-1, CDC NCM (NO ZLP), 00:00:11:12:13:18
root@OpenWrt:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux 6.1.86 xhci-hcd xHCI Host Controller
Bus 001 Device 002: ID 0951:1666 Kingston DataTraveler 3.0
Bus 002 Device 001: ID 1d6b:0003 Linux 6.1.86 xhci-hcd xHCI Host Controller
Bus 002 Device 003: ID 8087:095a Intel Corp. MODEM + 2 CDC-ACM + 3 CDC-NCM + SS
root@OpenWrt:~#
`

### OpenWrt version

r26117-5c833329ce

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPi-R3 Mini

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

reusu, 2024-05-06:
try those command

echo 443 > /sys/class/gpio/export
echo 431 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio431/direction
echo 0 > /sys/class/gpio/gpio431/value

QooGeek, 2024-05-13:
https://github.com/BPI-SINOVOIP/BPI-R3MINI-OPENWRT-V21.02.3/commit/9bd78779f267a21c04c5bb4d16c32e83aae8d1d3

How can I support the 5g module? I have installed the 5g module now, but it is not displayed.

```
root@OpenWrt:~# echo 443 > /sys/class/gpio/export
ash: write error: Invalid argument

root@OpenWrt:~# echo 431 > /sys/class/gpio/export
ash: write error: Invalid argument

root@OpenWrt:~# echo out > /sys/class/gpio/gpio431/direction
-ash: can't create /sys/class/gpio/gpio431/direction: nonexistent directory

root@OpenWrt:~# echo 0 > /sys/class/gpio/gpio431/value
-ash: can't create /sys/class/gpio/gpio431/value: nonexistent directory

root@Bananapi:~# lsmod | fgrep gpio
gpio_button_hotplug    16384  0
leds_gpio              12288  0

root@Bananapi:~# cat /sys/kernel/debug/gpio
gpiochip0: GPIOs 512-612, parent: platform/1001f000.pinctrl, pinctrl_moore:
 gpio-513 (                    |blue:wlan-1         ) out hi
 gpio-514 (                    |blue:wlan-2         ) out hi
 gpio-519 (                    |reset               ) in  hi IRQ ACTIVE LOW
 gpio-531 (                    |green:status        ) out hi
 gpio-532 (                    |regulator-usb-vbus  ) out lo ACTIVE LOW
 gpio-559 (                    |PHY reset           ) out hi ACTIVE LOW
 gpio-561 (                    |PHY reset           ) out hi ACTIVE LOW

root@Bananapi:~# ll /sys/class/gpio/
drwxr-xr-x    2 root     root             0 Jan  1  1970 ./
drwxr-xr-x   40 root     root             0 Jan  1  1970 ../
--w-------    1 root     root          4096 May 13 11:59 export
lrwxrwxrwx    1 root     root             0 Jan  1  1970 gpiochip512 -> ../../devices/platform/soc/1001f000.pinctrl/gpio/gpiochip512/
--w-------    1 root     root          4096 Jan  1  1970 unexport

root@Bananapi:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux 6.6.30 xhci-hcd xHCI Host Controller
Bus 002 Device 001: ID 1d6b:0003 Linux 6.6.30 xhci-hcd xHCI Host Controller
root@Bananapi:~# lspci
root@Bananapi:~#
```

Kanaka321, 2024-05-14:
the same response to the commands, it also didn't work on previous versions, only on SINOVOIP... of course, I understand that this is a snapshot, but maybe there is an opportunity to fix it?
@QooGeek is your module running on hotplug?

reusu, 2024-05-15:
> [BPI-SINOVOIP/BPI-R3MINI-OPENWRT-V21.02.3@9bd7877](https://github.com/BPI-SINOVOIP/BPI-R3MINI-OPENWRT-V21.02.3/commit/9bd78779f267a21c04c5bb4d16c32e83aae8d1d3)
> 
> [BPI-SINOVOIP/BPI-R3MINI-OPENWRT-V21.02.3@39b77d2](https://github.com/BPI-SINOVOIP/BPI-R3MINI-OPENWRT-V21.02.3/commit/39b77d2c0a0f4fc9a52085c2edb838111c46f7e3)
> 
> How can I support the 5g module? I have installed the 5g module now, but it is not displayed.
> 
> my 5g module fm350gl
> 
> ```
> root@OpenWrt:~# echo 443 > /sys/class/gpio/export
> ash: write error: Invalid argument
> 
> root@OpenWrt:~# echo 431 > /sys/class/gpio/export
> ash: write error: Invalid argument
> 
> root@OpenWrt:~# echo out > /sys/class/gpio/gpio431/direction
> -ash: can't create /sys/class/gpio/gpio431/direction: nonexistent directory
> 
> root@OpenWrt:~# echo 0 > /sys/class/gpio/gpio431/value
> -ash: can't create /sys/class/gpio/gpio431/value: nonexistent directory
> 
> root@Bananapi:~# lsmod | fgrep gpio
> gpio_button_hotplug    16384  0
> leds_gpio              12288  0
> 
> root@Bananapi:~# cat /sys/kernel/debug/gpio
> gpiochip0: GPIOs 512-612, parent: platform/1001f000.pinctrl, pinctrl_moore:
>  gpio-513 (                    |blue:wlan-1         ) out hi
>  gpio-514 (                    |blue:wlan-2         ) out hi
>  gpio-519 (                    |reset               ) in  hi IRQ ACTIVE LOW
>  gpio-531 (                    |green:status        ) out hi
>  gpio-532 (                    |regulator-usb-vbus  ) out lo ACTIVE LOW
>  gpio-559 (                    |PHY reset           ) out hi ACTIVE LOW
>  gpio-561 (                    |PHY reset           ) out hi ACTIVE LOW
> 
> root@Bananapi:~# ll /sys/class/gpio/
> drwxr-xr-x    2 root     root             0 Jan  1  1970 ./
> drwxr-xr-x   40 root     root             0 Jan  1  1970 ../
> --w-------    1 root     root          4096 May 13 11:59 export
> lrwxrwxrwx    1 root     root             0 Jan  1  1970 gpiochip512 -> ../../devices/platform/soc/1001f000.pinctrl/gpio/gpiochip512/
> --w-------    1 root     root          4096 Jan  1  1970 unexport
> 
> root@Bananapi:~# lsusb
> Bus 001 Device 001: ID 1d6b:0002 Linux 6.6.30 xhci-hcd xHCI Host Controller
> Bus 002 Device 001: ID 1d6b:0003 Linux 6.6.30 xhci-hcd xHCI Host Controller
> root@Bananapi:~# lspci
> root@Bananapi:~#
> ```

it works for me with 
OpenWrt SNAPSHOT r25845-7f85104ebb / LuCI Master git-24.089.54606~0ecb5ed / Kernel 6.1.82
i think the upgrade to kernel 6.6 caused some issues.

Kanaka321, 2024-05-15:
@QooGeek I've been surfing the Internet a little bit looking for information about your module, the datasheet does not indicate that it has a USB interface

@reusu please tell me, what kind of device do you have? maybe you just don't have a BPI R3 mini?

I just don't have a puzzle, in this board, only the USB interface is indicated on the m2 B-key connector, and only the PCI interface is indicated in the information for the module. Therefore, they cannot work in any way, my l860gl-16 module has both a USB and a PCI interface, therefore theoretically it should be run on this board via the usb bus... for comparison, I attached a BPI R4 board
[image]
[image]

bam80, 2024-05-15:
> it works for me with
> OpenWrt SNAPSHOT r25845-7f85104ebb / LuCI Master git-24.089.54606~0ecb5ed / Kernel 6.1.82
> i think the upgrade to kernel 6.6 caused some issues.

@reusu 
Could you update to the latest snapshot and check if it regressed or not?

reusu, 2024-05-16:
> > it works for me with
> > OpenWrt SNAPSHOT r25845-7f85104ebb / LuCI Master git-24.089.54606~0ecb5ed / Kernel 6.1.82
> > i think the upgrade to kernel 6.6 caused some issues.
> 
> @reusu Could you update to the latest snapshot and check if it regressed or not?

this command can restore the 5G module of M2, which I obtained from the latest snapshot.
echo 544 > /sys/class/gpio/export

Bananapi BPi-R3 Mini
OpenWrt SNAPSHOT r26308-4341901f05 / LuCI Master 24.120.25050~8f2dfa7 / 6.6.30

bam80, 2024-05-16:
@Kanaka321 did you try this command?

Kanaka321, 2024-05-16:
@bam80 I'll try it tonight, I'll unsubscribe, I tried the previous commands, both on core 6.1 and 6.6, but apparently the gpio was reassigned in 6.6

QooGeek, 2024-05-16:
I replaced the fm350gl module and it works fine. Maybe it was broken before.

```
root@R3mini-Router:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux 6.6.30 xhci-hcd xHCI Host Controller
Bus 002 Device 001: ID 1d6b:0003 Linux 6.6.30 xhci-hcd xHCI Host Controller
Bus 002 Device 002: ID 0e8d:7127 Fibocom Wireless Inc. FM350-GL
```

```
root@R3mini-Router:~# dmesg |grep usb
[    7.410871] usbcore: registered new interface driver usbfs
[    7.416412] usbcore: registered new interface driver hub
[    7.421751] usbcore: registered new device driver usb
[    7.439128] xhci-mtk 11200000.usb: xHCI Host Controller
[    7.444434] xhci-mtk 11200000.usb: new USB bus registered, assigned bus number 1
[    7.454886] xhci-mtk 11200000.usb: hcc params 0x01403f99 hci version 0x110 quirks 0x0000000000200010
[    7.464040] xhci-mtk 11200000.usb: irq 127, io mem 0x11200000
[    7.469866] xhci-mtk 11200000.usb: xHCI Host Controller
[    7.475084] xhci-mtk 11200000.usb: new USB bus registered, assigned bus number 2
[    7.482468] xhci-mtk 11200000.usb: Host supports USB 3.2 Enhanced SuperSpeed
[    7.497896] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[    7.971484] usb 1-2: new full-speed USB device number 2 using xhci-mtk
[    8.131493] usb 1-2: device descriptor read/64, error -71
[    8.401492] usb 1-2: device descriptor read/64, error -71
[    8.671487] usb 1-2: new full-speed USB device number 3 using xhci-mtk
[    8.831521] usb 1-2: device descriptor read/64, error -71
[    9.101492] usb 1-2: device descriptor read/64, error -71
[    9.227026] usb usb1-port2: attempt power cycle
[   11.348092] usbcore: registered new interface driver usbserial_generic
[   11.354694] usbserial: USB Serial support registered for generic
[   11.361552] usbcore: registered new interface driver cdc_ether
[   14.893512] usbcore: registered new interface driver rndis_host
[   14.901559] usbcore: registered new interface driver option
[   14.907164] usbserial: USB Serial support registered for GSM modem (1-port)
[   14.915829] usbcore: registered new interface driver qcserial
[   14.921624] usbserial: USB Serial support registered for Qualcomm USB modem
[   18.051631] usb usb1-port2: unable to enumerate USB device
[   26.281576] usb 2-1: new SuperSpeed USB device number 2 using xhci-mtk
[   26.317546] rndis_host 2-1:1.0 eth2: register 'rndis_host' at usb-11200000.usb-1, RNDIS device, 00:00:11:12:13:14
```

```
root@R3mini-Router:~# cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 2
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 6.06
S:  Manufacturer=Linux 6.6.30 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=11200000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=10000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.10 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 6.06
S:  Manufacturer=Linux 6.6.30 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=11200000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=5000 MxCh= 0
D:  Ver= 3.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS= 9 #Cfgs=  1
P:  Vendor=0e8d ProdID=7127 Rev= 0.01
S:  Manufacturer=Fibocom Wireless Inc.
S:  Product=FM350-GL
C:* #Ifs=10 Cfg#= 1 Atr=a0 MxPwr=896mA
A:  FirstIf#= 0 IfCount= 2 Cls=e0(wlcon) Sub=01 Prot=03
I:* If#= 0 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=ff Driver=rndis_host
E:  Ad=82(I) Atr=03(Int.) MxPS=  64 Ivl=125us
I:* If#= 1 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=rndis_host
E:  Ad=81(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=01(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 2 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=(none)
E:  Ad=83(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 3 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=(none)
E:  Ad=84(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 4 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=(none)
E:  Ad=85(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=04(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 5 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=42 Prot=01 Driver=(none)
E:  Ad=86(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=05(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 6 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=(none)
E:  Ad=87(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=06(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 7 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=(none)
E:  Ad=88(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=07(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 8 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=(none)
E:  Ad=89(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=08(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 9 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=(none)
E:  Ad=8a(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=09(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
```

Kanaka321, 2024-05-16:
@QooGeek congratulations on solving your problem, but my Fibocom L860-GL-16 still doesn't want to work... I don't know what to do anymore

@reusu "echo 544 > /sys/class/gpio/export" It doesn't help

Kanaka321, 2024-05-16:
Fibocom L860-GL-16 (change usb mode)
```
picocom -b 115200 /dev/ttyACM0
at@nvm:cal_usbmode.num=0 (or 7, 10, 11)
at@store_nvm(cal_usbmode)
AT+CFUN=15
ID 2cb7:0007 Fibocom MBIM + 3 CDC-ACM
ID 8087:095a Intel Corp. MODEM + 2 CDC-ACM + 3 CDC-NCM + SS

In both of these modes, the modem displays only when connected to a hotplug, along with starting the board, it does not want to start the modem:(

bam80, 2024-05-24:
Why to close this if the problem persists?

cwxiaos, 2025-03-03:
> I have the same problem with a RM520N-GL running OpenWRT 24.10.0 r28427 Kernel version 6.6.73
> 
> Any ideas what could be the issue?

You can solve this by adding startup commands in LuCI: System > Startup > Local Startup

```
# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.

# Fan trip point set
echo 30000 > /sys/class/thermal/thermal_zone0/trip_point_4_temp
echo 35000 > /sys/class/thermal/thermal_zone0/trip_point_3_temp
echo 47000 > /sys/class/thermal/thermal_zone0/trip_point_2_temp
echo 52000 > /sys/class/thermal/thermal_zone0/trip_point_1_temp
echo 85000 > /sys/class/thermal/thermal_zone0/trip_point_0_temp

# Enable 5G modem usb port
echo 544 > /sys/class/gpio/export

exit 0
```

which works for me

saforever99, 2025-03-28:
I have the same problem, running 24.10. I can see the modem:
root@OpenWrt:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux 6.6.73 xhci-hcd xHCI Host Controller
Bus 002 Device 001: ID 1d6b:0003 Linux 6.6.73 xhci-hcd xHCI Host Controller
Bus 002 Device 002: ID 0e8d:7127 Fibocom Wireless Inc. FM350-GL
This is my /etc/config/network:
config interface 'modem0'
        option proto 'atc'
        option device '/dev/ttyUSB3'
        option auth '0'
        option pdp 'IP'
        option atc_debug '0'
        option apn 'internet.it'
But the interface doesn't start. I tried
echo 544 > /sys/class/gpio/export
But it is unsuccessful.

cwxiaos, 2025-03-28:
> I have the same problem, running 24.10. I can see the modem:
> root@OpenWrt:~# lsusb
> Bus 001 Device 001: ID 1d6b:0002 Linux 6.6.73 xhci-hcd xHCI Host Controller
> Bus 002 Device 001: ID 1d6b:0003 Linux 6.6.73 xhci-hcd xHCI Host Controller
> Bus 002 Device 002: ID 0e8d:7127 Fibocom Wireless Inc. FM350-GL
> This is my /etc/config/network:
> config interface 'modem0'
>         option proto 'atc'
>         option device '/dev/ttyUSB3'
>         option auth '0'
>         option pdp 'IP'
>         option atc_debug '0'
>         option apn 'internet.it'
> But the interface doesn't start. I tried
> echo 544 > /sys/class/gpio/export
> But it is unsuccessful.
> When I restart the interface modem0 I see:
> root@OpenWrt:~# logread -f
> Fri Mar 28 17:59:33 2025 daemon.notice netifd: modem0 (4554): modem0 is disconnected
> Fri Mar 28 17:59:33 2025 daemon.notice netifd: modem0 (4554): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "modem0" } (Permission denied)
> Fri Mar 28 17:59:33 2025 daemon.notice netifd: Interface 'modem0' is now down
> Fri Mar 28 17:59:33 2025 daemon.notice netifd: Interface 'modem0' is setting up now
> Fri Mar 28 17:59:33 2025 daemon.notice netifd: modem0 (4562): Initiate modem with interface eth2
> Fri Mar 28 17:59:34 2025 daemon.warn odhcpd[2221]: No default route present, overriding ra_lifetime to 0!
> Fri Mar 28 17:59:36 2025 daemon.notice netifd: modem0 (4562): SIMcard ready
> Fri Mar 28 17:59:38 2025 daemon.notice netifd: modem0 (4562): Configure modem
> 
> Any suggestion?

cat /sys/kernel/debug/usb/devices

saforever99, 2025-03-29:
root@OpenWrt:~# cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 2
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 6.06
S:  Manufacturer=Linux 6.6.73 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=11200000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=10000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.10 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 6.06
S:  Manufacturer=Linux 6.6.73 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=11200000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=5000 MxCh= 0
D:  Ver= 3.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS= 9 #Cfgs=  1
P:  Vendor=0e8d ProdID=7127 Rev= 0.01
S:  Manufacturer=Fibocom Wireless Inc.
S:  Product=FM350-GL
C:* #Ifs=10 Cfg#= 1 Atr=a0 MxPwr=896mA
A:  FirstIf#= 0 IfCount= 2 Cls=e0(wlcon) Sub=01 Prot=03
I:* If#= 0 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=ff Driver=rndis_host
E:  Ad=82(I) Atr=03(Int.) MxPS=  64 Ivl=125us
I:* If#= 1 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=rndis_host
E:  Ad=81(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=01(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 2 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=option
E:  Ad=83(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 3 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=option
E:  Ad=84(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 4 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=option
E:  Ad=85(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=04(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 5 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=42 Prot=01 Driver=(none)
E:  Ad=86(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=05(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 6 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=option
E:  Ad=87(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=06(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 7 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=option
E:  Ad=88(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=07(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 8 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=option
E:  Ad=89(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=08(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 9 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=option
E:  Ad=8a(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=09(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms

cwxiaos, 2025-03-29:
> root@OpenWrt:~# cat /sys/kernel/debug/usb/devices
>
> T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 2
> B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
> D:  Ver= 2.00 Cls=09(hub  ) [...]

It looks like you are using fm350-gl with rndis, so i think you should add a dhcp client in interfaces instead of using another protocol

saforever99, 2025-03-29:
Can you please share how the network file should look like then ? so I can try it..

cwxiaos, 2025-03-29:
> Can you please share how the network file should look like then ? so I can try it..

```
config interface 'wwan4'
	option proto 'dhcp'
	option device '/dev/cdc-wdm0'

config interface 'wwan6'
	option proto 'dhcpv6'
	option device '/dev/cdc-wdm0'
	option reqaddress 'try'
	option reqprefix 'auto'
	option norelease '1'

```

Just in luci > network > interfaces, add a dhcp client and a dhcpv6 client.

And you can ls /dev to determine which device is exactly the network interface

cwxiaos, 2025-03-29:
> Can you please share how the network file should look like then ? so I can try it..
> 

In fact, you should first ls /dev

, I dont know what a fibcom modem looks like, i only use quectel

saforever99, 2025-03-29:
root@OpenWrt:/tmp# ls -la /dev
drwxr-xr-x    4 root     root          1640 Mar 29 09:00 .
drwxr-xr-x    1 root     root          3488 Feb  3 23:09 ..
drwxr-xr-x    3 root     root            60 Jan  1  1970 bus
crw-------    1 root     root        5,   1 Jan  1  1970 console
lrwxrwxrwx    1 root     root            13 Jan  1  1970 fd -> /proc/self/fd
brw-------    1 root     root      259,   1 Jan  1  1970 fit0
brw-------    1 root     root      259,   2 Jan  1  1970 fitrw
crw-rw-rw-    1 root     root        1,   7 Jan  1  1970 full
crw-------    1 root     root      254,   0 Jan  1  1970 gpiochip0
crw-------    1 root     root       10, 183 Jan  1  1970 hwrng
crw-------    1 root     root       89,   0 Jan  1  1970 i2c-0
crw-------    1 root     root        1,  11 Jan  1  1970 kmsg
srw-rw-rw-    1 root     root             0 Mar 29 08:50 log
crw-------    1 root     root       10, 237 Jan  1  1970 loop-control
brw-------    1 root     root        7,   0 Jan  1  1970 loop0
brw-------    1 root     root        7,   1 Jan  1  1970 loop1
brw-------    1 root     root        7,   2 Jan  1  1970 loop2
brw-------    1 root     root        7,   3 Jan  1  1970 loop3
brw-------    1 root     root        7,   4 Jan  1  1970 loop4
brw-------    1 root     root        7,   5 Jan  1  1970 loop5
brw-------    1 root     root        7,   6 Jan  1  1970 loop6
brw-------    1 root     root        7,   7 Jan  1  1970 loop7
brw-------    1 root     root      179,   0 Jan  1  1970 mmcblk0
brw-------    1 root     root      179,   8 Jan  1  1970 mmcblk0boot0
brw-------    1 root     root      179,  16 Jan  1  1970 mmcblk0boot1
brw-------    1 root     root      179,   1 Jan  1  1970 mmcblk0p1
brw-------    1 root     root      259,   0 Jan  1  1970 mmcblk0p128
brw-------    1 root     root      179,   2 Jan  1  1970 mmcblk0p2
brw-------    1 root     root      179,   3 Jan  1  1970 mmcblk0p3
brw-------    1 root     root      179,   4 Jan  1  1970 mmcblk0p4
brw-------    1 root     root      179,   5 Jan  1  1970 mmcblk0p5
crw-------    1 root     root      249,   0 Jan  1  1970 mmcblk0rpmb
crw-------    1 root     root       90,   0 Jan  1  1970 mtd0
crw-------    1 root     root       90,   1 Jan  1  1970 mtd0ro
crw-------    1 root     root       90,   2 Jan  1  1970 mtd1
crw-------    1 root     root       90,   3 Jan  1  1970 mtd1ro
brw-------    1 root     root       31,   0 Jan  1  1970 mtdblock0
brw-------    1 root     root       31,   1 Jan  1  1970 mtdblock1
crw-rw-rw-    1 root     root        1,   3 Jan  1  1970 null
crw-------    1 root     root        1,   4 Jan  1  1970 port
crw-------    1 root     root      108,   0 Mar 29 08:50 ppp
crw-rw-rw-    1 root     root        5,   2 Mar 29 09:34 ptmx
drwxr-xr-x    2 root     root             0 Jan  1  1970 pts
crw-rw-rw-    1 root     root        1,   8 Jan  1  1970 random
lrwxrwxrwx    1 root     root             8 Jan  1  1970 shm -> /tmp/shm
lrwxrwxrwx    1 root     root            15 Jan  1  1970 stderr -> /proc/self/fd/2
lrwxrwxrwx    1 root     root            15 Jan  1  1970 stdin -> /proc/self/fd/0
lrwxrwxrwx    1 root     root            15 Jan  1  1970 stdout -> /proc/self/fd/1
crw-rw-rw-    1 root     root        5,   0 Mar 29 09:34 tty
crw-rw----    1 root     dialout     4,  64 Jan  1  1970 ttyS0
crw-rw----    1 root     dialout     4,  65 Jan  1  1970 ttyS1
crw-rw----    1 root     dialout     4,  74 Jan  1  1970 ttyS10
crw-rw----    1 root     dialout     4,  75 Jan  1  1970 ttyS11
crw-rw----    1 root     dialout     4,  76 Jan  1  1970 ttyS12
crw-rw----    1 root     dialout     4,  77 Jan  1  1970 ttyS13
crw-rw----    1 root     dialout     4,  78 Jan  1  1970 ttyS14
crw-rw----    1 root     dialout     4,  79 Jan  1  1970 ttyS15
crw-rw----    1 root     dialout     4,  66 Jan  1  1970 ttyS2
crw-rw----    1 root     dialout     4,  67 Jan  1  1970 ttyS3
crw-rw----    1 root     dialout     4,  68 Jan  1  1970 ttyS4
crw-rw----    1 root     dialout     4,  69 Jan  1  1970 ttyS5
crw-rw----    1 root     dialout     4,  70 Jan  1  1970 ttyS6
crw-rw----    1 root     dialout     4,  71 Jan  1  1970 ttyS7
crw-rw----    1 root     dialout     4,  72 Jan  1  1970 ttyS8
crw-rw----    1 root     dialout     4,  73 Jan  1  1970 ttyS9
crw-rw----    1 root     dialout   188,   0 Mar 29 09:01 ttyUSB0
crw-rw----    1 root     dialout   188,   1 Mar 29 09:01 ttyUSB1
crw-rw----    1 root     dialout   188,   2 Mar 29 09:01 ttyUSB2
crw-rw----    1 root     dialout   188,   3 Mar 29 09:01 ttyUSB3
crw-rw----    1 root     dialout   188,   4 Mar 29 09:01 ttyUSB4
crw-rw----    1 root     dialout   188,   5 Mar 29 09:01 ttyUSB5
crw-rw----    1 root     dialout   188,   6 Mar 29 09:01 ttyUSB6
crw-------    1 root     root      250,   0 Jan  1  1970 ubi0
crw-------    1 root     root      250,   1 Jan  1  1970 ubi0_0
crw-------    1 root     root      250,   2 Jan  1  1970 ubi0_1
crw-------    1 root     root      250,   3 Jan  1  1970 ubi0_2
crw-------    1 root     root       10, 127 Jan  1  1970 ubi_ctrl
brw-------    1 root     root      254,   0 Jan  1  1970 ubiblock0_1
crw-rw-rw-    1 root     root        1,   9 Jan  1  1970 urandom
crw-------    1 root     root       10, 130 Jan  1  1970 watchdog
crw-------    1 root     root      252,   0 Jan  1  1970 watchdog0
crw-rw-rw-    1 root     root        1,   5 Jan  1  1970 zero

Indeed I cannot identify the network interface

cwxiaos, 2025-03-29:
> root@OpenWrt:/tmp# ls -la /dev
> drwxr-xr-x    4 root     root          1640 Mar 29 09:00 .
> drwxr-xr-x    1 root     root          3488 Feb  3 23:09 ..
> drwxr-xr-x    3 root     root            60 Jan [...]

You can try add dhcp client in luci, and see the interfaces in luci > network > interfaces > add new interfaces > devices,
Check if there has a interface that looks like a modem interface

cwxiaos, 2025-03-29:
> You can try add dhcp client in luci, and see the interfaces in luci > network > interfaces > add new interfaces > devices,
> Check if there has a interface that looks like a modem interface

And maybe one of ttyUSB* is the wwan interface

saforever99, 2025-03-29:
it is ttyUSB3.
root@OpenWrt:~# sms_tool -D -d /dev/ttyUSB3 at AT+CREG?
AT+CREG?

+CREG: 0,0

OK
The problem is that the interface that use ATC protocol doesn't startup, I have 0 traffic

cwxiaos, 2025-03-29:
Hope this help you. In fact, I dont even know how to use fibocom modem.

https://github.com/drozdi70/bananapi_bpir4/blob/main/fm350gl.sh

KenjiTakahashi, 2025-10-03:
Did anyone find a way to fix the initial issue?
I am facing the same problem with OpenWRT 24.10.3 and RM520N-GL modem.
None of the methods mentioned in the thread make any difference. Only hot plugging the modem forces it to appear, which is obviously not ideal.

saforever99, 2025-10-04:
did you try installing the following packages ?
opkg update
opkg install kmod-usb-net-rndis kmod-usb3 kmod-usb-serial kmod-usb-serial-option comgt
reboot

KenjiTakahashi, 2025-10-06:
Yes, I have these and other relevant packages installed. When the modem is hot plugged and recognised by the OS, it works fine. So it doesn't seem to be a missing driver issue.
It just doesn't appear at all when connected before booting for some reason.