openwrt/openwrt #15347: WPA3 Enterprise 192-bit (WPA-EAP-SUITE-B-192) does not work

ncleis, 2024-04-30:
### Describe the bug

With the hostapd-openssl package installed, WPA3 Enterprise 192-bit (WPA-EAP-SUITE-B-192) does not work

Although it was fixed eap192 in hostapd.sh file in previous issues, but when I set encryption option 'eap192' - the system does not understand this option and the network becomes unencrypted

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

mediatek/filogic

### Device

MERCUSYS MR90X v1

### Image kind

Official downloaded image

### Steps to reproduce

Go to /etc/config/wireless and set encryption option 'eap192'

### Actual behaviour

[image]
The encryption option does not work

### Expected behaviour

The option will work

### Additional info

For example, we took another non-openwrt router from keenetic, on which this mode works as it should. On openwrt instead of option encryption 'eap192', which leads to disabling encryption, option encryption 'wpa3+gcmp256' was selected, which still does not give the expected result
[image]

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-04-30:
Can you extract the encryptions from /var/run/hostapd*conf?

ncleis, 2024-05-01:
> Can you extract the encryptions from /var/run/hostapd*conf?

sae_require_mfp=1
sae_pwe=2

ncleis, 2024-05-01:
but if you change those parameters to:

wpa_pairwise=GCMP-256
wpa_key_mgmt=WPA-EAP-SUITE-B-192
group_mgmt_cipher=BIP-GMAC-256
[image]

and after that make wifi reload, then in the webinterface will be processed wpa3 everything seems to be correct, but if you look with the phone, everything is the same as in the topic header

csharper2005, 2024-05-01:
I can confirm that RSN-EAP_SUITE_B doesn't work on 23.05.3.

And it works on 22.03.6.
[image]
[image]

rany2, 2024-05-01:
I think maybe https://github.com/openwrt/openwrt/commit/472312f83f886a0749672a634948726fda9c2401 should be backported?

csharper2005, 2024-05-01:
It was a bunch of changes with EAP-192...
https://github.com/openwrt/openwrt/commits/openwrt-23.05/package/network/services/hostapd/files/hostapd.sh

csharper2005, 2024-05-01:
wireless:
```
config wifi-iface 'wifinet1'
	option device 'radio0'
	option mode 'ap'
	option ssid 'eap'
	option encryption 'wpa3'
	option auth_server '127.0.0.1'
	option auth_secret 'secret'
```
`hostapd.conf` 22.03.6 (correct):
```
wpa_pairwise=GCMP-256
wpa_key_mgmt=WPA-EAP-SUITE-B-192
group_mgmt_cipher=BIP-GMAC-256
```
`hostapd.conf` 23.05.3 (wrong):
```
wpa_pairwise=CCMP
wpa_key_mgmt=WPA-EAP-SHA256
group_mgmt_cipher=AES-128-CMAC
```

Ansuel, 2024-05-05:
@csharper2005 maybe there is a logic problem in the wifi-scripts migration? hostapd.conf data comes from 2 script hostapd.sh and mac80211.sh

Should be easy to notice a logic error in those scripts.

ncleis, 2024-05-05:
> @csharper2005 maybe there is a logic problem in the wifi-scripts migration? hostapd.conf data comes from 2 script hostapd.sh and mac80211.sh
> 
> Should be easy to notice a logic error in those scripts.

It turned out that there is an option "wpa3-192", which is not in the official openwrt documents.  This option is located in /etc/config/wireless.  And you can only set it through a configuration file; there is no such option in luci.  As a result, this option enables the desired encryption mode

csharper2005, 2024-05-05:
@Ansuel as it turned out, since https://github.com/openwrt/openwrt/commit/8c03dc962f8e10f9fef0877a0f8d8235f619ef7d was merged we have a new encryption `wpa3-192` for 192 bit support. It isn't documented on wiki and still not supported in Luci.

Gunni, 2024-11-19:
Are you using Suite-B approved certs, the entire chain must be:

- Certificates must be ECC using elliptic curve `secp384r1`
- Protocol must be `TLSV1.2`, all others are disabled (in practice, TLS 1.3 seems to work, it just isn’t officially NSA-grade wifi)
- Cipher algorithm must be `AES-256`
- Key exchange algorithm must be `ECDH`
- Digital signature algorithm must be `ECDSA`
- Hashing algorithm must be `SHA384`
- Cipher suites allowed for NSA Suite B 192 bit are `TLS_ECDHE_ECDSA_W_AES_256_CBC_SHA384` and `TLS_ECDHE_ECDSA_W_AES_256_GCM_SHA384`

Source: https://smallstep.com/blog/home-network-eap-tls-wifi/