openwrt/openwrt #15348: ramips: mt7530 not works autodetection 100Mb speed ethernet link on kernel 6.6

AuthorReflex, 2024-05-01:
### Describe the bug

With testing kernel version 6.6.* on devices ramps/mt7621 the link speed auto-negotiation is broken (again). If another device uses 100Mbps speed (two-pairs cable or 100Mbps speed port), the link on any port (WAN or LAN*) speed shows up as 1Gbps full duplex and not functional. 

Setting a proper speed and duplex (ex. ethtool -s wan speed 100 duplex full autoneg off) solves the problem. 

On kernel version 6.1.* everything works fine.

### OpenWrt version

r25874-98f9154316

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

probably all devices with mt7530 (ASUS RT-AC57U v1, Xiaomi Mi Router 4A GE, Xiaomi Redmi Router AC2100, Wifire NBN-s1500)

### Image kind

Self-built image

### Steps to reproduce

Build image for device ramips/mt7621 using tesing kernel vesion (with any version on branch 6.6 autonegotiation does not work)

### Diffconfig

```text
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_ramips_mt7621_DEVICE_asus_rt-ac57u-v1=y
CONFIG_TARGET_DEVICE_ramips_mt7621_DEVICE_wifire_s1500-nbn=y
CONFIG_TARGET_DEVICE_ramips_mt7621_DEVICE_xiaomi_mi-router-3g=y
CONFIG_TARGET_DEVICE_ramips_mt7621_DEVICE_xiaomi_mi-router-4a-gigabit=y
CONFIG_TARGET_DEVICE_ramips_mt7621_DEVICE_xiaomi_redmi-router-ac2100=y
CONFIG_EXPERIMENTAL=y
# CONFIG_JSON_OVERVIEW_IMAGE_INFO is not set
# CONFIG_KERNEL_DEBUG_FS is not set
# CONFIG_KERNEL_DEBUG_INFO is not set
CONFIG_KERNEL_IO_URING=y
# CONFIG_KERNEL_KALLSYMS is not set
CONFIG_KERNEL_PROC_STRIPPED=y
# CONFIG_KERNEL_WERROR is not set
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
CONFIG_LINUX_6_6=y
# CONFIG_LUA_ECO_MBEDTLS is not set
CONFIG_LUA_ECO_OPENSSL=y
CONFIG_LUCI_SRCDIET=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_PSK=y
[... 217 lines cut ...]
CONFIG_PACKAGE_wpad-basic-openssl=y
CONFIG_PACKAGE_xl2tpd=y
CONFIG_PACKAGE_xtables-legacy=y
CONFIG_PACKAGE_zlib=y
# CONFIG_SIGNATURE_CHECK is not set
# CONFIG_SIGNED_PACKAGES is not set
CONFIG_STRONGSWAN_ROUTING_TABLE="220"
CONFIG_STRONGSWAN_ROUTING_TABLE_PRIO="220"
CONFIG_TESTING_KERNEL=y
CONFIG_USE_GC_SECTIONS=y
CONFIG_USE_MKLIBS=y
# CONFIG_WPA_MBO_SUPPORT is not set
# CONFIG_LIBCURL_NGHTTP2 is not set
# CONFIG_LIBCURL_UNIX_SOCKETS is not set
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
# CONFIG_PACKAGE_libnghttp2 is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-05-01:
Paste the output of the command: mii-tool wan -v

AuthorReflex, 2024-05-01:
normal state: 
```
wan: negotiated 1000baseT-FD flow-control, link ok
  product info: vendor 00:e8:a5, model 1 rev 2
  basic mode:   autonegotiation enabled
  basic status: autonegotiation complete, link ok
  capabilities: 1000baseT-FD 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD
  advertising:  1000baseT-FD 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD flow-control
  link partner: 1000baseT-FD 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD
```

two-pair cable:
```
root@OpenWrt:~# mii-tool -v wan
wan: negotiated 100baseTx-FD, link ok
  product info: vendor 00:e8:a5, model 1 rev 2
  basic mode:   autonegotiation enabled
  basic status: autonegotiation complete, link ok
  capabilities: 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD
  advertising:  100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD flow-control
  link partner: 1000baseT-FD 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD
```

namiltd, 2024-05-01:
"two-pair cable" is before or after running "ethtool -s wan speed 100 duplex full autoneg off"?

AuthorReflex, 2024-05-01:
speed 100 duplex full autoneg on
```
wan: negotiated 100baseTx-FD, link ok
  product info: vendor 00:e8:a5, model 1 rev 2
  basic mode:   autonegotiation enabled
  basic status: autonegotiation complete, link ok
  capabilities: 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD
  advertising:  100baseTx-FD flow-control
  link partner: 1000baseT-FD 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD
```
speed 100 duplex full autoneg off
```
wan: 100 Mbit, full duplex, link ok
  product info: vendor 00:e8:a5, model 1 rev 2
  basic mode:   100 Mbit, full duplex
  basic status: link ok
  capabilities: 1000baseT-FD 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD
  advertising:  flow-control
  link partner: 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD
```

AuthorReflex, 2024-05-01:
kernel 6.1 link 100Mbps
```
wan: negotiated 100baseTx-FD, link ok
  product info: vendor 00:e8:a5, model 1 rev 2
  basic mode:   autonegotiation enabled
  basic status: autonegotiation complete, link ok
  capabilities: 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD
  advertising:  100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD flow-control
  link partner: 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD
```

ivanich, 2024-05-20:
Any updates with latest master?

meox, 2024-08-23:
I have the same issue with a Gli.Net Flint2

danpawlik, 2024-10-17:
I have similar issue with Ubiquiti u6 lite and isp router - orange funbox 2.0.
When it's connected, each 5 second it disconnects and connects back, the disconnect. It says that it negotiate 1000Mbps (where isp router had 1gbit port), setting100mbps does not do anything. Only adding a switch between router and ubiquti - then it works.
What is odd here - uap ac lr got similar issue. All started with kernel 6.6. In version 23.05 snapshot it does not happen.

kofec, 2024-10-30:
Hello, the same in my case (today's build - 2024-10-30):
```
root@Xiaomi3G:~# ubus call system board
{
	"kernel": "6.6.58",
	"hostname": "Xiaomi3G",
	"system": "MediaTek MT7621 ver:1 eco:3",
	"model": "Xiaomi Mi Router 3G",
	"board_name": "xiaomi,mi-router-3g",
	"rootfs_type": "squashfs",
	"release": {
		"distribution": "OpenWrt",
		"version": "SNAPSHOT",
		"description": "OpenWrt SNAPSHOT",
		"revision": "r27967-cbefc64cb3",
		"target": "ramips/mt7621",
		"builddate": "1730233504"
	}
}
```
I have to force using 100MB/s : ethtool -s lan2 speed 100 duplex full autoneg off
logs:
```
[  126.933728] br-lan: port 5(phy1-ap1) entered disabled state
[  127.000706] br-lan: port 5(phy1-ap1) entered blocking state
[  127.006413] br-lan: port 5(phy1-ap1) entered disabled state
[  127.012049] mt76x2e 0000:02:00.0 phy1-ap1: entered allmulticast mode
[  127.019002] mt76x2e 0000:02:00.0 phy1-ap1: entered promiscuous mode
[  127.025793] br-lan: port 5(phy1-ap1) entered blocking state
[  127.031440] br-lan: port 5(phy1-ap1) entered forwarding state
[  305.735737] mt7530-mdio mdio-bus:1f lan2: Link is Down
[  305.741143] br-lan: port 2(lan2) entered disabled state
[  322.804794] mt7530-mdio mdio-bus:1f lan2: Link is Up - 1Gbps/Full - flow control rx/tx
[  322.804907] br-lan: port 2(lan2) entered blocking state
[  322.818043] br-lan: port 2(lan2) entered forwarding state
[  324.320104] mt7530-mdio mdio-bus:1f lan2: Link is Down
[  324.325619] br-lan: port 2(lan2) entered disabled state
[  326.726379] mt7530-mdio mdio-bus:1f lan2: Link is Up - 1Gbps/Full - flow control rx/tx
[  326.734424] br-lan: port 2(lan2) entered blocking state
[  326.739737] br-lan: port 2(lan2) entered forwarding state
[  486.011633] mt7530-mdio mdio-bus:1f wan: Link is Down
[  487.424004] mt7530-mdio mdio-bus:1f lan2: Link is Down
[  487.429578] br-lan: port 2(lan2) entered disabled state
[  487.602395] mt7530-mdio mdio-bus:1f wan: Link is Up - 100Mbps/Full - flow control rx/tx
[  501.205859] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[  501.206094] br-lan: port 1(lan1) entered blocking state
[  501.219069] br-lan: port 1(lan1) entered forwarding state
[  887.428660] mt7530-mdio mdio-bus:1f lan1: Link is Down
[  887.433967] br-lan: port 1(lan1) entered disabled state
[  889.799295] mt7530-mdio mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
[  889.804967] br-lan: port 2(lan2) entered blocking state
[  889.812439] br-lan: port 2(lan2) entered forwarding state
[  914.795201] usb 1-1: reset full-speed USB device number 2 using xhci-mtk
```

solipse, 2024-10-31:
This issue was most likely resolved by a patch https://github.com/openwrt/openwrt/commit/73fd9f79ced3e50ddf34e50d5bfaf2dc65767188#diff-67b76ad9608e576e7ff40411cc5e96aa9f58dec8481afa2882bd810131747c38R4
entered version 22.03.0-rc1

AuthorReflex, 2024-10-31:
> This issue was most likely resolved by a patch [73fd9f7#diff-67b76ad9608e576e7ff40411cc5e96aa9f58dec8481afa2882bd810131747c38R4](https://github.com/openwrt/openwrt/commit/73fd9f79ced3e50ddf34e50d5bfaf2dc65767188#diff-67b76ad9608e576e7ff40411cc5e96aa9f58dec8481afa2882bd810131747c38R4) entered version 22.03.0-rc1

this patch already included in kernel-6.6

kofec, 2024-11-26:
I add info that when using:
`ethtool -s eth0 speed 100 duplex full autoneg off`
I had low performance - about 20Mb/s +/- 3Mb/s
so I had to change to 
`ethtool -s eth0 speed 100 duplex full autoneg on`
And now all is fine

ivanich, 2024-11-27:
Is there any update on this? Is anyone actively working on the issue?

ivanich, 2024-12-15:
Have it working on edgerouterx(with mt7530 switch btw) on a 24.10 snapshot build without any ethtool tweaks.
```
root@edgerouterx:~# ethtool eth0
Settings for eth0:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full 
                                             100baseT/Half 100baseT/Full 
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 100Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: external
        Auto-negotiation: on
        MDI-X: Unknown
        Current message level: 0x000000ff (255)
                               drv probe link timer ifdown ifup rx_err tx_err
        Link detected: yes

```

Nuck-TH, 2025-01-22:
Goth this on WG3526 with rc5(both upgraded config and default) after update from 23.05.5. Link is autodetected as 1Gbps/Full - flow control off(sometimes as full rx/tx) and doesn't work. On 23.05 it was 100Mbps/Full - flow control off. Setting it to that with ethtool fixes the problem.
Don't know what switch ISP has, but cable was laid when only 100Mbit/s was available, it is rather long(~20m).

From RC5 fresh reboot:

```
root@OpenWrt:~# ethtool wan
Settings for wan:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 4
        Transceiver: external
        Auto-negotiation: on
        MDI-X: Unknown
        Supports Wake-on: d
        Wake-on: d
        Link detected: yes
root@OpenWrt:~# uname -a
Linux OpenWrt 6.6.69 #0 SMP Sat Jan  4 21:35:37 2025 mips GNU/Linux
```

inoperable, 2025-02-05:
try to disable energy-efficient-ethernet on both both endpoints and it ceased
on my end it was the crappy intel i225v, unreliable piece of s**t with pme/eee isssues, ipv6 instability and random drops - worst nic (but hey, it shuffles up to 2.5gbps in name!)

f****g intel 
  
```ethtool --set-eee **ethX** eee off```

danpawlik, 2025-02-08:
Linking solution for issue:  https://github.com/openwrt/openwrt/issues/17351

Sechorda, 2025-02-12:
Kinda scared to upgrade only for this issue. will it be patched in the next release?

inoperable, 2025-02-13:
> Kinda scared to upgrade only for this issue. will it be patched in the next release?

scared? why? what can possible go wrong?

AuthorReflex, 2025-04-25:
fixed in https://github.com/openwrt/openwrt/pull/18585
```
 mii-tool -v lan1
lan1: negotiated 100baseTx-FD, link ok
  product info: vendor 00:e8:a5, model 1 rev 2
  basic mode:   autonegotiation enabled
  basic status: autonegotiation complete, link ok
  capabilities: 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD
  advertising:  100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD flow-control
  link partner: 100baseTx-FD 100baseTx-HD 10baseT-FD 10baseT-HD
```

tazz4843, 2026-01-15:
Still seems to be an issue on a brand new Netgear R6350 using a MT7621, OpenWRT 24.10.5