openwrt/openwrt #15351: asus ax53u: weakly amplifies the signal in repeater mode

saber716rus, 2024-05-01:
### Describe the bug

I found that the router is weakly amplifying the signal

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

I raised 2 frequencies at the same time as a repeater. I did it according to this instruction, only for each frequency I created 2 wwan interfaces and a repeater bridge. And each wwan assigned its own device.
https://openwrt.org/docs/guide-user/network/wifi/relay_configuration

### Actual behaviour

The signal is amplified somewhere only up to 60 dbm

### Expected behaviour

signal gain up to at least 30dbm

### Additional info

but the Internet is stable.

### Diffconfig

```text
-ash: ./scripts/diffconfig.sh: not found
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-05-01:
Please attach `/etc/config/wireless` and is it 2 ax53u or 2 different devices?

brada4, 2024-05-01:
Note that it is not analog amplifier, it is protocol repeater.

saber716rus, 2024-05-01:
> Please attach `/etc/config/wireless` and is it 2 ax53u or 2 different devices?

No, it's a single device. In general, the network is built like this. The main router is keenetic sprinter kn-3710. he is in the same room. The asus ax53u is in another room, and it is configured as a repeater.

`root@OpenWrt:~# cat /etc/config/wireless 

config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option channel 'auto'
        option band '2g'
        option htmode 'HE40'
        option cell_density '0'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
        option channel 'auto'
        option band '5g'
        option htmode 'HE80'
        option cell_density '0'

config wifi-iface 'wifinet2'
        option device 'radio0'
        option mode 'sta'
        option network 'wwan'
        option ssid 'SecNET'
        option encryption 'sae-mixed'
        option key 'KKP2nFERStvBEdT'

config wifi-iface 'wifinet3'
        option device 'radio1'
        option mode 'sta'
        option network 'wwan1'
        option ssid 'SecNET'
        option encryption 'sae-mixed'
        option key 'KKP2nFERStvBEdT'

root@OpenWrt:~# 
`
once there was such a failure of the 5 GHz network. This is from the keenetic sprinter kn-3710 log.

`[W] May  1 18:22:16 ndm: Hotspot::Discovery::Explorer: "Bridge0": network conflict: hosts fc:34:97:0a:18:ac and fc:34:97:0a:18:a8 have the same IPv4 address 192.168.1.101. 
[W] May  1 18:22:17 ndm: Hotspot::Discovery::Explorer: "Bridge0": network conflict: hosts fc:34:97:0a:18:a8 and fc:34:97:0a:18:ac have the same IPv4 address 192.168.1.101. 
[W] May  1 18:22:17 ndm: Hotspot::Discovery::Explorer: "Bridge0": network conflict: hosts fc:34:97:0a:18:ac and fc:34:97:0a:18:a8 have the same IPv4 address 192.168.1.101. 
[W] May  1 18:22:18 ndm: Hotspot::Discovery::Explorer: "Bridge0": network conflict: hosts fc:34:97:0a:18:a8 and fc:34:97:0a:18:ac have the same IPv4 address 192.168.1.101. 
[W] May  1 18:22:18 ndm: Hotspot::Discovery::Explorer: "Bridge0": network conflict: hosts fc:34:97:0a:18:ac and fc:34:97:0a:18:a8 have the same IPv4 address 192.168.1.101. 
[W] May  1 18:22:19 ndm: Hotspot::Discovery::Explorer: "Bridge0": network conflict: hosts fc:34:97:0a:18:a8 and fc:34:97:0a:18:ac have the same IPv4 address 192.168.1.101. 
[W] May  1 18:22:20 ndm: Hotspot::Discovery::Explorer: "Bridge0": network conflict: hosts fc:34:97:0a:18:ac and fc:34:97:0a:18:a8 have the same IPv4 address 192.168.1.101. 
[W] May  1 18:22:21 ndm: Hotspot::Discovery::Explorer: "Bridge0": network conflict: hosts fc:34:97:0a:18:a8 and fc:34:97:0a:18:ac have the same IPv4 address 192.168.1.101. 
[W] May  1 18:22:21 ndm: Hotspot::Discovery::Explorer: "Bridge0": network conflict: hosts fc:34:97:0a:18:ac and fc:34:97:0a:18:a8 have the same IPv4 address 192.168.1.101. `

saber716rus, 2024-05-01:
In general, the network is built like this. The main router is keenetic sprinter kn-3710. he is in the same room. The asus x53u is in another room and it is configured as a repeater.
[image]
[image]
[image]
[image]
[image]

saber716rus, 2024-05-01:
before that, I tried asus ax53u with tp-link ax53, then asus gave a good signal. but with kinetics, for some reason asus gives out a weaker signal. although it should give out more signal.

brada4, 2024-05-01:
Please format code blocks between triple backticks, remove passwords and change IRL passwords.

brada4, 2024-05-01:
You need to set regulatory domain country code to get any power out of radio
```
option country 'ZZ'
```

saber716rus, 2024-05-01:
> Please format code blocks between triple backticks, remove passwords and change IRL passwords.

> You need to set regulatory domain country code to get any power out of radio
> 
> ```
> option country 'ZZ'
> ```

but as for the country option, "ZZ" is missing. the most recent one is "ZW"

rany2, 2024-05-01:
@saber716rus "ZZ" is just a placeholder for your country code.

saber716rus, 2024-05-01:
> @saber716rus "ZZ" - это просто заполнитель для кода вашей страны.

I'm sorry, I misunderstood you.

saber716rus, 2024-05-04:
I set the country to "RU", but the signal did not increase. Although I had to get up.

brada4, 2024-05-04:
Indeed, you can only (auto-select) less used channel. 
https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt?id=HEAD#n1547

saber716rus, 2024-05-04:
> Indeed, you can only (auto-select) less used channel. 
> https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt?id=HEAD#n1547

judging by the configuration, the signal should be stronger than it actually is. and judging by the indicators next to the repeater router, it does not exceed 60dbm.

brada4, 2024-05-04:
?60dbm 1kW? does it microwave a chicken nearby too?

i get over -30dBm 50cm from AX54 (same construction, just 1 more gige port) in EU 
Check iwinfo and power when clicking 'edit' on any accesspoint of radio. Should be 20dBm.
then `iw list` shows exact radio intersected with regulatory rule and power in each channel.
PS used this at all defaults https://github.com/VREMSoftwareDevelopment/WiFiAnalyzer and samsung, both jump over scale.

saber716rus, 2024-05-06:
> ?60dbm 1kW? does it microwave a chicken nearby too?
>
> i get over -30dBm 50cm from AX54 (same construction, just 1 more gige port) in EU
> Check iwinfo and power when clicking 'edit' on any accesspoint [...]

I'll watch the signal level.

brada4, 2024-05-06:
There is 0.001pct chsmce to have received a defective or damaged device, but then radio problems must show with oem software alike.

brada4, 2024-05-07:
Obviously keenetic oem software does not support your relayd. Only interop you can get is wired dumb-ap.

saber716rus, 2024-05-30:
> Obviously keenetic oem software does not support your relayd. Only interop you can get is wired dumb-ap.

they wrote that it is supported by the repeater method via relayd.

closed 2024-05-30 by saber716rus