openwrt/openwrt #15357: Neufbox6 fails to install openwrt 23

neutrogena44, 2024-05-01:
### Describe the bug

Hello

I tried to install openwrt 23, but it fails.
Image used : 
openwrt-23.05.3-bcm63xx-smp-sfr_neufbox-6-sercomm-r0-squashfs-cfe.bin

```
CPUI                                                                            
L1CI                                                                            
HELO                                                                            
CPUI                                                                            
L1CI                                                                            
DRAM                                                                            
----                                                                            
PHYS                                                                            
STRF                                                                            
400H                                                                            
PHYE                                                                            
DDR2                                                                            
SIZ4                                                                            
SIZ3                                                                            
SIZ2                                                                            
DINT                                                                            
USYN                                                                            
LSYN                                                                            
MFAS                                                                            
LMBE                                                                            
RACE                                                                            
PASS                                                                            
----                                                                            
ZBSS                                                                            
CODE                                                                            
DATA                                                                            
L12F                                                                            
MAIN                                                                            
NAN0                                                                            
BT01                                                                            
0001                                                                            
NAN9                                                                            
BT01                                                                            
0992                                                                            
NAN9                                                                            
NAN3                                                                            
RFS2                                                                            
NAN5                                                                            
                                                                                
                                                                                
CFE version 1.0.38-112.70 for BCM963268 (32bit,SP,BE)                           
Build Date: Fri Oct 18 19:11:22 CEST 2013 (gmu@gmu-thinkpad)                    
Copyright (C) 2000-2011 Broadcom Corporation.                                   
                                                                                
boot...                                                                         
NAND flash device: name ST NAND512W3A2CN6, id 0x2076 block 16KB size 65536KB    
RTL: chip 6000.1000 probed                                                      
TX/RX delay set on switch side!                                                 
Chip ID: BCM63168D0, MIPS: 400MHz, DDR: 400MHz, Bus: 200MHz                     
Main Thread: TP0                                                                
Memory Test Passed                                                              
Total Memory: 134217728 bytes (128MB)                                           
Boot Address: 0xb8000000                                                        
                                                                                
Board IP address                  : 192.168.22.22:ffffff00                      
Host IP address                   : 192.168.22.68                               
Gateway IP address                :                                             
Run from flash/host (f/h)         : f                                           
Default host run file name        : vmlinux                                     
Default host flash file name      : bcm963xx_fs_kernel                          
Boot delay (0-9 seconds)          : 1                                           
[... 324 lines cut ...]
Main Thread Number [0|1]          : 0                                           
                                                                                
*** Press any key to stop auto run (1 seconds) ***                              
Auto run second count down: 1                                                   
CFE> r openwrt.bin                                                              
Loading 192.168.22.68:openwrt.bin ...                                           
Finished loading 5505028 bytes                                                  
Try upgrade firmware....                                                        
WFI TAG :                                                                       
        wfiVersion : 0x0                                                        
        wfiChipId : 0x0                                                         
        wfiFlashType : 0x0                                                      
        wfiCrc : 0x00000000                                                     
        wfiReserved : -559038242                                                
Illegal whole flash image                                                       
Retry loading it as a compressed image.                                         
Code Address: 0x36000000, Entry Address: 0x42726f61                             
LZMA: Prossible old LZMA format, trying to decompress..  
```
and then, nothing happens : if I reboot the device, it's immediately going back to manufacturer's one.
Any help? thank you

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

KanjiMonster, 2024-05-02:
>     Board Id (0-13)                   : NB6V-SER-r0
>                                         ^^^^

You have a Neufbox 6**V**, not a Neufbox 6. These are two different machines internally, and the images are incompatible for each other. Neufbox 6V is currently unsupported in OpenWrt.

closed 2024-05-02 by KanjiMonster

neutrogena44, 2024-05-02:
> > ```
> > Board Id (0-13)                   : NB6V-SER-r0
> >                                     ^^^^
> > ```
> 
> You have a Neufbox 6**V**, not a Neufbox 6. These are two different machines internally, and the images are incompatible for each other. Neufbox 6V is currently unsupported in OpenWrt.

thank you for this answer, the difference is not well easy to find. Well, how to know about the port of openwrt on that device? is there somebody who manages it? thank you!