openwrt/openwrt #15359: Raspberry pi 4 crash and reboot when get internet connectivity

leobsky, 2024-05-02:
### Describe the bug

Device crash as soon as it get internet connectivity, the last known working snapshot is  45a8e962a591dfcac252b0de6324319abd080469 so probably the bug is caused by 7116d2f2b093559516a59a8c5037d3580f04534c
```
[   49.512665] ------------[ cut here ]------------
[   49.517315] kernel BUG at include/linux/skbuff.h:2625!
[   49.522459] Internal error: Oops - BUG: 00000000f2000800 [#1] SMP
[   49.528557] Modules linked in: brcmfmac_wcc pppoe ppp_async nft_fib_inet nf_flow_table_inet brcmfmac pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack cfg80211 cdc_ncm cdc_ether usbnet usbhid slhc r8169 r8152 nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c lan78xx hid_generic crc_ccitt compat brcmutil snd_bcm2835(C) hid evdev snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore nls_utf8 nls_iso8859_1 nls_cp437 vfat fat realtek microchip
[   49.596162] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C         6.1.89 #0
[   49.603566] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[   49.609399] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   49.616366] pc : skb_pull+0x48/0x50
[   49.619869] lr : skb_gro_receive_list+0x54/0xec
[   49.624401] sp : ffffffc008003c40
[   49.627713] x29: ffffffc008003c40 x28: ffffff804105b948 x27: 0000000000000020
[   49.634861] x26: 0000000000000000 x25: 00000000000005a8 x24: 00000000000005a8
[   49.642007] x23: ffffff8043879128 x22: 00000000f5011080 x21: ffffff8043879e28
[   49.649153] x20: ffffff8043879e00 x19: ffffff8043879100 x18: 0000000000000000
[   49.656298] x17: ffffffc0770b7000 x16: ffffffc008000000 x15: 0000000000000000
[   49.663443] x14: 0000000000000002 x13: ab0f31f96ac4f288 x12: 0000000000000006
[   49.670588] x11: 0000000000000040 x10: 0000000000000000 x9 : 0000000000000000
[   49.677734] x8 : 00000000ffff76ff x7 : 0000000000000020 x6 : 000000000000003f
[   49.684879] x5 : 0000000000000001 x4 : 00000000f5011080 x3 : 00000000000005dc
[   49.692023] x2 : ffffff8043879100 x1 : 0000000000000034 x0 : 00000000000005a8
[   49.699168] Call trace:
[   49.701612]  skb_pull+0x48/0x50
[   49.704758]  tcp_gro_receive+0x2b0/0x2dc
[   49.708684]  tcp4_gro_receive+0x11c/0x300
[   49.712695]  inet_gro_receive+0x234/0x2e4
[   49.716706]  dev_gro_receive+0x3e8/0x6e0
[   49.720629]  napi_gro_frags+0x120/0x264
[   49.724464]  rtl8152_get_version+0xedd4/0x11618 [r8152]
[   49.729707]  __napi_poll+0x38/0x174
[   49.733204]  net_rx_action+0x15c/0x31c
[   49.736956]  _stext+0x100/0x274
[   49.740098]  ____do_softirq+0x10/0x20
[   49.743765]  call_on_irq_stack+0x24/0x58
[   49.747689]  do_softirq_own_stack+0x1c/0x30
[   49.751874]  __irq_exit_rcu+0xa4/0xd0
[   49.755542]  irq_exit_rcu+0x10/0x1c
[   49.759032]  el1_interrupt+0x38/0x5c
[   49.762613]  el1h_64_irq_handler+0x18/0x2c
[   49.766714]  el1h_64_irq+0x68/0x6c
[   49.770116]  arch_cpu_idle+0x18/0x2c
[   49.773696]  default_idle_call+0x2c/0x70
[   49.777622]  do_idle+0x214/0x270
[   49.780852]  cpu_startup_entry+0x34/0x40
[   49.784774]  kernel_init+0x0/0x130
[   49.788176]  arch_post_acpi_subsys_init+0x0/0x18
[   49.792799]  start_kernel+0x55c/0x580
[   49.796466]  __primary_switched+0xb4/0xbc
[   49.800483] Code: d65f03c0 d2800000 d50323bf d65f03c0 (d4210000)
[   49.806578] ---[ end trace 0000000000000000 ]---
[   49.811196] Kernel panic - not syncing: Oops - BUG: Fatal exception in interrupt
[   49.818593] SMP: stopping secondary CPUs
[   49.822517] Kernel Offset: disabled
[   49.826001] CPU features: 0x40000,00034080,0000400b
[   49.830881] Memory Limit: none
[   49.833935] Rebooting in 3 seconds..
```

### OpenWrt version

r26144-219018185e

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.1

### Image kind

Self-built image

### Diffconfig

```text
CONFIG_TARGET_bcm27xx=y
CONFIG_TARGET_bcm27xx_bcm2711=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_bcm27xx_bcm2711_DEVICE_rpi-4=y
CONFIG_TARGET_DEVICE_PACKAGES_bcm27xx_bcm2711_DEVICE_rpi-4=""
CONFIG_TARGET_PER_DEVICE_ROOTFS=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_kmod-crypto-sha256=y
CONFIG_PACKAGE_kmod-mii=y
CONFIG_PACKAGE_kmod-usb-net=y
CONFIG_PACKAGE_kmod-usb-net-cdc-ether=y
CONFIG_PACKAGE_kmod-usb-net-cdc-ncm=y
CONFIG_PACKAGE_kmod-usb-net-rtl8152=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libiwinfo=y
CONFIG_PACKAGE_libiwinfo-data=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libuv=y
CONFIG_PACKAGE_libwebsockets-full=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-filebrowser=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-app-ttyd=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_r8152-firmware=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ttyd=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_zlib=y
CONFIG_TARGET_ALL_PROFILES=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

regae, 2024-05-02:
try to revert these commits
7116d2f2b093559516a59a8c5037d3580f04534c
45a8e962a591dfcac252b0de6324319abd080469
b5c53848c3792355a2dd4ff0e06bf4d230e1f5fb

leobsky, 2024-05-02:
> try to revert these commits
> [7116d2f](https://github.com/openwrt/openwrt/commit/7116d2f2b093559516a59a8c5037d3580f04534c)
> [45a8e96](https://github.com/openwrt/openwrt/commit/45a8e962a591dfcac252b0de6324319abd080469)
> [b5c5384](https://github.com/openwrt/openwrt/commit/b5c53848c3792355a2dd4ff0e06bf4d230e1f5fb)
`git revert 7116d2f2b093559516a59a8c5037d3580f04534c` fixed for self build images.

nbd168, 2024-05-03:
Does the error go away if you add this kernel change? https://nbd.name/p/9cca9e2e
Thanks

leobsky, 2024-05-03:
> Does the error go away if you add this kernel change? https://nbd.name/p/9cca9e2e

@nbd168 yes applying the the patch fixed the crash.

closed 2024-05-03 by openwrt-bot

nbd168, 2024-05-03:
Thanks for testing. Fix pushed