openwrt/openwrt #15378: asus ax53u: the host name is not displayed

saber716rus, 2024-05-04:
### Describe the bug

The hostname on one of the networks connected to the same router is not displayed. The router is configured in dual-band repeater mode.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

I set it up using this wiki https://openwrt.org/docs/guide-user/network/wifi/relay_configuration
Only I have configured for two ranges. I.e., for each network, I have set my own wwan interface and a repeater bridge

### Actual behaviour

one network shows the host "192.168.1.1" and the second network "?"

### Expected behaviour

both networks will show the host "192.168.1.1"

### Additional info

root@OpenWrt:~#cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option channel 'auto'
        option band '2g'
        option htmode 'HE40'
        option cell_density '0'
        option country 'RU'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
        option channel 'auto'
        option band '5g'
        option htmode 'HE80'
        option cell_density '0'
        option country 'RU'

config wifi-iface 'wifinet0'
        option device 'radio0'
        option mode 'sta'
        option network 'wwan'
        option ssid 'SecNET'
        option encryption 'sae-mixed'
        option key 

config wifi-iface 'wifinet1'
        option device 'radio1'
        option mode 'sta'
        option network 'wwan1'
        option ssid 'SecNET'
        option encryption 'sae-mixed'
        option key 

[image]

### Diffconfig

```text
root@OpenWrt:~# ./scripts/diffconfig.sh
-ash: ./scripts/diffconfig.sh: not found
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

saber716rus, 2024-05-04:
[image]

saber716rus, 2024-05-04:
[image]

brada4, 2024-05-06:
repeater does not capture dhcp-derived dns names in path, you have MAC address and wifi connection here, and MAC address and DHCP lease + dns record in other router.

jow-, 2024-05-06:
Not a bug.

closed 2024-05-06 by jow-