openwrt/openwrt #15380: x86_64: 6.6 kernel: Intel i915 graphics card driver loading error

yunxi993, 2024-05-04:
### Describe the bug

[    3.043697] AMD-Vi: AMD IOMMUv2 functionality not available on this system - This is not a bug.
[    3.052434] ACPI: bus type drm_connector registered
[    3.058117] i915 0000:00:02.0: [drm] Transparent Hugepage support is recommended for optimal performance on this platform!
[    3.069932] i915 0000:00:02.0: Direct firmware load for i915/adlp_dmc.bin failed with error -2
[    3.078605] i915 0000:00:02.0: Falling back to sysfs fallback for: i915/adlp_dmc.bin
[   65.976493] i915 0000:00:02.0: [drm] *ERROR* GT0: GuC firmware i915/tgl_guc_70.bin: fetch failed -ETIMEDOUT
[   65.976508] i915 0000:00:02.0: [drm] Failed to load DMC firmware i915/adlp_dmc.bin. Disabling runtime power management.
[   65.986280] i915 0000:00:02.0: [drm] GT0: GuC firmware(s) can be downloaded from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
[   65.997090] i915 0000:00:02.0: [drm] DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
[   66.012819] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 0.0.0
[   66.033246] i915 0000:00:02.0: [drm] *ERROR* GT0: GuC initialization failed -ENOEXEC
[   66.041024] i915 0000:00:02.0: [drm] *ERROR* GT0: Enabling uc failed (-5)
[   66.047847] i915 0000:00:02.0: [drm] *ERROR* GT0: Failed to initialize GPU, declaring it wedged!
[   66.056992] i915 0000:00:02.0: [drm:add_taint_for_CI] CI tainted:0x9 by intel_gt_set_wedged_on_init+0x2f/0x40
[   66.207280] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[   66.215173] ACPI: video: Video Device [GFX0] (multi-head: yes  rom: no  post: no)
[   66.223044] input: Video Bus as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input3
[   66.233186] loop: module loaded
[   66.236386] Guest personality initialized and is inactive
[   66.241823] VMCI host device registered (name=vmci, major=10, minor=127)
[   66.248574] Initialized host personality
[   66.252592] VMware PVSCSI driver - version 1.0.7.0-k
[   66.256089] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[   66.257601] hv_vmbus: registering driver hv_storvsc
[   66.269349] ahci 0000:00:17.0: version 3.0
[   66.269418] nvme nvme0: pci function 0000:05:00.0
[   66.269628] ahci 0000:00:17.0: AHCI 0001.0301 32 slots 1 ports 6 Gbps 0x2 impl SATA mode
[   66.282330] ahci 0000:00:17.0: flags: 64bit ncq sntf pm clo only pio slum part deso sadm sds 
[   66.285556] nvme nvme0: 4/0/0 default/read/poll queues
[   66.286988] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[   66.291088] scsi host0: ahci
[   66.298319]  nvme0n1: p1 p2 p3 p128
[   66.302715] scsi host1: ahci
[   66.311722] ata1: DUMMY
[   66.314215] ata2: SATA max UDMA/133 abar m2048@0x81102000 port 0x81102180 irq 128
[   66.317947] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes

### OpenWrt version

r26171-3f28c422ba

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

x86

### Image kind

Self-built image

### Additional info

23.05 branch does not have this error.CONFIG_TARGET_x86=y

### Diffconfig

```text
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_GNUTLS_ALPN=y
[... 269 lines cut ...]
# CONFIG_PACKAGE_libnetfilter-conntrack is not set
# CONFIG_PACKAGE_libnfnetlink is not set
# CONFIG_PACKAGE_libpcre2 is not set
# CONFIG_PACKAGE_libsodium is not set
# CONFIG_PACKAGE_libudns is not set
# CONFIG_PACKAGE_luci-app-passwall2_INCLUDE_Haproxy is not set
# CONFIG_PACKAGE_luci-app-passwall2_INCLUDE_ShadowsocksR_Libev_Client is not set
# CONFIG_PACKAGE_luci-app-passwall2_INCLUDE_Shadowsocks_Libev_Client is not set
# CONFIG_PACKAGE_luci-app-passwall2_INCLUDE_Simple_Obfs is not set
# CONFIG_PACKAGE_luci-app-passwall2_INCLUDE_V2ray_Plugin is not set
# CONFIG_PACKAGE_shadowsocks-libev-config is not set
# CONFIG_PACKAGE_shadowsocks-libev-ss-local is not set
# CONFIG_PACKAGE_shadowsocks-libev-ss-redir is not set
# CONFIG_PACKAGE_shadowsocksr-libev-ssr-local is not set
# CONFIG_PACKAGE_shadowsocksr-libev-ssr-redir is not set
# CONFIG_PACKAGE_simple-obfs is not set
# CONFIG_PACKAGE_v2ray-plugin is not set
# CONFIG_SING_BOX_WITH_DHCP is not set
# CONFIG_SING_BOX_WITH_GVISOR is not set
# CONFIG_SING_BOX_WITH_WIREGUARD is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

schm0, 2024-05-05:
Did you include the firmware ?
You have to it manually in the kernel config I think...
There was a PR a few months ago to fix some of this but was never merged.

namiltd, 2024-05-05:
> There was a PR a few months ago to fix some of this but was never merged.

Which one?

yunxi993, 2024-05-05:
> Did you include the firmware ? You have to it manually in the kernel config I think... There was a PR a few months ago to fix some of this but was never merged.

No, there doesn't seem to be any modules related to Intel i915.
```
# Firmware
#

#
# ath10k Board-Specific Overrides
#
# end of ath10k Board-Specific Overrides

# CONFIG_PACKAGE_aircard-pcmcia-firmware is not set
# CONFIG_PACKAGE_amd64-microcode is not set
# CONFIG_PACKAGE_amdgpu-firmware is not set
# CONFIG_PACKAGE_ar3k-firmware is not set
# CONFIG_PACKAGE_ath10k-board-qca4019 is not set
# CONFIG_PACKAGE_ath10k-board-qca9377 is not set
# CONFIG_PACKAGE_ath10k-board-qca9887 is not set
# CONFIG_PACKAGE_ath10k-board-qca9888 is not set
# CONFIG_PACKAGE_ath10k-board-qca988x is not set
# CONFIG_PACKAGE_ath10k-board-qca9984 is not set
# CONFIG_PACKAGE_ath10k-board-qca99x0 is not set
# CONFIG_PACKAGE_ath10k-firmware-qca4019 is not set
# CONFIG_PACKAGE_ath10k-firmware-qca4019-ct is not set
# CONFIG_PACKAGE_ath10k-firmware-qca4019-ct-full-htt is not set
# CONFIG_PACKAGE_ath10k-firmware-qca4019-ct-htt is not set
# CONFIG_PACKAGE_ath10k-firmware-qca6174 is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9377 is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9887 is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9887-ct is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9887-ct-full-htt is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9888 is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9888-ct is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9888-ct-full-htt is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9888-ct-htt is not set
# CONFIG_PACKAGE_ath10k-firmware-qca988x is not set
# CONFIG_PACKAGE_ath10k-firmware-qca988x-ct is not set
# CONFIG_PACKAGE_ath10k-firmware-qca988x-ct-full-htt is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9984 is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9984-ct is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9984-ct-full-htt is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9984-ct-htt is not set
# CONFIG_PACKAGE_ath10k-firmware-qca99x0 is not set
# CONFIG_PACKAGE_ath10k-firmware-qca99x0-ct is not set
# CONFIG_PACKAGE_ath10k-firmware-qca99x0-ct-full-htt is not set
# CONFIG_PACKAGE_ath10k-firmware-qca99x0-ct-htt is not set
# CONFIG_PACKAGE_ath11k-firmware-ipq8074 is not set
# CONFIG_PACKAGE_ath11k-firmware-qca6390 is not set
# CONFIG_PACKAGE_ath11k-firmware-qcn9074 is not set
# CONFIG_PACKAGE_ath11k-firmware-wcn6750 is not set
# CONFIG_PACKAGE_ath11k-firmware-wcn6855 is not set
# CONFIG_PACKAGE_ath6k-firmware is not set
# CONFIG_PACKAGE_ath9k-htc-firmware is not set
# CONFIG_PACKAGE_bnx2-firmware is not set
# CONFIG_PACKAGE_bnx2x-firmware is not set
# CONFIG_PACKAGE_brcmfmac-firmware-4329-sdio is not set
# CONFIG_PACKAGE_brcmfmac-firmware-4339-sdio is not set
# CONFIG_PACKAGE_brcmfmac-firmware-43430a0-sdio is not set
# CONFIG_PACKAGE_brcmfmac-firmware-43602a1-pcie is not set
# CONFIG_PACKAGE_brcmfmac-firmware-4366b1-pcie is not set
# CONFIG_PACKAGE_brcmfmac-firmware-4366c0-pcie is not set
# CONFIG_PACKAGE_brcmfmac-firmware-usb is not set
# CONFIG_PACKAGE_brcmfmac-nvram-43430-sdio is not set
# CONFIG_PACKAGE_brcmfmac-nvram-43455-sdio is not set
# CONFIG_PACKAGE_brcmsmac-firmware is not set
# CONFIG_PACKAGE_broadcom-4306-sprom is not set
# CONFIG_PACKAGE_broadcom-43112-sprom is not set
# CONFIG_PACKAGE_broadcom-4313-sprom is not set
# CONFIG_PACKAGE_broadcom-43131-sprom is not set
[... 86 lines cut ...]
# CONFIG_PACKAGE_rt2800-usb-firmware is not set
# CONFIG_PACKAGE_rt61-pci-firmware is not set
# CONFIG_PACKAGE_rt73-usb-firmware is not set
# CONFIG_PACKAGE_rtl8188eu-firmware is not set
# CONFIG_PACKAGE_rtl8192ce-firmware is not set
# CONFIG_PACKAGE_rtl8192cu-firmware is not set
# CONFIG_PACKAGE_rtl8192de-firmware is not set
# CONFIG_PACKAGE_rtl8192eu-firmware is not set
# CONFIG_PACKAGE_rtl8192se-firmware is not set
# CONFIG_PACKAGE_rtl8723au-firmware is not set
# CONFIG_PACKAGE_rtl8723bu-firmware is not set
# CONFIG_PACKAGE_rtl8761a-firmware is not set
# CONFIG_PACKAGE_rtl8761b-firmware is not set
# CONFIG_PACKAGE_rtl8761bu-firmware is not set
# CONFIG_PACKAGE_rtl8821ae-firmware is not set
# CONFIG_PACKAGE_rtl8822be-firmware is not set
# CONFIG_PACKAGE_rtl8822ce-firmware is not set
# CONFIG_PACKAGE_ti-3410-firmware is not set
# CONFIG_PACKAGE_ti-5052-firmware is not set
# CONFIG_PACKAGE_wil6210-firmware is not set
# CONFIG_PACKAGE_wireless-regdb is not set
# CONFIG_PACKAGE_wl12xx-firmware is not set
# CONFIG_PACKAGE_wl18xx-firmware is not set
# end of Firmware
```

yunxi993, 2024-05-05:
> @yunxi993 The problem occurs with kernel 6.6, but what about 6.1?

Currently the 6.1 kernel has not been used.

trippleflux, 2024-05-06:
I think @schm0 is referring about [this PR](https://github.com/openwrt/openwrt/pull/5027).

If i am not mistaken i915 has never works recently in OpenWrt.

schm0, 2024-05-06:
> I think @schm0 is referring about [this PR](https://github.com/openwrt/openwrt/pull/5027).
> 
> If i am not mistaken i915 has never works recently in OpenWrt.
Yes, that's it.

Anyway, you can run:
make kernel_menuconfig
or
make kernel_menuconfig CONFIG_TARGET=subtarget
and then include the firmware under:
Device Drivers -> Generic Driver Options -> Firmware loader -> Firmware loading facility -> Build named firmware blobs into the kernel binary
if you want to include multiple firmwares you can separate them by space.
The appear to be taken from the host system, so you have to install the linux firmware package on the host.
Maybe there is a env variable that can be used here to use the firmware from the build system but idk...

vincejv, 2024-05-25:
I cannot replicate this with i7 2600 and i5 9500t
I've tested GPU HW Transcoding via Handbrake and Jellyfin thru docker, both detected the loaded i915 drivers and accelerated my Video Encode/Decode

I've encountered this error on i5 9500t, but i just loaded the firmware and the message is gone, I followed the instructions here: https://forum.openwrt.org/t/silly-little-i915-issue-on-x86/150405/24?page=2

> [ 65.997090] i915 0000:00:02.0: [drm] DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
[ 66.012819] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 0.0.0
[ 66.033246] i915 0000:00:02.0: [drm] ERROR GT0: GuC initialization failed -ENOEXEC

yunxi993, 2024-05-31:
> I cannot replicate this with i7 2600 and i5 9500t on 6.6 I've tested GPU HW Transcoding via Handbrake and Jellyfin thru docker, both detected the loaded i915 drivers and accelerated my Video Encode/Decode
> 
> I've encountered this error on i5 9500t, but i just loaded the firmware and the message is gone, I followed the instructions here: https://forum.openwrt.org/t/silly-little-i915-issue-on-x86/150405/24?page=2
> 

Mine is the built-in core display of Intel N100.

vincejv, 2024-06-05:
Did you follow the instructions on how to install the GPU firmware?

Joee-D, 2024-08-06:
You can use the following script to add the i915, you need to change the url of the bin to your target first.

mkdir -p /openwrt/firmware/i915
curl -L https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/i915/glk_dmc_ver1_04.bin -o /openwrt/firmware/i915/glk_dmc_ver1_04.bin  
echo 'CONFIG_FIRMWARE_IN_KERNEL=y' >> target/linux/x86/64/config-5.15  
echo 'CONFIG_EXTRA_FIRMWARE="i915/glk_dmc_ver1_04.bin"' >> target/linux/x86/64/config-5.15  
echo 'CONFIG_EXTRA_FIRMWARE_DIR="/openwrt/firmware"' >> target/linux/x86/64/config-5.15  

it works
`
Sun Aug  4 09:48:49 2024 kern.info kernel: [    3.742607] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/glk_dmc_ver1_04.bin (v1.4)
`

yunxi993, 2024-08-08:
Thank you very much, I solved it successfully according to your method.

yunxi993, 2024-08-08:
> You can use the following script to add the i915, you need to change the url of the bin to your target first.
> 

Thank you very much, I solved it successfully according to your method.

closed 2024-08-08 by yunxi993

vincejv, 2024-08-26:
@yunxi993 this issue is addressed in this commit https://github.com/openwrt/openwrt/commit/b6ac37110f19d130f608707b0bfcf8c1df60fbdf

no need to build your own image/firmware.

schm0, 2024-08-26:
Actually on my system this does not work.
It tries to load the firmware but later in the log it states that it failed.
I have to build the firmware directly into the kernel to make it work.

yunxi993, 2024-08-27:
> Actually on my system this does not work. It tries to load the firmware but later in the log it states that it failed. I have to build the firmware directly into the kernel to make it work.

You are right, the patch still needs to be compiled into the firmware.
So the 3 kernel options are useless

joez, 2024-09-03:
One way to fix this issue is to build the necessary i915 firmware into the kernel, another way is to make the i915 driver be a kernel module, so that when it is loading the firmwares are available in the rootfs.
PR #16276 is under review, feel free to try it out

yunxi993, 2024-11-16:
Based on your PR.
Do I need to compile `kmod-drm-i915` and `i915-firmware` at the same time? Select `kmod-drm-i915` to compile `i915-firmware-dmc` by default.
Is it possible to compile only `i915-firmware` and default dependencies?

yunxi993, 2024-11-16:
> One way to fix this issue is to build the necessary i915 firmware into the kernel, another way is to use the i915 driver as a kernel module, so that the firmware can be found in the rootfs at load time PR #16276 is under review, feel free to try it out

Based on your PR.

Do I need to compile both `i915-firmware` and `kmod-drm-i915`.

Or just compile `i915-firmware`.

I see that `kmod-drm-i915` only depends on `i915-firmware-dmc`.