openwrt/openwrt #15381: ipq40xx: VLAN Bridging Fails After "generic: 6.1: backport various qca8k fixes patch" on EA3800

jeffsf, 2024-05-04:
### Describe the bug

A configuration that bridges a VLAN from a trunk on the "wan" VIF to the "lan" VIFs ceases to bridge traffic on or after

```
commit a62b1544d1
Author: Christian Marangi <ansuelsmth@gmail.com>
Date:   Tue Sep 26 13:47:44 2023 +0200

    generic: 6.1: backport various qca8k fixes patch
    
    Backport various QCA8K fixes patch merged upstream. Refresh any changed
    patches due to backports.
    
    Signed-off-by: Christian Marangi <ansuelsmth@gmail.com>
```

Connectivity to/from the router over a VLAN on that trunk was still present. 

 

### OpenWrt version

a62b1544d1

### OpenWrt release

a62b1544d1

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys EA8300 (Dallas)

### Image kind

Self-built image

### Steps to reproduce

Observe that the device is configured as a "dumb AP" has a trunk wired to the local network on the "wan" port and three, wired devices supporting IPv4 and IPv6 with DHCP assignment plugged into "lan1" through "lan3". DHCP is supplied elsewhere on the network. These three devices are able to obtain addresses and there is bidirectional communication with the rest of the network on their VLAN 200. (Numbering and other data obfuscated or redacted. Original config available on request.) The router is available through its address on VLAN 10. The router does not participate on VLAN 200.

```
port              vlan-id  
lan1              200 PVID Egress Untagged
lan2              200 PVID Egress Untagged
lan3              200 PVID Egress Untagged
lan4              200 PVID Egress Untagged
wan               10
                  11
                  12
                  200
                  1000
                  1001
br-switch         10
                  11
ap5_1             200 PVID Egress Untagged
ap2_2             1001 PVID Egress Untagged
ap2_3             200 PVID Egress Untagged
ap2_4             200 PVID Egress Untagged
ap2_5             1000 PVID Egress Untagged
ap2_6             200 PVID Egress Untagged
```

Build from `main` branch at commit a62b1544d1 or later. Flash to device running known-good version and preserve config. After reboot and a reasonable period of time, the devices on the "lan" ports are not accessible. Their DHCP leases expire.

Flashing a build from a commit prior to a62b1544d1 has the expected behavior for Ethernet connectivity.
[... 203 lines cut ...]
CONFIG_PACKAGE_syslog-ng=y
CONFIG_PACKAGE_tcpdump-mini=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_uwsgi=y
CONFIG_PACKAGE_uwsgi-cgi-plugin=y
CONFIG_PACKAGE_uwsgi-luci-support=y
CONFIG_PACKAGE_uwsgi-syslog-plugin=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_PACKAGE_xz=y
CONFIG_PACKAGE_xz-utils=y
CONFIG_PACKAGE_zlib=y
CONFIG_PCRE2_JIT_ENABLED=y
CONFIG_RSYNC_acl=y
CONFIG_RSYNC_xattr=y
CONFIG_WPA_MBO_SUPPORT=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jeffsf, 2024-05-04:
It is straightforward for me to provide additional information as the device has two independent firmware installed.

jeffsf, 2024-05-05:
The bot is wrong. The commit is from the `main` branch of the official OpenWrt git repository.

jeffsf, 2024-05-05:
Potentially related to https://github.com/openwrt/openwrt/issues/15327 which involves a non-trivial use of VLAN bridging.

jeffsf, 2024-05-05:
Blindly removing the `target/linux/generic/backport-6.1/780-v6.6-*` patch sequence from commit a62b154 appears to remove the failure. 

```
jeff@deb-devel:~/devel/openwrt-ea8300$ git diff --name-status HEAD^ HEAD
D       target/linux/generic/backport-6.1/780-v6.6-01-net-dsa-qca8k-make-learning-configurable-and-keep-of.patch
D       target/linux/generic/backport-6.1/780-v6.6-02-net-dsa-qca8k-limit-user-ports-access-to-the-first-C.patch
D       target/linux/generic/backport-6.1/780-v6.6-03-net-dsa-qca8k-move-qca8xxx-hol-fixup-to-separate-fun.patch
D       target/linux/generic/backport-6.1/780-v6.6-04-net-dsa-qca8k-use-dsa_for_each-macro-instead-of-for-.patch
```

robimarko, 2024-05-13:
@Ansuel Any ideas?

dchard, 2024-05-16:
@jeffsf just out of curiosity:  did you looked at tcpdump or wireshark to see how those leases expire? 

I am asking as I am also experiencing weird ARP behavior since kernel 6.1, but only with some "easy smart" switches inbetween. In my case the DHCP process goes through, OWRT sends a proper ARP reply, which never reaches the client device when an "easy smart" switch is inbetween. On kernel 5.15.132 it works flawlessly. FYI, kernel 6.6 is also affected which is pretty bad as these patches are already upstreamed...

jeffsf, 2024-05-25:
> @jeffsf just out of curiosity: did you looked at tcpdump or wireshark to see how those leases expire?

I don't know an easy way to snoop a virtual port. All the traffic is between the device on the port and devices on the other end of the trunk. I'm open to suggestions.

The switch here is a Cisco SG500.

jeffsf, 2024-08-25:
Initial test based off `commit 7a7ea98400 (openwrt/master, openwrt/main, openwrt/HEAD, main)` does not immediately exhibit the previous symptoms. I will continue to watch things and update as appropriate.

namiltd, 2024-08-26:
> Leaving open as I don't know if the next release is slated for 6.1 or 6.6.

6.6

motolav, 2024-09-24:
My TP-Link C2600 (IPQ806x with qca8k) still doesn't work when the switch ports are bridged with WAN, I lose connectivity until the setting is auto reset.

dchard, 2024-09-24:
@jeffsf likely this fixed it:

https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=0b6b54eca939a2cb757f854594863548fb93f21f

My issues are gone after the above commit for sure

Djfe, 2025-05-17:
Shall this be closed now? It's been months