openwrt/openwrt #15386: githum-actions bot Improperly Marks Commit on Offical Repo as Invalid

jeffsf, 2024-05-05:
https://github.com/openwrt/openwrt/issues/15381 was marked as Invalid by the bot with 

> Invalid Version reported. `a62b1544d1`
> Is this from a clean repository?

This is not correct. 

As can be seen in the rendering of that issue, the occurrence of the text is auto-linked to the commit.

https://github.com/openwrt/openwrt/commit/a62b1544d15d5cfd9622efcd22b7b835297a1a41

https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=a62b1544d15d5cfd9622efcd22b7b835297a1a41

rany2, 2024-05-05:
These are the values the bot expects: https://github.com/openwrt/openwrt/issues/15327

Instead of providing the commit, you need to either call `./scripts/getver.sh` on the commit you built from or from the MOTD or from os-release.

Ansuel, 2024-05-05:
Yes we had wrong issue from a long time and they got impossible to manage... Having these scary and annoying message seems to be working tho since i'm noticing less and less invalid issue.

I know for advanced user that knows what they are reporting is annoying but the labeller action expects the output of those commands soo any deviation results in Invalid.

Hope you can understand these reasons!

jeffsf, 2024-05-05:
Thanks -- I totally understand the challenges between "It says OpenWrt Chaos Calmer" and incomplete and not-actionable reports.

Updating the template and something in the "invalid" message about what the expected format is might have been helpful. It presently includes the text "The OpenWrt release or commit hash where this bug occurs (use command below)." for both "invalid" fields, so my expectation was that pasting in a valid commit hash on `main` was sufficient.