openwrt/openwrt #15388: Incorrect mapping of port link LED's in MR52

vochong, 2024-05-05:
### Describe the bug

With the official 23.05.3 in Meraki MR52:

eth0 = non-PoE port
eth1 = PoE port
If I plug a cable in the non-PoE port, eth0 is up but the link LEDs of the PoE port blink.
If I plug a cable in the PoE port, eth1 is up but the link LEDs of the non-PoE port blink.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

 23.05.3

### OpenWrt target/subtarget

ipq806x/generic

### Device

 Cisco Meraki MR52

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-05-05:
Swap all lan1 with lan2 in qcom-ipq8068-mr52.dts and recompile (4 times lan1 and 4 times lan2).

vochong, 2024-05-07:
Thanks a lot @namiltd.   I built a 23.05 image using a qcom-ipq8068-mr52.dts that was modified per your instructions and the problem was resolved.

namiltd, 2024-10-24:
@vochong Look at: https://github.com/openwrt/openwrt/pull/16775
Can you test the snapshot version?

PR #15410 (2024-05-07): ipq806x: Meraki MR52: swap lan leds
LAN Leds on Meraki MR52 are wrong and needs to be swapped to actually reflect real ports (lan1<->lan2). 
Fixes: #15388
commit ec8f647d (2024-05-07): ipq806x: swap lan leds for Meraki MR52

LAN Leds on Meraki MR52 are wrong and needs to be swapped to actually
reflect real ports (lan1<->lan2).

Fixes: #15388

Link: https://github.com/openwrt/openwrt/pull/15410
[ wrap commit to 80 columns and improve commit title ]

(cherry picked from commit bd6852185b7c7362fc5485afa961c686d9cd74b4)