openwrt/openwrt #15389: In master snapshots, a STA/Client can never get an IP address from uplink if an AP is configure and ENABLED on the same radio.

vochong, 2024-05-05:
### Describe the bug

In master snapshot images for (for different device platforms, e.g.  ipq40xx, ipq806x etc.), a STA/client can never get any IP address from the uplink if an AP is configured and enabled on the same radio. The AP needs to be temporarily disabled in order for the STA/Client to acquire an IP address from the uplink, and then the AP can be enabled after that.

Such problem is NOT seen in 23.05.3 images.

### OpenWrt version

r26181-d1504a1e76

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq806x/generic

### Device

NETGEAR Nighthawk X4S R7800

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

vochong, 2024-05-12:
@nbd

vochong, 2024-05-15:
@dibdot

Can travelmate be enhanced to overcome this issue in master?

vochong, 2024-05-16:
@robimarko @Ansuel

Can you please give some thoughts about this issue in master images?  I have tried multiple different devices and they all exhibit the same symptom.   There is no such problem in 23.05.x.

robimarko, 2024-05-16:
Sadly, this issue is beyond me, I dont have any ideas

vochong, 2024-05-18:
Hi @nbd168

Can you please give some thoughts about this issue in master images?

vochong, 2024-05-26:
@castillofrancodamian

Hello, are you encountering this issue in master snapshots?

vochong, 2024-05-31:
@brada4

Do you have any idea about this issue?  Thanks.

brada4, 2024-05-31:
@vochong hi, I think it is worth posting ap and sta config that you think are conflicting, also identify wifi device like lspci and/or ethtool phy0-ap0

Works like problem-free on ath10k 10.4, mt76, ath9k (i dont have any other, like I try to go with the mass flow to not stand out)

vochong, 2024-05-31:
@brada4,

I used Luci to configure the STA and AP so I'm sure there's no config error.    If the STA/client gets IP and AP is working at the same time on your device with master snapshot, please try to restart the radio (I used the restart radio function in LUCI) on which STA and AP are configured.   You will see that the STA/Client cannot get an IP address at all unless you temporarily disable AP.   I don't have such problem at all with 23.05.3.  Thanks.

vochong, 2024-06-04:
@brada4
Have you been able to reproduce the problem as I mentioned for master snapshots?  With both STA and AP configured and enabled on the same radio, just reset the radio and see if your STA can get an IP address from the uplink.   Repeat the radio reset several times to see if the problem can be reproduced consistently on your setup.  For me, it's 100% reproducible for master snapshots on different platforms.   However 23.05.03 does not have this problem.

brada4, 2024-06-04:
Do i look like a witch with a crystal ball to guess your config files?

vochong, 2024-06-04:
@brada4 

**WIRELESS CONFIG**
```
'# cat /etc/config/wireless 

config wifi-device 'radio0'
	option type 'mac80211'
	option path 'soc/1b900000.pci/pci0002:00/0002:00:00.0/0002:01:00.0'
	option channel '1'
	option band '2g'
	option htmode 'HT20'
	option cell_density '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'MYSSID-2G'
	option encryption 'psk2'
	option key 'MYPASSWORD'
	option wpa_disable_eapol_key_retries '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0'
	option channel '36'
	option band '5g'
	option htmode 'VHT80'
	option cell_density '0'

config wifi-device 'radio2'
	option type 'mac80211'
	option path 'soc/1b700000.pci/pci0001:00/0001:00:00.0/0001:01:00.0'
	option channel '44'
	option band '5g'
	option htmode 'VHT80'
	option cell_density '0'

config wifi-iface 'wifinet3'
	option device 'radio2'
	option mode 'sta'
	option network 'wwan'
	option ssid 'MYSSID'
	option encryption 'psk2'
	option key 'MYPASSWORD'

config wifi-iface 'wifinet4'
	option device 'radio2'
	option mode 'ap'
	option ssid 'MYSSID-5G'
	option encryption 'psk2'
	option key 'MYPASSWORD'
	option wpa_disable_eapol_key_retries '1'
	option network 'lan'
```

**RESTART RADIO phy1   (on which phy1-sta0 and phy1-ap0 are configured and enabled)**
```
Tue Jun  4 08:55:31 2024 daemon.notice hostapd: Set new config for phy phy1:
Tue Jun  4 08:55:31 2024 daemon.notice hostapd: Remove interface 'phy1'
Tue Jun  4 08:55:31 2024 daemon.notice hostapd: phy1-ap0: interface state ENABLED->DISABLED
Tue Jun  4 08:55:31 2024 daemon.notice hostapd: phy1-ap0: AP-DISABLED
Tue Jun  4 08:55:31 2024 daemon.notice hostapd: phy1-ap0: CTRL-EVENT-TERMINATING
Tue Jun  4 08:55:31 2024 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Tue Jun  4 08:55:31 2024 daemon.notice hostapd: nl80211: deinit ifname=phy1-ap0 disabled_11b_rates=0
Tue Jun  4 08:55:31 2024 kern.info kernel: [ 1303.774403] ath10k_pci 0001:01:00.0 phy1-ap0: left allmulticast mode
Tue Jun  4 08:55:31 2024 kern.info kernel: [ 1303.774451] ath10k_pci 0001:01:00.0 phy1-ap0: left promiscuous mode
Tue Jun  4 08:55:31 2024 kern.info kernel: [ 1303.780125] br-lan: port 3(phy1-ap0) entered disabled state
Tue Jun  4 08:55:31 2024 daemon.notice netifd: Network device 'phy1-ap0' link is down
[... 124 lines cut ...]
Tue Jun  4 08:53:45 2024 daemon.notice netifd: Wireless device 'radio2' is now up
Tue Jun  4 08:53:45 2024 daemon.notice netifd: Interface 'wwan' is enabled
Tue Jun  4 08:53:45 2024 daemon.notice netifd: Interface 'wwan' has link connectivity
Tue Jun  4 08:53:45 2024 daemon.notice netifd: Interface 'wwan' is setting up now
Tue Jun  4 08:53:45 2024 daemon.notice netifd: wwan (6603): udhcpc: started, v1.36.1
Tue Jun  4 08:53:45 2024 daemon.notice netifd: wwan (6603): udhcpc: broadcasting discover
Tue Jun  4 08:53:48 2024 daemon.notice netifd: wwan (6603): udhcpc: broadcasting discover
Tue Jun  4 08:53:48 2024 daemon.notice netifd: wwan (6603): udhcpc: broadcasting select for 192.168.1.210, server 192.168.1.1
Tue Jun  4 08:53:49 2024 daemon.notice netifd: wwan (6603): udhcpc: lease of 192.168.1.210 obtained from 192.168.1.1, lease time 43200
Tue Jun  4 08:53:49 2024 daemon.notice netifd: Interface 'wwan' is now up
Tue Jun  4 08:53:49 2024 user.notice firewall: Reloading firewall due to ifup of wwan (phy1-sta0)

**NOTE THAT phy1-sta0 CAN NOW GET AN IP ADDRESS FROM THE UPLINK**

root@host:/sys/class/leds/green:user# ifconfig phy1-sta0
phy1-sta0 Link encap:Ethernet  HWaddr E2:CB:BC:B4:68:97  
          inet addr:192.168.1.210  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::e0cb:bcff:feb4:6897/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3458 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1076 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:4097175 (3.9 MiB)  TX bytes:230189 (224.7 KiB)

```

castillofrancodamian, 2024-06-07:
> @castillofrancodamian
> 
> Hello, are you encountering this issue in master snapshots?

How did you come to me and assume that I have problems with STA?

vochong, 2024-06-07:
@castillofrancodamian

You filed a similar defect last year:   https://github.com/openwrt/openwrt/issues/13449

brada4, 2024-06-07:
First you have two 5GHz radios in same 80MHz channel, and no country code.
Worth altering ath10k-ct to non-ct kmod and firmware (not touching 3rd board data package)

vochong, 2024-06-08:
The driver default for the country code in  my devices is US.   Even if I manually set it in the config, it does not help anything with the problem.

The problem is observed on multiple devices (EA8500, EA8100v2, R7500v2, R7800, Google Wifi, etc.) when running master snapshot images with either the ath10k or ath10-ct driver/firmware.  No problem with 23.05.3.

How to reproduce the problem:

- Configure both the STA and AP on the same radio, using a specific channel that matches the channel used by the WIFI uplink for STA.
- Reboot the router and check whether the STA interface can acquire an IP address from the uplink. It fails most of the time.
- If the STA does not obtain an IP address, temporarily disable the AP (on the same radio) to allow the STA to get an IP address, then re-enable the AP.
- In LuCI, restart the radio on which the STA and AP are configured and enabled. You will see that the STA will fail to get an IP address.  You can try to restart the radio multiple times to see that the problem persists.

```
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#2
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#1
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)
```

brada4, 2024-06-08:
Are you able to perform simple diagnostic steps:
- set proper country code
- use non-ct mainline driver and firmware
- attach similar error from mt76 device in your list.

Your config is similar to what travelmate does and should work right away

castillofrancodamian, 2024-07-09:
> @castillofrancodamian
> 
> You filed a similar defect last year: #13449
> 
> For me, with master snapshots (kernel 6.6), neither automatic channel nor fixed channel for STA would work if both STA and AP are configured and enabled on the same radio. The 23.05.x images do not have this problem.

I can configure that I have the same problems (MR70X device). I don't know if it's the same issue I reported as you mentioned or if it's something related. Plus there are more problems. Even having only AP interfaces, they are not activated when rebooting the LuCI button or directly rebooting the router.

```
root@mr70x:~# cat /etc/config/wireless 

config wifi-device '2g'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option band '2g'
        option htmode 'HE20'
        option country 'AR'
        option cell_density '1'
        option channel 'auto'

config wifi-iface '2g_lag'
        option device '2g'
        option network 'lan'
        option mode 'ap'
        option ssid 'LAG'
        option encryption 'psk2+ccmp'
        option ifname '2g_lag'
        option key 'password'
        option ieee80211k '1'
        option time_advertisement '2'
        option time_zone '<-03>3'
        option wnm_sleep_mode '1'
        option wnm_sleep_mode_no_keys '1'
        option bss_transition '1'
        option proxy_arp '1'
        option ieee80211r '1'
        option ft_over_ds '0'
        option macfilter 'allow'
        option ft_psk_generate_local '1'
        option ieee80211w '1'

config wifi-device '5g'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
        option channel '36'
        option band '5g'
        option htmode 'HE80'
        option country 'AR'
        option cell_density '1'

config wifi-iface '2g_invitados'
        option device '2g'
        option mode 'ap'
        option ssid 'Invitados'
        option encryption 'psk2+ccmp'
        option ifname '2g_invitados'
        option key 'password'
        option network 'invitados'
        option ieee80211r '1'
        option ft_over_ds '0'
        option ieee80211k '1'
        option time_advertisement '2'
        option time_zone '<-03>3'
        option wnm_sleep_mode '1'
        option wnm_sleep_mode_no_keys '1'
        option bss_transition '1'
        option proxy_arp '1'
        option ft_psk_generate_local '0'
        option ieee80211w '1'

config wifi-iface '5g_lag'
        option device '5g'
        option mode 'ap'
        option ssid 'LAG'
        option encryption 'psk2+ccmp'
        option ifname '5g_lag'
        option key 'password'
        option ieee80211r '1'
        option ft_over_ds '0'
        option ieee80211k '1'
        option time_advertisement '2'
        option time_zone '<-03>3'
        option wnm_sleep_mode '1'
        option wnm_sleep_mode_no_keys '1'
        option bss_transition '1'
        option proxy_arp '1'
        option network 'lan'
        option macfilter 'allow'
        option ft_psk_generate_local '1'
        option ieee80211w '1'

config wifi-iface '5g_invitados'
        option device '5g'
        option mode 'ap'
        option ssid 'Invitados'
        option encryption 'psk2+ccmp'
        option ifname '5g_invitados'
        option key 'password'
        option network 'invitados'
        option ieee80211r '1'
        option ft_over_ds '0'
        option ieee80211k '1'
        option time_advertisement '2'
        option time_zone '<-03>3'
        option wnm_sleep_mode '1'
        option wnm_sleep_mode_no_keys '1'
        option bss_transition '1'
        option proxy_arp '1'
        option ft_psk_generate_local '0'
        option ieee80211w '1'
```

[image]

[image]

vochong, 2024-11-03:
The problem seems to have been resolved.   I have no longer observed it in recent Master snapshots (around November 2024 and later).

closed 2024-11-03 by vochong