openwrt/openwrt #15391: ramips/mt7621: SamKnows Whitebox 8: both phy0 and phy1 identified as 5Ghz

nunojpg, 2024-05-05:
### Describe the bug

/etc/config/wireless have both radio0 and radio1 set to band "5g".
iw phy phy0 info shows support for 5GHz .
iw phy phy1 info shows support for 5GHz AND 2.4GHz.
DTS at: target/linux/ramips/dts/mt7621_samknows_whitebox-v8.dts

### OpenWrt version

master (r26182-49a64c9541)

### OpenWrt release

master (SNAPSHOT)

### OpenWrt target/subtarget

ramips/mt7621

### Device

SamKnows Whitebox 8

### Image kind

Self-built image

### Expected behaviour

There should be a wireless radio configured by default to 2.4GHz.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-05-05:
Add to mt7621_samknows_whitebox-v8.dts line 132 (block &pcie1 mt76@0,0):
```
		ieee80211-freq-limit = <2400000 2500000>;
```
and recompile.

nunojpg, 2024-05-05:
Yes, same in 23.05.03.

namiltd, 2024-05-05:
So if this line helps, a patch is needed for both branches: main and 23.05.

nunojpg, 2024-05-05:
All great. You want me to open the PR?

namiltd, 2024-05-05:
PRs done.

hauke, 2024-05-09:
Could someone with a  SamKnows Whitebox 8 please post the `dmesg` and `iwinfo` output. 
This  page says it uses MediaTek MT7612E and MediaTek MT7603E, see https://openwrt.org/toh/samknows/sk-wb8
The MT7603E is a 2.4 GHz only device and should not use 5GHz.

commit 1a2fc49b (2024-05-25): ramips: samknows whitebox v8: set wifi frequency

Set the 2.4GHz frequency for WiFi.

Fixes: #15391
PR #15395 (2024-05-25): ramips: samknows whitebox v8: set wifi frequency
Set the 2.4GHz frequency for WiFi.
Fixes: #15391
commit 3287d82f (2024-06-06): ramips: samknows whitebox v8: set wifi frequency

Set the 2.4GHz frequency for WiFi.

Fixes: #15391

(cherry picked from commit 1a2fc49b711fdde5d74aacd044ddf51f099c2e16)
PR #15396 (2024-06-06): [23.05] ramips: samknows whitebox v8: set wifi frequency
Set the 2.4GHz frequency for WiFi.
Fixes: #15391
closed 2024-05-25 by openwrt-bot