openwrt/openwrt #15393: ipq40xx: mr33: Unable to read device correct mac address

nasbdh9, 2024-05-05:
### Describe the bug

eth0 mac address changes randomly on every reboot
wlan mac address is different from stock firmware

### OpenWrt version

r26181-d1504a1e76

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Meraki MR33 Access Point

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

nasbdh9, 2024-05-05:
The error also applies to the 23.05 branch
22.03 branch can read mac address correctly

nasbdh9, 2024-05-05:
It can be confirmed that the mac address can be read correctly until 6cdf974a3d0cfebf4a37aaf2876b2ece0f11363d (use kernel 5.15)

Leo-PL, 2024-05-07:
cc @rmilecki

john-tho, 2024-05-16:
Can you test if changing the nvmem definition in the eeprom DTS node to an [nvmem layout](https://elixir.bootlin.com/linux/latest/source/Documentation/devicetree/bindings/nvmem/layouts/fixed-layout.yaml) fixes this?
Current:
https://github.com/openwrt/openwrt/blob/cffd3ad8d7cad2214d44695ad4f2405ea53a26c7/target/linux/ipq40xx/files-6.6/arch/arm/boot/dts/qcom/qcom-ipq4029-insect-common.dtsi#L126-L139

Example layout:
https://github.com/openwrt/openwrt/blob/cffd3ad8d7cad2214d44695ad4f2405ea53a26c7/target/linux/ipq806x/files-6.6/arch/arm/boot/dts/qcom/qcom-ipq8068-mr42.dts#L84-L101

rmilecki, 2024-05-16:
> Can confirm that MAC address cannot be read correctly starting from [b649b0b](https://github.com/openwrt/openwrt/commit/b649b0bf7100bdc6adb7e857c74738cab7c47b4c)

This commits adds extra behaviour only when there is `fixed-layout` in your DTS. I don't see that being used for mr33. Did you actually bisect it? Did you test that commit and one before?

rmilecki, 2024-05-16:
I'd expect such breakage to be rather caused by the b5956700702b ("kernel: backport nvmem v6.6 fixes and v6.7 changes").

nasbdh9, 2024-05-16:
> I'd expect such breakage to be rather caused by the [b595670](https://github.com/openwrt/openwrt/commit/b5956700702b5649ec70bfa0d115af5b439ce6d1) ("kernel: backport nvmem v6.6 fixes and v6.7 changes").

I compiled e46559215573c58fb35ccaceee5fcc961c0813b2 which can read the mac address correctly
After compiling b649b0bf7100bdc6adb7e857c74738cab7c47b4c and the three subsequent submissions and the latest submission, the mac address cannot be read correctly.

nasbdh9, 2024-05-18:
Still cannot read mac address after conversion

```
&blsp1_i2c3 {
        pinctrl-0 = <&i2c_0_pins>;
        pinctrl-names = "default";
        status = "okay";

        eeprom@50 {
                compatible = "atmel,24c64";
                pagesize = <32>;
                reg = <0x50>;
                read-only; /* This holds our MAC & Meraki board-data */

                nvmem-layout {
                        compatible = "fixed-layout";
                        #address-cells = <1>;
                        #size-cells = <1>;

                        mac_address: mac-address@66 {
                                compatible = "mac-base";
                                reg = <0x66 0x6>;
                                #nvmem-cell-cells = <1>;
                        };
                };
        };
};
```

john-tho, 2024-05-18:
Any dmesg messages about NVMEM, atmel, eeprom, or the MAC address assignment?

Can you read the MAC address from the NVMEM SYSFS interface?
Maybe `hexdump -C /sys/bus/nvmem/devices/*/cells/mac-address@66`

nasbdh9, 2024-05-18:
sysfs can read the correct mac address

```
root@OpenWrt:~# hexdump -C /sys/bus/nvmem/devices/*/cells/mac-address@66
00000000  e0 cb bc 48 50 f3                                 |...HP.|
00000006
```

<details>
  <summary>dmesg</summary>
  <pre><code>root@OpenWrt:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.6.30 (nasbdh9@nasbdh9-virtual-build) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 13.2.0 r26310+1-e9e474cfa4) 13.2.0, GNU ld (GNU Binutils) 2.42) #0 SMP Fri May 17 13:14:39 2024
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Meraki MR33 Access Point
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] OF: reserved mem: 0x87e00000..0x87e7ffff (512 KiB) nomap non-reusable smem@87e00000
[    0.000000] OF: reserved mem: 0x87e80000..0x87ffffff (1536 KiB) nomap non-reusable tz@87e80000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080000000-0x000000008fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x0000000087dfffff]
[    0.000000]   node   0: [mem 0x0000000087e00000-0x0000000087ffffff]
[    0.000000]   node   0: [mem 0x0000000088000000-0x000000008fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x000000008fffffff]
[    0.000000] percpu: Embedded 13 pages/cpu s21268 r8192 d23788 u53248
[    0.000000] pcpu-alloc: s21268 r8192 d23788 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Kernel command line:
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 245124K/262144K available (7595K kernel code, 622K rwdata, 1992K rodata, 1024K init, 247K bss, 17020K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000001] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000018] Switching to timer-based delay loop, resolution 20ns
[    0.000325] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000348] CPU: Testing write buffer coherency: ok
[    0.000394] pid_max: default: 32768 minimum: 301
[    0.010213] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.010237] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.018765] qcom_scm: convention: smc legacy
[    0.020217] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.020391] Setting up static identity map for 0x80300000 - 0x8030003c
[    0.020579] rcu: Hierarchical SRCU implementation.
[    0.020586] rcu:     Max phase no-delay instances is 1000.
[    0.021277] smp: Bringing up secondary CPUs ...
[    0.024900] smp: Brought up 1 node, 4 CPUs
[    0.024924] SMP: Total of 4 processors activated (384.00 BogoMIPS).
[    0.024935] CPU: All CPU(s) started in SVC mode.
[    0.033333] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.033550] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.033579] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.038612] pinctrl core: initialized pinctrl subsystem
[    0.041790] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.042134] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.043618] thermal_sys: Registered thermal governor 'step_wise'
[    0.043717] cpuidle: using governor ladder
[... 190 lines cut ...]
[   14.686513] ath10k_ahb a800000.wifi: board_file api 2 bmi_id 0:21 crc32 2d972a4b
[   16.100354] ath10k_ahb a800000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 512 raw 0 hwcrypto 1
[   16.112008] ath: EEPROM regdomain: 0x0
[   16.112039] ath: EEPROM indicates default country code should be used
[   16.112046] ath: doing EEPROM country->regdmn map search
[   16.112056] ath: country maps to regdmn code: 0x3a
[   16.112064] ath: Country alpha2 being used: US
[   16.112072] ath: Regpair used: 0x3a
[   16.118969] kmodloader: done loading kernel modules from /etc/modules.d/*
[   19.092165] zram0: detected capacity change from 0 to 245760
[   19.108546] Adding 122876k swap on /dev/zram0.  Priority:100 extents:1 across:122876k SSDsc
[   25.549129] ipqess-edma c080000.ethernet eth0: Link is Down
[... cut ...]
[   25.581310] br-lan: port 1(lan) entered disabled state
[   25.586422] qca8k-ipq4019 c000000.switch lan: entered allmulticast mode
[   25.591569] ipqess-edma c080000.ethernet eth0: entered allmulticast mode
[   25.598613] qca8k-ipq4019 c000000.switch lan: entered promiscuous mode
[   28.728373] qca8k-ipq4019 c000000.switch lan: Link is Up - 1Gbps/Full - flow control rx/tx
[   28.728474] br-lan: port 1(lan) entered blocking state
[   28.735559] br-lan: port 1(lan) entered forwarding state
  </code></pre>
</details>

john-tho, 2024-05-18:
Okay, so the nvmem cell is created and functional, just not being applied to eth & wifi?
I cannot spot anything obvious in that dmesg.
You might have to add some debug messages to kernel. You can modify in-place in build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_mikrotik/linux-6.6.30/ to skip full Linux rebuild.

I would add messages to [`of_get_mac_address`](https://elixir.bootlin.com/linux/v6.6.31/source/net/core/of_net.c#L126), [`of_get_mac_address_nvmem`](https://elixir.bootlin.com/linux/v6.6.31/source/net/core/of_net.c#L61), [`nvmem_mac_base_read_raw`](https://github.com/openwrt/openwrt/blob/a96354bcfb0ad1c0deccbfdb3264c149447d474b/target/linux/generic/pending-6.6/804-nvmem-core-support-mac-base-fixed-layout-cells.patch#L40), and ipqess: It looks like ipqess no longer prints if it sets a random MAC:
https://github.com/openwrt/openwrt/blob/cffd3ad8d7cad2214d44695ad4f2405ea53a26c7/target/linux/ipq40xx/patches-6.6/700-net-ipqess-introduce-the-Qualcomm-IPQESS-driver.patch#L664-L666
23.05 did: https://github.com/openwrt/openwrt/blob/e61f06c51d11dc5846023cd0fb28322345ca03b3/target/linux/ipq40xx/files/drivers/net/ethernet/qualcomm/ipqess/ipqess.c#L1183-L1191

nasbdh9, 2024-05-18:
is this because ipqess doesn't use the mac_address defined in the device tree

john-tho, 2024-06-28:
Hi @nasbdh9, (or anyone else with an mr33) I _may_ have bumped into the fix for this.
If you could test out https://github.com/openwrt/openwrt/pull/15831 for us, at least that would give us the `of_get_ethdev_address` return code. It might even fix this.

Cheers

nasbdh9, 2024-06-29:
Patch does not work, still uses random mac address. 
```
root@OpenWrt:~# dmesg | grep ipqess
[    1.425929] ipqess-edma c080000.ethernet: using random MAC address 32:f9:4f:59:9b:67
[    1.540704] ipqess-edma c080000.ethernet eth0: entered promiscuous mode
[    4.807997] ipqess-edma c080000.ethernet eth0: configuring for fixed/internal link mode
[    4.814922] ipqess-edma c080000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   23.320887] ipqess-edma c080000.ethernet eth0: Link is Down
[   23.326945] ipqess-edma c080000.ethernet eth0: configuring for fixed/internal link mode
[   23.327226] ipqess-edma c080000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   23.360493] ipqess-edma c080000.ethernet eth0: entered allmulticast mode

```

john-tho, 2024-06-29:
> Patch does not work, still uses random mac address.

No such luck. Thanks.

Sorry, I had just changed the patch so that the return code of `of_get_ethdev_address` no longer print by default; https://github.com/openwrt/openwrt/compare/d2a47e8333499abf135e10177ea679b4f1af21d9..b4a48603dd5b0e7f0afad6f5a06c23eb96b5650b.

Could you please make that print for us? This specific commit that still includes it: https://github.com/openwrt/openwrt/commit/d2a47e8333499abf135e10177ea679b4f1af21d9 ([patch](https://github.com/openwrt/openwrt/commit/d2a47e8333499abf135e10177ea679b4f1af21d9.patch)), or you could manually change the change the `dev_dbg` to a `dev_info` (opposite that compare) in `target/linux/ipq40xx/patches-6.6/700-net-ipqess-introduce-the-Qualcomm-IPQESS-driver.patch`

This below prints the error, and also prints the the OF node:
```
      err = of_get_ethdev_address(np, netdev);
      if (err) {
              dev_info(&pdev->dev, "failed to get MAC address from DT: %d %pOF\n", err, np);
```

Cheers

john-tho, 2024-06-29:
> Patch does not work, still uses random mac address.

No such luck. Thanks.

Sorry, I had just changed the patch so that the return code of `of_get_ethdev_address` to no longer print by default; https://github.com/openwrt/openwrt/compare/d2a47e8333499abf135e10177ea679b4f1af21d9..b4a48603dd5b0e7f0afad6f5a06c23eb96b5650b.

Could you please make that print for us? This specific commit that still includes it: https://github.com/openwrt/openwrt/commit/d2a47e8333499abf135e10177ea679b4f1af21d9 ([patch](https://github.com/openwrt/openwrt/commit/d2a47e8333499abf135e10177ea679b4f1af21d9.patch)), or you could manually change the change the `dev_dbg` to a `dev_info` (opposite that compare) in `target/linux/ipq40xx/patches-6.6/700-net-ipqess-introduce-the-Qualcomm-IPQESS-driver.patch`

This below prints the error, and also prints the the OF node:
```
      err = of_get_ethdev_address(np, netdev);
      if (err) {
              dev_info(&pdev->dev, "failed to get MAC address from DT: %d %pOF\n", err, np);
```

Kernel 6.6.31 did include a fix for `at24.c`: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/log/drivers/misc/eeprom/at24.c?h=linux-6.6.y

Cheers

john-tho, 2024-06-30:
Those ipqess changes were merged in main, so if you build an image that includes them (https://github.com/openwrt/openwrt/commit/515ad3ce8a8fbb062d366a08b6d4e8ce04f7f8bb or newer), with OpenWrt's kernel DYNAMIC_DEBUG config option set: `CONFIG_KERNEL_DYNAMIC_DEBUG=y`, we can do some more debugging:

The below will enable dynamic debug for the
- eeprom driver https://elixir.bootlin.com/linux/latest/source/drivers/misc/eeprom/at24.c#L360
- nvmem https://elixir.bootlin.com/linux/v6.6.36/C/ident/of_get_mac_address_nvmem
- ethernet driver https://github.com/openwrt/openwrt/blob/b4069fad025abb26e17988b769a6589b6c504dfb/target/linux/ipq40xx/patches-6.6/700-net-ipqess-introduce-the-Qualcomm-IPQESS-driver.patch#L1201

Then remove and re-activate the ethernet driver:
```
echo 'file "drivers/misc/eeprom/*" +p; file "drivers/nvmem/*" +p; file "drivers/net/ethernet/qualcomm/ipqess/*" +p;'>/sys/kernel/debug/dynamic_debug/control
$(echo "c080000.ethernet">/sys/bus/platform/drivers/ipqess-edma/unbind;
echo "c080000.ethernet">/sys/bus/platform/drivers/ipqess-edma/bind;
echo "c000000.switch">/sys/bus/platform/drivers/qca8k-ipq4019/bind) &
```
Then at least we should see the `of_get_ethdev_address` return code in `dmesg`.

From a quick look through drivers/nvmem/core.c for ipq40xx, it looks like, because it is not a `fixed-layout`, the `mac_address` DTS node won't get the suitable nvmem `read_post_process` function to be able to adjust the MAC address. I haven't seen anything that would make calling the NVMEM cell with an unused `index` stop things from working, though.

nasbdh9, 2024-07-06:
```
BusyBox v1.36.1 (2024-07-05 08:05:58 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r26809+59-7c9644a7b5
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# dmesg | grep ipqess
[    1.427062] ipqess-edma c080000.ethernet: using random MAC address 56:80:bb:8c:d1:09
[    1.541646] ipqess-edma c080000.ethernet eth0: entered promiscuous mode
[    4.840732] ipqess-edma c080000.ethernet eth0: configuring for fixed/internal link mode
[    4.847661] ipqess-edma c080000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   45.528956] ipqess-edma c080000.ethernet eth0: Link is Down
[   45.535165] ipqess-edma c080000.ethernet eth0: configuring for fixed/internal link mode
[   45.535448] ipqess-edma c080000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   45.568590] ipqess-edma c080000.ethernet eth0: entered allmulticast mode
root@OpenWrt:~# echo 'file "drivers/misc/eeprom/*" +p; file "drivers/nvmem/*" +p; file "drivers/net/ethernet/qualcomm/ipqess/*" +p;'>/sys/kernel/debug/dynamic_debug/control
root@OpenWrt:~# $(echo "c080000.ethernet">/sys/bus/platform/drivers/ipqess-edma/unbind;
> echo "c080000.ethernet">/sys/bus/platform/drivers/ipqess-edma/bind;
> echo "c000000.switch">/sys/bus/platform/drivers/qca8k-ipq4019/bind) &
root@OpenWrt:~#
Using username "root".

BusyBox v1.36.1 (2024-07-05 08:05:58 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r26809+59-7c9644a7b5
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# dmesg | grep ipqess
[    1.427062] ipqess-edma c080000.ethernet: using random MAC address 56:80:bb:8c:d1:09
[    1.541646] ipqess-edma c080000.ethernet eth0: entered promiscuous mode
[    4.840732] ipqess-edma c080000.ethernet eth0: configuring for fixed/internal link mode
[    4.847661] ipqess-edma c080000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   45.528956] ipqess-edma c080000.ethernet eth0: Link is Down
[   45.535165] ipqess-edma c080000.ethernet eth0: configuring for fixed/internal link mode
[   45.535448] ipqess-edma c080000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   45.568590] ipqess-edma c080000.ethernet eth0: entered allmulticast mode
[  179.774997] ipqess-edma c080000.ethernet eth0: left allmulticast mode
[  179.789228] ipqess-edma c080000.ethernet eth0: left promiscuous mode
[  179.852735] ipqess-edma c080000.ethernet eth0: Link is Down
[  180.341371] ipqess-edma c080000.ethernet: failed to get MAC address from DT: -2
[  180.341418] ipqess-edma c080000.ethernet: using random MAC address 2e:da:fa:f7:42:d8
[  180.613711] ipqess-edma c080000.ethernet eth0: entered promiscuous mode
[  180.621876] ipqess-edma c080000.ethernet eth0: configuring for fixed/internal link mode
[  180.631644] ipqess-edma c080000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  180.664985] ipqess-edma c080000.ethernet eth0: entered allmulticast mode
```

john-tho, 2024-07-07:
> ```
> [  180.341371] ipqess-edma c080000.ethernet: failed to get MAC address from DT: -2
> ```

Thanks. `#define	ENOENT		 2	/* No such file or directory */`

Could I please double check that you can access the NVMEM cell via `hexdump -C /sys/bus/nvmem/devices/*/cells/mac-address@66` with kernel 6.6?
If you can, I have not worked out how yet, given no `add_legacy_fixed_of_cells` for `at24.c`…

If not, could you please try again with [a DTB `fixed-layout`](https://github.com/openwrt/openwrt/issues/15393#issuecomment-2118646753).
Or set `nvmem_config.add_legacy_fixed_of_cells = true;` in `at24_probe` in `drivers/misc/eeprom/at24.c`.

Only a few places in `nvmem/core.c` return `-ENOENT`:
`of_nvmem_device_get`—not used
`nvmem_cell_get_from_lookup`—used from `nvmem_cell_get` if `of_nvmem_cell_get` errors, and error is not `-EPROBE_DEFER`.
`of_nvmem_cell_get`

My current guess is that the OF cells are not being added (no `config->add_legacy_fixed_of_cells`, and no `fixed-layout`), so `of_nvmem_cell_get` fails in `nvmem_find_cell_entry_by_node`. This is called through:
`of_get_ethdev_address`
`of_get_mac_address_nvmem`
`&gmac` has a platform_device, so we take that branch, and use
`nvmem_get_mac_address`
`nvmem_cell_get(dev, "mac-address")`
`of_nvmem_cell_get(&gmac, "mac-address")`
(guessing,) eventually [`nvmem_find_cell_entry_by_node(nvmem, cell_np)` finds no cells](https://elixir.bootlin.com/linux/v6.8/source/drivers/nvmem/core.c#L1470), so `return ERR_PTR(-ENOENT);`

Leo-PL, 2024-10-06:
@john-tho this is correct. I've just tried your suggestion (independently) and it fixes the issue. Will push a fix in a few moments.

closed 2024-10-07 by openwrt-bot