openwrt/openwrt #15398: Linksys E8450 Phy not found for 5GHz on custom snapshot build from the latest commit

CharlesMengCA, 2024-05-05:
### Describe the bug

After upgraded to latest build from the main branch, the radio1 became unknown and not started.

Found following message in the system log

```

Sun May  5 18:45:45 2024 daemon.notice netifd: radio1 (2229): WARNING: Variable 'data' does not exist or is not an array/object
Sun May  5 18:45:45 2024 daemon.notice netifd: radio1 (2229): Phy not found
Sun May  5 18:45:45 2024 daemon.notice netifd: radio1 (2229): Could not find PHY for device 'radio1'
Sun May  5 18:45:45 2024 daemon.notice netifd: Wireless device 'radio1' set retry=0
Sun May  5 18:45:45 2024 daemon.crit netifd: Wireless device 'radio1' setup failed, retry=0
Sun May  5 18:45:45 2024 daemon.notice netifd: radio1 (2301): WARNING: Variable 'data' does not exist or is not an array/object
Sun May  5 18:45:45 2024 daemon.notice netifd: radio1 (2301): Bug: PHY is undefined for device 'radio1'
Sun May  5 18:45:45 2024 daemon.notice netifd: Wireless device 'radio1' is now down
```

### OpenWrt version

r26186-064d4a8083

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Additional info

<details>
  <summary>System Log</summary>
  
  ```
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Sun May  5 18:45:47 2024 kern.notice kernel: [    0.000000] Linux version 6.6.30 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.2.0 r26186-064d4a8083) 13.2.0, GNU ld (GNU Binutils) 2.42) #0 SMP Sun May  5 19:45:52 2024
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000] Machine model: Linksys E8450 (UBI)
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000] earlycon: uart8250 at MMIO32 0x0000000011002000 (options '')
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000] printk: bootconsole [uart8250] enabled
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000] OF: reserved mem: 0x0000000042ff0000..0x0000000042ffffff (64 KiB) map non-reusable ramoops@42ff0000
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000] OF: reserved mem: 0x0000000043000000..0x000000004302ffff (192 KiB) nomap non-reusable secmon@43000000
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000] Zone ranges:
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000]   DMA32    empty
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000]   Normal   empty
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000] Early memory node ranges
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000043030000-0x000000005fffffff]
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000] psci: PSCIv1.1 detected in firmware.
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.4
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000] percpu: Embedded 18 pages/cpu s34536 r8192 d31000 u73728
Sun May  5 18:45:47 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s34536 r8192 d31000 u73728 alloc=18*4096
Sun May  5 18:45:47 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1
Sun May  5 18:45:47 2024 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
[... 338 lines cut ...]
Sun May  5 18:45:51 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Sun May  5 18:45:51 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Sun May  5 18:45:51 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Sun May  5 18:45:51 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Sun May  5 18:45:51 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Sun May  5 18:45:51 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Sun May  5 18:45:51 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Sun May  5 18:45:51 2024 daemon.warn dnsmasq[1]: no servers found in /tmp/resolv.conf.d/resolv.conf.auto, will retry
Sun May  5 18:45:51 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sun May  5 18:45:51 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Sun May  5 18:45:53 2024 kern.info kernel: [   14.213286] mt7530-mdio mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control rx/tx
Sun May  5 18:45:53 2024 kern.info kernel: [   14.221261] br-lan: port 5(wan) entered blocking state
Sun May  5 18:45:53 2024 kern.info kernel: [   14.226414] br-lan: port 5(wan) entered forwarding state
Sun May  5 18:45:53 2024 daemon.notice netifd: Network device 'wan' link is up
Sun May  5 18:45:53 2024 daemon.notice netifd: bridge 'br-lan' link is up
Sun May  5 18:45:53 2024 daemon.notice netifd: Interface 'lan' has link connectivity
Sun May  5 18:45:54 2024 daemon.notice ksmbd: Stopping Ksmbd userspace service.
Sun May  5 18:45:54 2024 daemon.notice ksmbd: Starting Ksmbd userspace service.
Sun May  5 18:52:28 2024 daemon.info procd: - init complete -
  ```
</details>

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

CharlesMengCA, 2024-05-05:
<details>
  <summary>diffconfig</summary>
  
  ```
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_mt7622=y
CONFIG_TARGET_mediatek_mt7622_DEVICE_linksys_e8450-ubi=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_AUTOREMOVE=y
CONFIG_BPF_TOOLCHAIN_BUILD_LLVM=y
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_BUILDBOT=y
CONFIG_COLLECT_KERNEL_DEBUG=y
# CONFIG_GDB is not set
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_IB=y
CONFIG_IMAGEOPT=y
CONFIG_JSON_CYCLONEDX_SBOM=y
CONFIG_KERNEL_BUILD_DOMAIN="buildhost"
CONFIG_KERNEL_BUILD_USER="builder"
# CONFIG_KERNEL_KALLSYMS is not set
CONFIG_MAKE_TOOLCHAIN=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_bridger=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_kmod-asn1-decoder=y
CONFIG_PACKAGE_kmod-crypto-des=y
CONFIG_PACKAGE_kmod-crypto-ecb=y
CONFIG_PACKAGE_kmod-crypto-md5=y
CONFIG_PACKAGE_kmod-crypto-sha256=y
CONFIG_PACKAGE_kmod-fs-exfat=y
CONFIG_PACKAGE_kmod-fs-ksmbd=y
CONFIG_PACKAGE_kmod-fs-netfs=y
CONFIG_PACKAGE_kmod-fs-ntfs3=y
CONFIG_PACKAGE_kmod-fs-smbfs-common=y
CONFIG_PACKAGE_kmod-fs-vfat=y
CONFIG_PACKAGE_kmod-ifb=m
CONFIG_PACKAGE_kmod-lib-crc-ccitt=m
CONFIG_PACKAGE_kmod-nls-cp437=y
CONFIG_PACKAGE_kmod-nls-iso8859-1=y
CONFIG_PACKAGE_kmod-nls-ucs2-utils=y
CONFIG_PACKAGE_kmod-nls-utf8=y
CONFIG_PACKAGE_kmod-oid-registry=y
# CONFIG_PACKAGE_kmod-ppp is not set
CONFIG_PACKAGE_kmod-sched-act-vlan=y
CONFIG_PACKAGE_kmod-sched-bpf=y
CONFIG_PACKAGE_kmod-sched-cake=m
CONFIG_PACKAGE_kmod-sched-core=y
CONFIG_PACKAGE_kmod-sched-flower=y
CONFIG_PACKAGE_kmod-scsi-core=y
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_ksmbd-server=y
CONFIG_PACKAGE_libbpf=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_liblzo=y
CONFIG_PACKAGE_libnl-core=y
CONFIG_PACKAGE_libnl-genl=y
CONFIG_PACKAGE_libpcap=m
CONFIG_PACKAGE_libxdp=m
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-ksmbd=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
[... 840 lines cut ...]
# CONFIG_PACKAGE_r8169-firmware is not set
# CONFIG_PACKAGE_rs9113-firmware is not set
# CONFIG_PACKAGE_rt2800-pci-firmware is not set
# CONFIG_PACKAGE_rt2800-usb-firmware is not set
# CONFIG_PACKAGE_rtl8192ce-firmware is not set
# CONFIG_PACKAGE_rtl8192cu-firmware is not set
# CONFIG_PACKAGE_rtl8192de-firmware is not set
# CONFIG_PACKAGE_rtl8192se-firmware is not set
# CONFIG_PACKAGE_rtl8723de-firmware is not set
# CONFIG_PACKAGE_rtl8821ae-firmware is not set
# CONFIG_PACKAGE_rtl8821ce-firmware is not set
# CONFIG_PACKAGE_rtl8822be-firmware is not set
# CONFIG_PACKAGE_rtl8822ce-firmware is not set
# CONFIG_PACKAGE_snapshot-tool is not set
# CONFIG_PACKAGE_umbim is not set
# CONFIG_PACKAGE_uqmi is not set
# CONFIG_PACKAGE_wil6210-firmware is not set
# CONFIG_PACKAGE_wl12xx-firmware is not set
# CONFIG_PACKAGE_wl18xx-firmware is not set
# CONFIG_PACKAGE_wwan is not set
# CONFIG_PACKAGE_xtables-legacy is not set
# CONFIG_PACKAGE_xtables-nft is not set

  ```
</details>

rany2, 2024-05-05:
I fixed it in this PR: #15397

CharlesMengCA, 2024-05-05:
Thanks, rany2. I am building it with you fix and see how it goes. :)

Is there a bug with the github-actions bot? I used the suggested command to generate the version and Target/Subtarget.

rany2, 2024-05-05:
Re the GitHub actions bot; I don't think you did anything wrong, but the bot seems to think you didn't fill anything. Maybe there's an extra newline/whitespace that fooled it?

Anyway let me know if the fix worked, it is working fine on my Redmi AX6S and afaik Linksys E8450 is basically identical.

CharlesMengCA, 2024-05-06:
The PR is working. Thanks a lot.

mrkiko, 2024-05-06:
Why not close this issue then?

closed 2024-05-06 by nbd168