openwrt/openwrt #15404: ASUS RT-AC51U 5G LED does not works.

pw44, 2024-05-06:
### Describe the bug

Installed 23.05.3 release and 5G wifi indicatior does not lights.

Tried some of the hints, but simply unable to make it work

### OpenWrt version

r23497-6637af95aa

### OpenWrt release

23.05.0

### OpenWrt target/subtarget

ramips/mt7620

### Device

Asus RT-AC51U

### Image kind

Official downloaded image

### Steps to reproduce

connect to device on 5G channel

### Actual behaviour

5G led does not lights

### Expected behaviour

blinking, like the 2G led

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-05-06:
ASUS RT-AC51U has only one WiFi LED, it is configured for 2.4GHz.
[image]

Shine-, 2024-05-06:
Not sure which model **your**  photo comes from, but it's not a RT-AC51U for sure. Here's a photo of a RT-AC51U (just taken minutes ago), and you can clearly see the 5G WiFi LED:
[image]
The 5G LED not working has always been a known issue with this device, though.

namiltd, 2024-05-06:
You need to enable gpio1 and gpio2 and look for the address of this LED according to https://openwrt.org/docs/techref/hardware/port.gpio

pw44, 2024-05-06:
> Not sure which model **your** photo comes from, but it's not a RT-AC51U for sure. Here's a photo of a RT-AC51U (just taken minutes ago), and you can clearly see the 5G WiFi LED: ![AC51U](https://private-user-images.githubusercontent.com/6599044/328311495-d145947a-5cb4-4a15-b3d4-0a645935b251.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUwMzgwMDYsIm5iZiI6MTcxNTAzNzcwNiwicGF0aCI6Ii82NTk5MDQ0LzMyODMxMTQ5NS1kMTQ1OTQ3YS01Y2I0LTRhMTUtYjNkNC0wYTY0NTkzNWIyNTEuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MDZUMjMyMTQ2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MWNiZWYxYmE2YmFlMjZlMDg3MTczZDQyMjk3ZmM0OWViMjQ3ZTdhNDc0ZDZhYWY0ZmJkNWQwZTI4NGQwMmFhOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.plGbR44OSVDk3OWEpxC0EpX0wyp3WCdZE73rjeFpoSs) The 5G LED not working has always been a known issue with this device, though.

exactly the one i have.

pw44, 2024-05-06:
> You need to enable gpio1 and gpio2 and look for the address of this LED according to https://openwrt.org/docs/techref/hardware/port.gpio
> 
> Add this to mt7620a_asus_rt-ac51u.dts:
> 
> ```
> &gpio1 {
> 	status = "okay";
> };
> &gpio2 {
> 	status = "okay";
> };
> ```

where to add it? and how to make it permanent?

namiltd, 2024-05-07:
The mt7620a_asus_rt-ac51u.dts file is a source file (e.g. in https://github.com/openwrt/openwrt/archive/refs/tags/v23.05.3.zip), changing it requires recompiling the image.

pw44, 2024-05-07:
> The mt7620a_asus_rt-ac51u.dts file is a source file (e.g. in https://github.com/openwrt/openwrt/archive/refs/tags/v23.05.3.zip), changing it requires recompiling the image.

i see... i guess i'm not that skilled ;-) anyway, i will take a look.

namiltd, 2024-05-07:
So test gpio0 and gpio3 which are enabled for this router.

pw44, 2024-05-07:
> So test gpio0 and gpio3 which are enabled for this router.

and how do i test those? and later, how do i enable those and make permanent?

namiltd, 2024-05-07:
The testing method is described at https://openwrt.org/docs/techref/hardware/port.gpio
First you need to find the 5 GHz Wi-Fi LED and then configure it in the dts file. If everything works, create a Pull Request with the patch.

pw44, 2024-05-07:
> The testing method is described at https://openwrt.org/docs/techref/hardware/port.gpio First you need to find the 5 GHz Wi-Fi LED and then configure it in the dts file. If everything works, create a Pull Request with the patch.

ok, and where am i able to create a file? seams it's not writeable. :-(

namiltd, 2024-05-07:
You need to copy the sources to your local computer (there are several ways) and make changes locally and then compile.

Lanchon, 2024-05-10:
> ok, and where am i able to create a file? seams it's not writeable. :-(

there is a details page on the wiki describing building. but it takes some effort for sure.

start by testing the GPIO pins to find the right pin for the missing led and the value required to turn it on, then post that info here. if not you, then at least someone else could pick it up later.

DragonBluep, 2024-07-06:
SSH try: `echo $pin_num > /sys/kernel/debug/ieee80211/phy0/mt76/led_pin`. Replace `$pin_num` with 1, 2, 3...