openwrt/openwrt #15408: No WiFI radio is detected in MR52 when running master snapshot images (kernel 6.6)

vochong, 2024-05-07:
### Describe the bug

With master snapshot images (kernel 6.6) for Meraki MR52, none of its WIFI radios is detected and no WIFI firmware is loaded.

### OpenWrt version

r26199-0b0e3e22f8

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq806x/generic

### Device

Cisco Meraki MR52

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

danpawlik, 2024-05-07:
Need to check once again, but I also have an issue after upgrade to image that contains kernel 6.6.30 on my AX3200.

namiltd, 2024-05-07:
Attach Kernel Log.

vochong, 2024-05-08:
@namiltd 

Some other observations:

In 23.05.3,  PoE port = WAN,  non-PoE port = LAN.   It's the reverse in master kernel 6.6.

In master kernel 6.6:
root@OpenWrt:~# uname -a
Linux OpenWrt 6.6.30 #0 SMP Tue May  7 20:26:10 2024 armv7l GNU/Linux

root@OpenWrt:~# opkg list-installed | grep ath10k
ath10k-board-qca9887 - 20240220-r1
ath10k-board-qca9984 - 20240220-r1
ath10k-firmware-qca9887-ct - 2020.11.08-r1
ath10k-firmware-qca9984-ct - 2020.11.08-r1
kmod-ath10k-ct - 6.6.30.2023.06.05~fadd0768-r1

root@OpenWrt:~# lsmod | grep qca
root@OpenWrt:~# 

root@OpenWrt:~# iw dev
root@OpenWrt:~# 

[mr52_23.05.3_boot.log](https://github.com/openwrt/openwrt/files/15243053/mr52_23.05.3_boot.log)
[mr52_kernel6.6.30_boot.log](https://github.com/openwrt/openwrt/files/15243054/mr52_kernel6.6.30_boot.log)

namiltd, 2024-05-08:
I see that "Generic PHY" is loaded twice instead of "Qualcomm Atheros AR8031/AR8033". Under investigation.

vochong, 2024-05-08:
Sorry, I never tried any image with kernel 6.1 on MR52.  It's quite time-consuming to disassemble and recover MR52 via USB-TTL so I'm careful with using master images.

namiltd, 2024-05-08:
Images with kernel 6.6 is only available from Fri, 26 Apr 2024. Maybe you uploaded older snapshots?

namiltd, 2024-05-08:
Last shapshot with kernel 6.1.86+LuCI: [mr52-6.1.86.zip](https://github.com/openwrt/openwrt/files/15251551/mr52-6.1.86.zip)

vochong, 2024-05-08:
Same problem with kernel 6.1
[mr52-kernel-6.1.log](https://github.com/openwrt/openwrt/files/15251883/mr52-kernel-6.1.log)

namiltd, 2024-05-08:
Thank you. Now at least we know that it's not a 6.6 kernel issue, but an update from 5.15 to 6.1.

namiltd, 2024-05-11:
Try it: [mr52.zip](https://github.com/openwrt/openwrt/files/15284491/mr52.zip)
Check interface and led order.

vochong, 2024-05-12:
No change for r26289-61e8728d86

- No Wifi detected
- LAN port is still the PoE port.   As a matter of fact, the PoE port must be the WAN port by default.

[mr52_r26289-61e8728d86.log](https://github.com/openwrt/openwrt/files/15284934/mr52_r26289-61e8728d86.log)

vochong, 2024-05-12:
FYI,  the same missing WIFI issue is seen in MR42 with master images.

[mr42_kernel6.6_nowifi.log](https://github.com/openwrt/openwrt/files/15285136/mr42_kernel6.6_nowifi.log)

namiltd, 2024-05-12:
It's strange because I swapped eth0 and eth1 in this image.

robimarko, 2024-05-12:
Well, its obviously an PCIe issue since none of the 3 ports have a PHY linkup at the end of configuration.

Its hard to tell what it could be, but I find it extremely weird that this device uses no PERST GPIO-s at all

vochong, 2024-05-15:
Hi @robimarko,

If MR42/MR52 don't use PERST GPIO-s, any reason why they work well in 23.05.x ?  Thanks.

robimarko, 2024-05-15:
Without having the device or somebody at least doing a bisect in OpenWrt to see which commit in master breaks stuff it's impossible to say.

Ansuel, 2024-05-15:
@robimarko regression caused by pcie rework done ""recently"" ? maybe?

robimarko, 2024-05-15:
I looked at 6.1 DWC QCOM commits and there isn't anything screaming out but as you know it could literaly be anything

vochong, 2024-05-15:
@robimarko @Ansuel 

I did a hard reset to the commit 9a5b1af670185f07a797920f861a6033e61c5c61 (July 23 2023) around the time ipq806x was switched to 6.1.    I saw the same failures.

[   10.897510] pcieport 0002:00:00.0: Failed to create device link (0x180) with 0-0056
[   10.902880] pcieport 0001:00:00.0: Failed to create device link (0x180) with 0-0056
[   10.910417] pcieport 0000:00:00.0: Failed to create device link (0x180) with 0-0056

[mr42_early_kernel6.1_dmesg.log](https://github.com/openwrt/openwrt/files/15327611/mr42_early_kernel6.1_dmesg.log)

Ansuel, 2024-05-16:
Oh nice finding... Any clue why the property delete wasn't working?

robimarko, 2024-05-16:
Because the older `-gpio` (Not -gpios) suffix was used in the SoC DTSI but `/delete-property/ perst-gpios;` in OpenWrt DTS so it was not deleted as there was no such property.

And then upstream got updated in:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/arch/arm/boot/dts/qcom-ipq8064.dtsi?h=v6.1.90&id=a01bae600a0d07038ec686a18604bf86452633d8

So now they are removed.

That explains it as not using PERST GPIOs made no sense, I cannot find the stock DTS to see what they are doing but I suspect that all of those PCIE deletes are incorrect

Ansuel, 2024-05-16:
This is similar to usb being broken just cause they renamed the entry
upstream...

robimarko, 2024-05-16:
Well, in this case our downstream DTS seems that it was wrong all of the time and it worked purely by luck (Or bad copy/paste).

@vochong Can you try and remove `/delete-property/ perst-gpios;` from:
https://github.com/openwrt/openwrt/blob/main/target/linux/ipq806x/files-6.6/arch/arm/boot/dts/qcom/qcom-ipq8068-cryptid-common.dtsi

And then flash that?

Ansuel, 2024-05-16:
No idea why they remove that on original firmware... They probably don't trust the kernel pcie driver and lease U-Boot take the init... Not the case for us. Pretty sure with the property removed the thing will start working.

vochong, 2024-05-16:
@robimarko @Ansuel 

Yes, I already built images and tested both MR42 and MR52 after reading the comment by @rx78gp01.   WiFi works perfectly on them now.  Thank you everyone for the quick solution.

namiltd, 2024-05-16:
What about the port order? Do the LEDs indicate the correct ports?

namiltd, 2024-05-16:
@vochong What about the port order? Do the LEDs indicate the correct ports?

closed 2024-05-16 by openwrt-bot

vochong, 2024-05-16:
@namiltd

Yes, both port order and their LED's are correct after removing these /delete-property/ fields.    I don't know see any relationship but everything works correctly thereafter.

namiltd, 2024-05-25:
Another problem with mr42/mr52: https://forum.openwrt.org/t/mr42-mr52-leds/198842

vochong, 2024-05-25:
I don't think we should care about the exact system LED color scheme from the original OEM firmware because we can achieve much more with WiFi radios in OpenWrt. There's no one-to-one matching.

-----------------------------------
LED Indicators and Run Dark Mode
Your MR42 is equipped with a multi-color LED light on the front of the unit to convey information about system functionality and performance:

Orange - AP is booting (permanent Orange suggests hardware issue)
Rainbow - AP is initializing/scanning
Blinking Blue - AP is upgrading
Green - AP in Gateway mode with no clients
Blue - AP in Gateway mode with clients 
Blinking Orange - AP can't find uplink 
-----------------------------------

commit 96e402ae (2024-05-16): ipq806x: mr42/mr52: fix PCIe ports

All three PCIe ports are reported non working on Meraki MR42/MR52 boards
since kernel 6.1 with the issue of PCIe PHY link never coming up thus
no WLAN cards are available on the boards.

After debugging it seems that PCIe worked on 5.15 and older purely by
accident as device DTS was using /delete-property/ perst-gpios; in each
of the 3 PCIe nodes but there was no "perst-gpios" property in the SoC DTSI
as it was still using the older "perst-gpio" property so it was not getting
removed from the device DTS.

However, in kernel 6.1 commit ("ARM: dts: qcom-*: rep [... cut]
PR #15509 (2024-05-16): ipq806x: mr42/mr52: fix PCIe ports
All three PCIe ports are reported non working on Meraki MR42/MR52 boards since kernel 6.1 with the issue of PCIe PHY link never coming up thus no WLAN cards are available on the boards.

After debugging it seems that PCIe worked on 5.15 and older purely by accident as device DTS was using /delete-property/ perst-gpios; in each of the 3 PCIe nodes but there was no "perst-gpios" property in the SoC DTSI as it was still using the older "perst-gpio" property so it was not getting removed from the device DTS.

However, in kernel 6.1 commit ("ARM: dts: qcom-*: repl [... cut]
commit e61f06c5 (2024-05-17): ipq806x: mr42/mr52: fix PCIe ports

All three PCIe ports are reported non working on Meraki MR42/MR52 boards
since kernel 6.1 with the issue of PCIe PHY link never coming up thus
no WLAN cards are available on the boards.

After debugging it seems that PCIe worked on 5.15 and older purely by
accident as device DTS was using /delete-property/ perst-gpios; in each
of the 3 PCIe nodes but there was no "perst-gpios" property in the SoC DTSI
as it was still using the older "perst-gpio" property so it was not getting
removed from the device DTS.

However, in kernel 6.1 commit ("ARM: dts: qcom-*: rep [... cut]
PR #15512 (2024-05-17): [23.05] ipq806x: mr42/mr52: fix PCIe ports
All three PCIe ports are reported non working on Meraki MR42/MR52 boards since kernel 6.1 with the issue of PCIe PHY link never coming up thus no WLAN cards are available on the boards.

After debugging it seems that PCIe worked on 5.15 and older purely by accident as device DTS was using /delete-property/ perst-gpios; in each of the 3 PCIe nodes but there was no "perst-gpios" property in the SoC DTSI as it was still using the older "perst-gpio" property so it was not getting removed from the device DTS.

However, in kernel 6.1 commit ("ARM: dts: qcom [... cut]