openwrt/openwrt #15412: "Smp_affinity" script does not work in the stable and development versions of the QNAP QHora-301W router

shevchenko1987, 2024-05-07:
### Describe the bug

The "Smp_affinity" script does not work in the stable and development versions of the QNAP QHora-301W router.

No core response to the above-mentioned scripts.

The only one that works properly is this one: https://forum.openwrt.org/t/dynalink-dl-wrx36-askey-rt5010w-ipq8072a-technical-discussion/110454/1708

### OpenWrt version

23.05.x / Snapshot / Master

### OpenWrt release

23.05.x / Snapshot / Master

### OpenWrt target/subtarget

ipq807x

### Device

QNAP QHora-301W

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

shevchenko1987, 2024-05-07:
> Invalid Version reported. `23.05.x / Snapshot / Master` Is this from a clean repository?

Yes

pwned-pixel, 2024-05-10:
A bit less information. If you claim it is not working how do you determine it is not working? Regarding this script it is printing its output in to logread. There should be sth. like:

```
Thu May  9 19:26:47 2024 kern.info kernel: [    0.013465] SMP: Total of 4 processors activated.
Thu May  9 19:26:51 2024 user.notice smp_affinity: Pinning IRQ(74) reo2host-destination-ring1 to CPU 0
Thu May  9 19:26:51 2024 user.notice smp_affinity: Pinning IRQ(75) reo2host-destination-ring2 to CPU 1
Thu May  9 19:26:51 2024 user.notice smp_affinity: Pinning IRQ(76) reo2host-destination-ring3 to CPU 2
Thu May  9 19:26:51 2024 user.notice smp_affinity: Pinning IRQ(77) reo2host-destination-ring4 to CPU 3
Thu May  9 19:26:51 2024 user.notice smp_affinity: Pinning IRQ(56) wbm2host-tx-completions-ring1 to CPU 1
Thu May  9 19:26:51 2024 user.notice smp_affinity: Pinning IRQ(61) wbm2host-tx-completions-ring2 to CPU 2
Thu May  9 19:26:51 2024 user.notice smp_affinity: Pinning IRQ(64) wbm2host-tx-completions-ring3 to CPU 3
Thu May  9 19:26:51 2024 user.notice smp_affinity: Pinning IRQ(68) ppdu-end-interrupts-mac1 to CPU 1
Thu May  9 19:26:51 2024 user.notice smp_affinity: Pinning IRQ(72) ppdu-end-interrupts-mac2 to CPU 2
Thu May  9 19:26:51 2024 user.notice smp_affinity: Pinning IRQ(70) ppdu-end-interrupts-mac3 to CPU 3
Thu May  9 19:26:51 2024 user.notice smp_affinity: Pinning IRQ(37) edma_txcmpl              to CPU 3
Thu May  9 19:26:51 2024 user.notice smp_affinity: Pinning IRQ(38) edma_rxfill              to CPU 3
Thu May  9 19:26:51 2024 user.notice smp_affinity: Pinning IRQ(39) edma_rxdesc              to CPU 3
Thu May  9 19:26:51 2024 user.notice smp_affinity: Pinning IRQ(40) edma_misc                to CPU 3
Thu May  9 19:26:51 2024 user.notice smp_affinity: Pinning IRQ(43) xhci-hcd:usb1            to CPU 2
Thu May  9 19:26:51 2024 user.notice smp_affinity: Pinning IRQ(44) xhci-hcd:usb3            to CPU 2
```

Beside that this script is only available within master/main and not as you claim within 23.05. So for 23.05 it cannot work because it is not there.

If it is not working it could be a runtime issue. But the script is starting very late already. As a workaround respectivley for testing you could also call the script another time within rc.local or manually and see if the logread output is there then. Another option is to the START=99 within the init.d script.

shevchenko1987, 2024-05-10:
This script: https://forum.openwrt.org/t/dynalink-dl-wrx36-askey-rt5010w-ipq8072a-technical-discussion/110454/1708 makes the load on the CPU cores even out.

pwned-pixel, 2024-05-10:
23.05 is not supported. So there is no bug for 23.05. If you want to discuss any bug you have to do it for a version where the script is in and provide proper log output (see above).

Despite that: Htop is telling you very little bout interrupt usage. You would have to monitor the interrupts to judge about if the script works or not. So atm I assume that the script is working correctly even for 23.05.

If you want to see cpu core usage regarding this interrupts choosen within this script you have to create load for them. Because his load is different from a load caused by a program shown within htop.

So open a shell put in:

`watch -n0.1 --no-title cat /proc/interrupts`

and start e.g. streaming a youtube video over wifi. You should see a "load" there which is spread over the cores assigned by the script.