openwrt/openwrt #15413: Bootloop kernel 6.6 on Cudy WR3000

Underworld6, 2024-05-07:
### Describe the bug

Installing the latest snapshot with kernel 6.6 on Cudy Wr3000 results into a bootloop.⁷

### OpenWrt version

Snapshot from 07.05.24

### OpenWrt release

795c6d66e39f

### OpenWrt target/subtarget

Mediatek/filogic

### Device

Cudy WR3000 v1

### Image kind

Official downloaded image

### Steps to reproduce

Flash new image trough luci , as well as attended sysupgrade

### Actual behaviour

Roter tries to boot but failea. Wifi leds quickly lid and then the router reboots.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Underworld6, 2024-05-07:
Snapshots from https://firmware-selector.openwrt.org/.

yogo1212, 2024-05-09:
"Computer says no" :-|
Entirely reasonable bug report marked as invalid by bot.
I'm not sure these automations really make life on earth better, at the end of the day.

Maybe it's because of the uppercase `Mediatek`?
I don't suppose the bot reacts to edits but you could try changing it to `mediatek/filogic`.
Otherwise, this is some extra work for the non-existent secretary for Github issues.

I'll try to get this confirmed.

ynezz, 2024-05-09:
Is it really that hard to actually read the form? 
[image]

ynezz, 2024-05-09:
> Roter tries to boot but failea. Wifi leds quickly lid and then the router reboots.

Ok, any chance to get somehow output from serial console during this occasion?

Underworld6, 2024-05-09:
Sorry, first time using github. I am using old snapshot OpenWrt SNAPSHOT r25637-8753022aea / LuCI Master git-24.077.26240~f2a4910 and I am adding luci with the firmware selector. When trying to install ( the snapshot version from 07.05.24 - what I marked was the snapshot naming convention extention) trough Luci- system- flash image I get bootloop. The same result with attended systemupdate. I can not give the correct 6.6 snapshot version, as I can not log to the router and didn't pay attention when upgrading it. I was able to recover via failsafe. I can buy UART, but I could not find any diagram where to connect it.

yogo1212, 2024-05-09:
@Underworld6 
It sounds like you might not have done this before.
Best if you use a voltage meter to help identify the correct pins or you'll be frying that new UART adapter.
If you want to solder on some pins for a reliable connection, you'll be taking everything apart - including the heatsink, which means you'll want to reapply the thermal paste.
That involves cleaning the components, likely using alcohol and avoiding static discharges.

What I mean to say is: It's a nice offer but please only do it if you like to tinker and understand the risk.

I have UART pins soldered on already and I'll to try to do get this done during the weekend.

Underworld6, 2024-05-10:
> @Underworld6 Thanks for reporting the details! The whole process can be a bit intimidating and making mistakes is a normal thing.
> 
> It sounds like you might not have used UART before. Best you use a voltage meter to help identify the correct pins or you'll be frying that new UART adapter. If you want to solder on some pins for a reliable connection, you'll be taking everything apart - including the heatsink, which means you'll want to apply some new thermal paste. That involves cleaning the components, likely using alcohol and avoiding static discharges.
> 
> What I mean to say is: It's a nice offer but please only do it if you like to tinker and understand the risk.
> 
> I have UART pins soldered on already and I'll to try to do get this done during the weekend (assuming I get the crash as well).

Well, I need to buy a soldering iron as well in that case :) Let's not experiment for now.
Thanks
Kind regards
K

Underworld6, 2024-05-10:
> Is it really that hard to actually read the form? [...]

It is hard, when you do not have access to your router. Pls look at the bug reported. Yes, I could find the correct details, but as you asked mentioned it was hard and it was manual.

yogo1212, 2024-05-10:
With the standard image, I don't get a crash.

I saw your edit on your previous comment.
Just to be sure:
You added `bridger` to the list of 'installed packages' on [this page](https://firmware-selector.openwrt.org/?version=23.05.3&target=mediatek%2Ffilogic&id=cudy_wr3000-v1)?

Underworld6, 2024-05-10:
Sorry for the confusion, I build up a snapshot image from the same page, but installed bridger from luci-syatem-sofware.
10x
Kr.
K

Underworld6, 2024-05-10:
Just to be, sure. I have added luci in customise packages for a snapshot version and than build it ip, not the latest official. Than I've installed bridger trough luci.
Kr
K

yogo1212, 2024-05-14:
@Underworld6 would you mind just sending me the image?

Underworld6, 2024-05-15:
[openwrt-795c6d66e39f-mediatek-filogic-cudy_wr3000-v1-squashfs-sysupgrade.zip](https://github.com/openwrt/openwrt/files/15322267/openwrt-795c6d66e39f-mediatek-filogic-cudy_wr3000-v1-squashfs-sysupgrade.zip)
Please see it attached. Thanks @yogo1212

Underworld6, 2024-05-25:
> @Underworld6 would you mind just sending me the image?

Hi Leon,
Do you think, it is safe to upgrade to the latestst snapshot, now?
.It took me 20-30 min. to put the router in safefail mode last time, in order to recover:) 
10x
Kr
K

yogo1212, 2024-05-27:
@Underworld6 I would assume it is.

If you want confirmation from me, you'll have to wait a bit - sorry!
My wife and I are occupied with a newborn :wink:

Underworld6, 2024-05-27:
No problem. Congrats to your new baby!  Wishing you a few extra hours of sleep and all the joy!

Underworld6, 2024-06-04:
I have resolved my issue and it was me doing wrong things. 
The problem was that I was choosing "keep current settings during the snapshot update from kernel 6.1 to 6.6". Although it is explicitly said in wiki not to keep the settings, I was confident enough since I was choosing this option for more than 10 years of use OpenWrt and constant updates - nothing will happen - well it happen this time. 
If I keep the 6.1 settings while I upgrade to any snapshot 6.6 the router went into a bootloop. You can resolve this issue by holding reset button, while powering your router on, or simply wait for the boot loop ( please be advised that you need to clear your cache or use your browser "incognito" mode as you may not reach 192.168.1.1 ( it took me at least 2 hours to understand that).
I believe that I went into a bootloop as I am using the router as a WDS Dumb AP with 802.11rk and v. I remember now that when I did my backup, I replaced wpad-mbedtls basic with wpad-mbedtls. However as it consumed a lot of space I removed it. 
The tricky part (if you look at the history of this issue ) is that I never added/ replaced wpad-mbedtls basic, while I was trying to upgrade to kernel 6.6, hence I got a bootloop.
Steps to resolve the issue:
1. Upgraded to latest snapshot without keeping the current settings - all went fine
2. I uploaded via WinSCP the network and wireless configs - all went fine, but the exception that the "Master" wirelsess networks reported that hey are disabled. WDS wireless clients were fine
3. I looked at the setup and I realized that I wam missing the full wpad package
4. I installed it and - all fine
5. Upgraded to the same snapshot version via attendedsysupgrade to keep the setup - all fine

I hope this will help someone and again thanks @yogo1212 
Kind regards
Krassi

Underworld6, 2024-06-04:
It is working fine for the moment.

closed 2024-06-04 by Underworld6