openwrt/openwrt #15423: base-files: LEDs in devices with external LED controllers still do not work in 23.05. should we backport?

Lanchon, 2024-05-08:
### Describe the bug

for many devices (i know of WHW03 V1 and V2, @robimarko [knows of others](https://github.com/openwrt/openwrt/issues/13803#issuecomment-1782923670)) with discreet, external LED controllers (ie, not GPIO LEDs), openwrt was unable to control the LEDs.

the bug was reported [here](https://github.com/openwrt/openwrt/issues/13803). i can confirm this is fixed in master. i believe the fix is [this commit](https://github.com/openwrt/openwrt/commit/e814acc59948943c776ac319a348f72c0d19f33c).

however the 23.05 branch does not include the fix and affected devices still cannot control the LEDs. with the next openwrt release probably not happening before 2025, it is worth considering cherry picking this fix.

but it is safe? could the change break any other device?

if desired, i can test that the suspected fix actually works on a WHW03 V1 on 23.05.

please let me know if i should do a PR.

CC: @DragonBluep @chunkeey @testuser7

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys WHW03 V2 (Velop)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Ansuel, 2024-05-09:
@Lanchon that commit is already in 23.05

Lanchon, 2024-05-09:
@Ansuel thank you, you are right!

i don't know why the github's UI isn't picking up this particular cherrypick, but it is here:
https://github.com/openwrt/openwrt/commit/7606dac661f60d378d2cf42c6434811e6234f252

WHW03 V2 then needs some cherrypicks too, to compensate for this change. i'll look into that.

closed 2024-05-09 by Lanchon

Lanchon, 2024-05-10:
cherrypicks are here: https://github.com/openwrt/openwrt/pull/15438