openwrt/openwrt #15425: wifi is not associated openwrt

omid2018vpn, 2024-05-08:
### Describe the bug

hi , when my router ( nanopi r1 ) reboot , wifi is not associated and radio is not active
When I edit and want to enable it, it is still disabled and it doesn't work. I tested all versions of 23, I also tested the snapshot, but it didn't work. With the help of this 
command 
rm /etc/config/wireless
wifi config
I delete the wifi information, then I set it again, it works again when it is rebooted.  There is a problem

### OpenWrt version

23.05.3

### OpenWrt release

(r23809-234f1a2efa)

### OpenWrt target/subtarget

 sunxi/cortexa7

### Device

FriendlyARM NanoPi R1

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-05-08:
check `mount` on device if /overlay is backed by real device.

aress31, 2024-06-22:
I am currently encountering the exact same problem, here are some info:

```bash
root@OpenWrt:~# cat /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd00:89a5:d5f1::/48'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth0'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.137.100'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option gateway '192.168.137.1'
        option dns '1.1.1.1 8.8.8.8'

config interface 'wan'
        option proto 'dhcp'
```

```bash
root@OpenWrt:~# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/axi/1001100000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
        option band '5g'
        option htmode 'HT20'
        option cell_density '0'

config wifi-iface 'wifinet0'
        option device 'radio0'
        option mode 'sta'
        option network 'wan'
        option ssid 'secret'
        option bssid 'secret'
        option encryption 'psk2'
        option key 'secret'
```

```bash
root@OpenWrt:~# cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 108.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x4
CPU part        : 0xd0b
CPU revision    : 1

processor       : 1
BogoMIPS        : 108.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x4
CPU part        : 0xd0b
CPU revision    : 1

processor       : 2
BogoMIPS        : 108.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x4
CPU part        : 0xd0b
CPU revision    : 1

processor       : 3
BogoMIPS        : 108.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x4
CPU part        : 0xd0b
CPU revision    : 1

Revision        : d04170
Serial          : 4d228dccca079558
Model           : Raspberry Pi 5 Model B Rev 1.0
```

```bash
root@OpenWrt:~# uname -a
Linux OpenWrt 6.6.34 #0 SMP Fri Jun 21 18:27:53 2024 aarch64 GNU/Linux
```

[image]

[image]

Any help?

brada4, 2024-06-22:
https://openwrt.org/faq/no_space_left_on_device

linxcow, 2024-08-16:
Tested on 23.05.4 :
I have the same problem on nanopi r1 (version without flash.)
Sometimes after reboot no wifi. 
Also setting to "sae-mixed" never possible to connect to radio from any device.