openwrt/openwrt #15427: wget throws a "SSL error: SSL - Bad input parameters to function"

McGiverGim, 2024-05-09:
### Describe the bug

I've upgraded to latest snapshot in one of my AX3600. Since then, for some URLs, I'm getting this error:
```
root@router-despacho:/tmp# wget https://github.com/jerrykuku/luci-theme-argon/releases/download/v2.3.1/luci-theme-argon_2.3.1_all.ipk
Downloading 'https://github.com/jerrykuku/luci-theme-argon/releases/download/v2.3.1/luci-theme-argon_2.3.1_all.ipk'
Connecting to 140.82.121.4:443
Redirected to /github-production-release-asset-2e65be/170895520/63bb58df-abc2-4421-aed8-0ace3f9aae17?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240509%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240509T061938Z&X-Amz-Expires=300&X-Amz-Signature=35dace05b00ba78ae0b67147c950cb21d38f6c87b26fea29e3eaa66cabcddc24&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=170895520&response-content-disposition=attachment%3B%20filename%3Dluci-theme-argon_2.3.1_all.ipk&response-content-type=application%2Foctet-stream on objects.githubusercontent.com
SSL error: SSL - Bad input parameters to function
Connection error: Connection failed
```
The same command works perfectly in my other AX3600, that contains a snapshot from 2/3 weeks ago.

### OpenWrt version

r26235-0d74b2a1e5

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

I suppose that:
- Install latest Snapshot
- Issue a `wget https://github.com/jerrykuku/luci-theme-argon/releases/download/v2.3.1/luci-theme-argon_2.3.1_all.ipk` command

### Actual behaviour

It throws an error: `SSL error: SSL - Bad input parameters to function`

### Expected behaviour

Download the file

### Additional info

As I said, my other AX3600 from a two or three week ago can download the file without problem. I suppose it is some incompatibility between packages.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-05-09:
`wget --version` ? ie if it is wget or ufetch

McGiverGim, 2024-05-09:
It seems they are totally different, between the one that works and the one that not (maybe some depedency installed it)?
The one that works:
```
root@router-comedor:~# wget --version
GNU Wget 1.24.5 built on linux-gnu.

-cares +digest -gpgme +https +ipv6 -iri +large-file -metalink -nls
+ntlm +opie -psl +ssl/openssl
...
```
The one that not:
```
root@router-despacho:~# wget --version
wget: unrecognized option: version
```
It seems the one that does not work is `uclient-fetch`. Looking at the software list appears as `2024.04.19~e8780fa7-r1`

The one that works I think is using `wget-ssl`. I tried to install it in the one that does not work and it works. So the prolem is with `uclient-fetch`. Is that expected?

brada4, 2024-05-09:
You need ca-certificates to connect to non-default sites.

McGiverGim, 2024-05-09:
I've tried to install the ca-certificates package, but the error remains. Do I need to do something more? 
I've executed this command since two years ago without problem.

rany2, 2024-05-09:
You need `wget-ssl` package, there are two variants: `wget` (without SSL support) and `wget-ssl`

brada4, 2024-05-09:
uclient-fetch - 2023-04-13-007d9454-1
certainly works as in 23..3

rany2, 2024-05-09:
I think it's a uclient-fetch bug, it worked on the other router because it was using gnu-wget. I can't get it to load that URL either with uclient-fetch (mbedtls).

McGiverGim, 2024-05-09:
I think I can give more data:
- I'm using this `wget` commando since more than a year ago, each 2 or 3 weeks I update to latest snapshot.
- It has worked perfectly, until now.
- I'm almost sure that in earlier snapshot, contained the same `uclient-fetch` version, and it worked.
So I think it must be a problem of maybe the library that it used (mbedtls?). It seems some incompatibility by the error message.

rany2, 2024-05-09:
I understand the cause now. I'll CC @nbd168 because it seems he did some recent work with uclient and SSL.

The issue occurs when uclient-fetch tries to redirect over SSL from one host to another. So a redirect that works like this: `https://rany.eu.org/hi` -> `https://rany.eu.org/hi2` will work fine but `https://rany.eu.org/subdomain` to `https://subdomain.rany.eu.org` does not.

kuleshov-aleksei, 2024-09-13:
Got the same error on freshly installed snapshot: `r27350-c4a9265160` for Bananapi BPi-R4

```
> opkg update
Downloading https://downloads.sourceforge.net/project/v2raya/openwrt/aarch64_cortex-a53/Packages.gz
SSL error: SSL - Bad input parameters to function

> wget https://downloads.sourceforge.net/project/v2raya/openwrt/v2raya.pub -O /etc/opkg/keys/94cc2a834fb0aa03
Downloading 'https://downloads.sourceforge.net/project/v2raya/openwrt/v2raya.pub'
Connecting to 204.68.111.105:443
Redirected to /project/v2raya/openwrt/v2raya.pub?viasf=1 on kumisystems.dl.sourceforge.net
SSL error: SSL - Bad input parameters to function
Connection error: Connection failed
```

Seems like I have the same version of uclient-fetch as author of this thread:

```
> opkg install wget
Package uclient-fetch (2024.04.19~e8780fa7-r1) installed in root is up to date.
```

And this has nothing to do with self-signed ssl certificates. For me **wget-ssl fixed the problem** `opkg install wget-ssl`

_commenting for visibility_

McGiverGim, 2024-10-25:
This seems fixed with latest snapshot. I close this. Thanks!

closed 2024-10-25 by McGiverGim