openwrt/openwrt #15428: Kernel panic - not syncing: PSGMII work is unstable !!! Repeated recalibration attempts did not help(0x3) !

pygoubet, 2024-05-09:
### Describe the bug

Since my Linksys WHW03 V2 APs use the SNAPSHOT release, they can only boot successfully if a cable is connected to one of the 2 ethernet ports. This is not very convenient because they are member of a BATMAN mesh ; only the gateway is connected by wire.

I have switched to SNAPSHOT release to benefit from the 5 GHz extra channel brought by 1c32cee348f54e9f64b5928ee1a1efe491eec137

### OpenWrt version

r26186-064d4a8083

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys WHW03 V2 (Velop)

### Image kind

Official downloaded image

### Steps to reproduce

The Sysupgrade was generated from the [firmware selector from Linksys WHW03 V2](https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=ipq40xx%2Fgeneric&id=linksys_whw03v2) with the following installed packages:
```
ath10k-board-qca4019 ath10k-firmware-qca4019-ct ath10k-firmware-qca9888-ct base-files busybox ca-bundle dnsmasq dropbear firewall4 fstools kmod-ath10k-ct kmod-bluetooth kmod-gpio-button-hotplug kmod-leds-gpio kmod-leds-pca963x kmod-nft-offload kmod-spi-dev kmod-usb-dwc3 kmod-usb-dwc3-qcom kmod-usb3 libc libgcc libustream-mbedtls logd luci mtd netifd nftables odhcp6c odhcpd-ipv6only opkg procd procd-seccomp procd-ujail uboot-envtools uci uclient-fetch urandom-seed urngd wpad-mesh-mbedtls batctl-full luci-proto-batman-adv kmod-batman-adv luci-app-advanced-reboot
```

### Actual behaviour

The kernel crashes and reboot the system:
```
...
[    4.827456] init: - preinit -
[    6.081787] ipqess-edma c080000.ethernet eth0: configuring for fixed/internal link mode
[    6.082555] qca8k-ipq4019 c000000.switch lan: configuring for phy/psgmii link mode
[    6.088630] qca8k-ipq4019 c000000.switch: PSGMII calibration!
[    6.096354] ipqess-edma c080000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   10.896528] qca8k-ipq4019 c000000.switch: PSGMII work is unstable !!! Let's try to wait a bit ... 10
[   16.269075] qca8k-ipq4019 c000000.switch: PSGMII work is unstable !!! Let's try to wait a bit ... 20
[   22.669048] qca8k-ipq4019 c000000.switch: PSGMII work is unstable !!! Let's try to wait a bit ... 30
[   30.029132] qca8k-ipq4019 c000000.switch: PSGMII work is unstable !!! Let's try to wait a bit ... 40
[   38.429107] qca8k-ipq4019 c000000.switch: PSGMII work is unstable !!! Let's try to wait a bit ... 50
[   47.789063] qca8k-ipq4019 c000000.switch: PSGMII work is unstable !!! Let's try to wait a bit ... 60
[   58.589104] qca8k-ipq4019 c000000.switch: PSGMII work is unstable !!! Let's try to wait a bit ... 70
[   70.109001] qca8k-ipq4019 c000000.switch: PSGMII work is unstable !!! Let's try to wait a bit ... 80
[   82.909048] qca8k-ipq4019 c000000.switch: PSGMII work is unstable !!! Let's try to wait a bit ... 90
[   96.349075] qca8k-ipq4019 c000000.switch: PSGMII work is unstable !!! Let's try to wait a bit ... 100
[  106.711041] Kernel panic - not syncing: PSGMII work is unstable !!! Repeated recalibration attempts did not help(0x3) !
[  106.711086] CPU2: stopping
[  106.711096] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G           O       6.6.30 #0
[  106.711110] Hardware name: Generic DT based system
[  106.711124]  unwind_backtrace from show_stack+0x10/0x14
[  106.711161]  show_stack from dump_stack_lvl+0x40/0x4c
[  106.711188]  dump_stack_lvl from do_handle_IPI+0xec/0x120
[  106.711206]  do_handle_IPI from ipi_handler+0x18/0x20
[  106.711221]  ipi_handler from handle_percpu_devid_irq+0x78/0x134
[  106.711238]  handle_percpu_devid_irq from generic_handle_domain_irq+0x28/0x38
[  106.711259]  generic_handle_domain_irq from gic_handle_irq+0x74/0x88
[  106.711280]  gic_handle_irq from generic_handle_arch_irq+0x34/0x44
[  106.711299]  generic_handle_arch_irq from call_with_stack+0x18/0x20
[  106.711325]  call_with_stack from __irq_svc+0x80/0x98
[  106.711347] Exception stack(0xc109df58 to 0xc109dfa0)
[  106.711357] df40:                                                       00000003 00000001
[  106.711370] df60: 00012724 40000000 00000002 c0e04f68 c1094000 c0e04fa4 00000000 00000000
[  106.711383] df80: 00000000 00000000 00000017 c109dfa8 c0a4a678 c0a4ade0 60000013 ffffffff
[  106.711391]  __irq_svc from default_idle_call+0x2c/0x30
[  106.711410]  default_idle_call from do_idle+0x1d8/0x228
[  106.711430]  do_idle from cpu_startup_entry+0x28/0x2c
[  106.711443]  cpu_startup_entry from secondary_start_kernel+0x144/0x148
[  106.711459]  secondary_start_kernel from 0x80301474
[  106.711478] CPU0: stopping
[  106.711488] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G           O       6.6.30 #0
[  106.711500] Hardware name: Generic DT based system
[  106.711508]  unwind_backtrace from show_stack+0x10/0x14
[  106.711535]  show_stack from dump_stack_lvl+0x40/0x4c
[  106.711562]  dump_stack_lvl from do_handle_IPI+0xec/0x120
[  106.711579]  do_handle_IPI from ipi_handler+0x18/0x20
[  106.711595]  ipi_handler from handle_percpu_devid_irq+0x78/0x134
[  106.711610]  handle_percpu_devid_irq from generic_handle_domain_irq+0x28/0x38
[  106.711630]  generic_handle_domain_irq from gic_handle_irq+0x74/0x88
[  106.711649]  gic_handle_irq from generic_handle_arch_irq+0x34/0x44
[  106.711668]  generic_handle_arch_irq from call_with_stack+0x18/0x20
[  106.711693]  call_with_stack from __irq_svc+0x80/0x98
[  106.711715] Exception stack(0xc0e01f28 to 0xc0e01f70)
[  106.711727] 1f20:                   00000003 00000001 00030ea4 40000000 00000000 c0e04f68
[  106.711740] 1f40: c0e08700 c0e04fa4 00000000 00000000 c0e04f08 00000000 00000017 c0e01f78
[  106.711750] 1f60: c0a4a678 c0a4ade0 60000013 ffffffff
[  106.711757]  __irq_svc from default_idle_call+0x2c/0x30
[  106.711774]  default_idle_call from do_idle+0x1d8/0x228
[  106.711790]  do_idle from cpu_startup_entry+0x28/0x2c
[  106.711804]  cpu_startup_entry from kernel_init+0x0/0x12c
[  106.711820]  kernel_init from arch_post_acpi_subsys_init+0x0/0x8
[  106.711849] CPU1: stopping
[  106.711860] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G           O       6.6.30 #0
[  106.711873] Hardware name: Generic DT based system
[  106.711880]  unwind_backtrace from show_stack+0x10/0x14
[  106.711908]  show_stack from dump_stack_lvl+0x40/0x4c
[  106.711935]  dump_stack_lvl from do_handle_IPI+0xec/0x120
[  106.711952]  do_handle_IPI from ipi_handler+0x18/0x20
[  106.711967]  ipi_handler from handle_percpu_devid_irq+0x78/0x134
[  106.711983]  handle_percpu_devid_irq from generic_handle_domain_irq+0x28/0x38
[  106.712003]  generic_handle_domain_irq from gic_handle_irq+0x74/0x88
[  106.712022]  gic_handle_irq from generic_handle_arch_irq+0x34/0x44
[  106.712039]  generic_handle_arch_irq from call_with_stack+0x18/0x20
[  106.712063]  call_with_stack from __irq_svc+0x80/0x98
[  106.712084] Exception stack(0xc109bf58 to 0xc109bfa0)
[  106.712094] bf40:                                                       00000003 00000001
[  106.712107] bf60: 00005a2c 40000000 00000001 c0e04f68 c1091800 c0e04fa4 00000000 00000000
[  106.712119] bf80: 00000000 00000000 00000001 c109bfa8 c0a4a678 c0a4ade0 60000013 ffffffff
[  106.712127]  __irq_svc from default_idle_call+0x2c/0x30
[  106.712144]  default_idle_call from do_idle+0x1d8/0x228
[  106.712161]  do_idle from cpu_startup_entry+0x28/0x2c
[  106.712174]  cpu_startup_entry from secondary_start_kernel+0x144/0x148
[  106.712189]  secondary_start_kernel from 0x80301474
[  107.098927] Rebooting in 1 seconds..
```

### Expected behaviour

Once a cable is connected, the PSGMII calibration works:
```
...
[    4.827224] init: - preinit -
[    6.083059] ipqess-edma c080000.ethernet eth0: configuring for fixed/internal link mode
[    6.083843] qca8k-ipq4019 c000000.switch lan: configuring for phy/psgmii link mode
[    6.089903] qca8k-ipq4019 c000000.switch: PSGMII calibration!
[    6.097713] ipqess-edma c080000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
Press the [f] key and hit [enter] to enter failsafe mode
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
...
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Ansuel, 2024-05-09:
Do we have GPL of that router?

pygoubet, 2024-05-10:
It is worth noting that, with no cable connected, if I interrupt the U-Boot autoboot and specifically enter `run bootcmd` command, then PSGMII calibration succeeds and OpenWrt works as expected.
Albeit, with autoboot, the kernel always crashes.
A `bootdelay` to 60 seconds does not help the PSGMII calibration to work.

robimarko, 2024-05-10:
Most likely because interrupting U-Boot has caused it to initialize the networking

testuser7, 2024-05-14:
> Do we have GPL of that router?

GPL source code can be downloaded here: https://downloads.linksys.com/support/assets/gpl/WHW03V2_2.1.19.209783.tgz

pygoubet, 2024-05-19:
> Most likely because interrupting U-Boot has caused it to initialize the networking

Good point @robimarko!
I have updated the default `bootcmd` parameter to run a `ping` command which makes U-Boot to initialize the network and so the Linux Kernel to not fail on `PSGMII calibration`.

* workaround
```bash
fw_setenv bootcmd 'ping $ipaddr ; if test $auto_recovery = no; then bootipq; elif test $boot_part = 1; then run bootpart1; else run bootpart2; fi'
```

* U-Boot output
```
...
Hit any key to stop autoboot:  0 
MAC0 addr:e8:9f:80:64:6:2e
PHY ID1: 0x4d
PHY ID2: 0xd0b2
ipq40xx_ess_sw_init done
eth0
eth0 PHY0 Down Speed :10 Half duplex
eth0 PHY1 Down Speed :10 Half duplex
eth0 PHY2 Down Speed :10 Half duplex
eth0 PHY3 Down Speed :10 Half duplex
eth0 PHY4 Down Speed :10 Half duplex
ping failed; host 192.168.1.11 is not alive

NAND read: device 0 offset 0xa800000, size 0xd00000
 13631488 bytes read: OK
## Booting kernel from FIT Image at 81000000 ...
...
[    4.804045] init: - preinit -
[    6.047466] ipqess-edma c080000.ethernet eth0: configuring for fixed/internal link mode
[    6.048263] qca8k-ipq4019 c000000.switch lan: configuring for phy/psgmii link mode
[    6.054344] qca8k-ipq4019 c000000.switch: PSGMII calibration!
[    6.062019] ipqess-edma c080000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
Press the [f] key and hit [enter] to enter failsafe mode
...
```

Ansuel, 2024-05-19:
Thea no... That is unacceptable and we need to find the real cause... Sadly
from what I can see the provided GPL doesn't have U-Boot source

Ansuel, 2024-05-19:
@pygoubet btw you said snapshot... it did work previously? cn you reference the working image?

pygoubet, 2024-05-19:
OpenWrt 23.05.3 (r23809-234f1a2efa) kernel 5.15.150 starts with no workaround (network initialized by U-Boot or ethernet cable plugged). I suppose the problem I am encountering has been introduced by Kernel 6.6 and especially by [706-net-dsa-qca8k-add-IPQ4019-built-in-switch-support.patch](https://github.com/openwrt/openwrt/blob/main/target/linux/ipq40xx/patches-6.6/706-net-dsa-qca8k-add-IPQ4019-built-in-switch-support.patch).

Ansuel, 2024-05-19:
Ok thanks for the info

robimarko, 2024-05-22:
I have pushed a change so that kernel will not panic if PSGMII fails to calibrate anymore as its not a fatal error, so it can actually be debugged.

pygoubet, 2024-05-23:
As expected, build 'r26387-9a11bc3682' makes the kernel to not panic after `PSGMII work is unstable !!! Repeated recalibration attempts did not help(0x3) !` message.
WLAN interfaces work, so it allows my Linksys device to operate as a BATMAN node.
But plugging a cable afterward on one of ethernet ports does not make the connection to work (link UP but no traffic).

robimarko, 2024-05-23:
I gotta say, I am quite intrigued how is this the only device with this issue, especially since the calibration code did not really change.

Is this board maybe available cheap to buy?

pygoubet, 2024-05-23:
Any action I could to do to help the debugging?

robimarko, 2024-05-23:
> Any action I could do to help the debugging?

You can dump the PSGMII register space, it can be done from regmap debugfs

pygoubet, 2024-05-24:
[failed-calibration-registers.txt](https://github.com/openwrt/openwrt/files/15428206/failed-calibration-registers.txt)
[uboot-ping-workaround-registers.txt](https://github.com/openwrt/openwrt/files/15428207/uboot-ping-workaround-registers.txt)
[wire_connected-registers.txt](https://github.com/openwrt/openwrt/files/15428208/wire_connected-registers.txt)

robimarko, 2024-05-24:
Well, the only difference seems to be the calibration results themselves, the config bits seem to be the same.

Also, a lot of the PSGMII register space is completely undocumented

restena-pyg, 2024-05-27:
A more recent version of the GPL here: https://downloads.linksys.com/support/assets/gpl/WHW03_v2.1.9.215389.tgz
Maybe something to learn from it?

pygoubet, 2024-05-27:
A more recent version of the GPL here: https://downloads.linksys.com/support/assets/gpl/WHW03_v2.1.9.215389.tgz
Maybe something to learn from it?

sasodoma, 2025-05-24:
I have both the V1 and V2 of the WHW03 and on the V2 I experience all of the same symptoms as reported originally. The V1 works without a problem. @robimarko I see that you're from Croatia. I'm from Slovenia, so maybe there's a way I could get a V2 to you for cheap, if you're still interested in this.

dwardor, 2026-10-20:
> 22\. Understandable. I'm currently using the bootcmd ping fix and that seems to work well, so it's not like this is an urgent issue.

@pygoubet & @sasodoma can you adapt and try the dts patch from [[22807#]](https://github.com/openwrt/openwrt/issues/22807#issuecomment-6294818633)

sasodoma, 2026-10-20:
I don't have access to the router right now but I can try it over the weekend.

sasodoma, 2026-10-24:
> 22\. [@sasodoma](https://github.com/sasodoma) can you adapt and try the dts patch from [[22807#]](https://github.com/openwrt/openwrt/issues/22807#issuecomment-6294818633)

@dwardor I can confirm that building v25.12.6 with the DTS patch fixes the problem. I no longer need the bootcmd ping.

dwardor, 2026-10-24:
> 25\. > 22. [@sasodoma](https://github.com/sasodoma) can you adapt and try the dts patch from [[22807#]](https://github.com/openwrt/openwrt/issues/22807#issuecomment-6294818633)
>      
>      
>      [@dwardor](https://github.com/dwardor) I can confirm that building v25.12.6 with the DTS patch fixes the problem. I no longer need the bootcmd ping.

PR was submited : https://github.com/openwrt/openwrt/pull/26094

pygoubet, 2026-10-28:
> 20\. > 22. Understandable. I'm currently using the bootcmd ping fix and that seems to work well, so it's not like this is an urgent issue.
>      
>      
>      [@pygoubet](https://github.com/pygoubet) & [@sasodoma](https://github.com/sasodoma) can you adapt and try the dts patch from [[22807#]](https://github.com/openwrt/openwrt/issues/22807#issuecomment-6294818633)

I also confirm it fixes the problem. The bootcmd ping workaround is no more needed.