openwrt/openwrt #15439: ipq40xx: whw03v2: nearly half of user-available space is being wasted

Lanchon, 2024-05-10:
### Describe the bug

whw03v2, a dual firmware 512 MB device, is ported in a way that only uses the stock space for kernel+rootfs. this results in a rootfs_data of 120 MB. 183 MB of the device end up being unused.

i proposed a simple PR to let advanced users optionally use that 183 MB area as a user-managed partition: https://github.com/openwrt/openwrt/pull/13754

but seemingly that PR was not favored. @Ansuel asked about the possibility of using that space to extend rootfs_data. i said not possible; at the time i didn't knew better.

but now i think it is possible, read on...

```
[    1.284466] 0x000000000000-0x000000100000 : "SBL1"
[    1.291446] 0x000000100000-0x000000200000 : "MIBIB"
[    1.296083] 0x000000200000-0x000000300000 : "QSEE"
[    1.300798] 0x000000300000-0x000000380000 : "CDT"
[    1.305094] 0x000000380000-0x000000580000 : "APPSBL"
[    1.311471] 0x000000580000-0x000000600000 : "ART"
[    1.314994] 0x000000600000-0x000000680000 : "u_env"
[    1.319582] 0x000000680000-0x0000006c0000 : "s_env"
[    1.324016] 0x0000006c0000-0x000000700000 : "devinfo"
[    1.328823] 0x000000700000-0x00000a800000 : "kernel"     : 161 MiB (6 MiB kernel + rootfs)
[    1.490871] 0x000000d00000-0x00000a800000 : "rootfs"     : 155 MiB
[    1.647219] 0x00000a800000-0x000014900000 : "alt_kernel" : 161 MiB (6 MiB kernel + rootfs)
[    1.809863] 0x00000ae00000-0x000014900000 : "alt_rootfs" : 155 MiB
[    1.959552] 0x000014900000-0x000014b00000 : "sysdiag"    :   2 MiB (unused)
[    1.961963] 0x000014b00000-0x000020000000 : "syscfg"     : 181 MiB (stock state, ext4)
```

the extra 183 MiB in the last two partitions could be split in half and the MTD concat driver can be used to expand both rootfs partitions. this would raise each 120 MB rootfs to 200 MB, nearly doubling them.

the downside is that syscfg would be wiped, without there being a backup.

for this reason, i don't think this change would be recommended to the standard whw03v2 port.

-----

however, a `whw03v2-large` port could be added that uses this space in this way.

cross flashing between `whw03v2`, `whw03v2-large`, and stock would be allowed using the factory images. sysupgrade would be allowed only from `whw03v2` to `whw03v2-large` (using the force flag, settings kept).

presumably the advanced user would do a syscfg backup using LuCI before upgrading to large. i would add a mechanism (access to the stock partitions) to allow easy return to stock.

@dangowrt @Ansuel 

so i could borrow a V2 (a month or so from now) and try my hand at this... if you'd like to merge it!

or... you could merge [my other proposal](https://github.com/openwrt/openwrt/pull/13754), that simply allows advanced users to format the extra space and use it to their advantage, if they want.

i don't wan't to waste time on this unless you green light it.

thanks!

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys WHW03 V2 (Velop)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

The13east, 2026-11-07:
did they merge this ? I just flashed the new 25.12 on this router and it still has 122mb of space ! 
that's not enough for passwall2 + xraycore + singbox ! and before a dev look at this and shrugs it off I will ask him to install these packages in a middle of internet blackout with 0 internet connection and data costing upward of 10$/GB ! 
it is important to have every vpn client under the sun on this router cause it's their only hope

Lanchon, 2026-11-13:
you write as if "the devs" owed you work. if you need this done then do it yourself, or collaborate following the established documented rules of the project, or pay for somebody to do it for you.