openwrt/openwrt #15447: Netgear WAX 206 - WAN unavailable - Kernel 6.6.30

gca31, 2024-05-10:
### Describe the bug

Using snapshot r26272-39f595d1d8 (Kernel 6.6.30) the WAN interface is not going up when plugging an ethernet cable.
logread provide this message while booting:
[    2.463546] mt7530-mdio mdio-bus:1f lan1 (uninitialized): PHY [mt7530-0:01] driver [MediaTek MT7531 PHY] (irq=133)
[    2.484031] mt7530-mdio mdio-bus:1f lan2 (uninitialized): PHY [mt7530-0:02] driver [MediaTek MT7531 PHY] (irq=134)
[    2.504301] mt7530-mdio mdio-bus:1f lan3 (uninitialized): PHY [mt7530-0:03] driver [MediaTek MT7531 PHY] (irq=135)
[    2.524506] mt7530-mdio mdio-bus:1f lan4 (uninitialized): PHY [mt7530-0:04] driver [MediaTek MT7531 PHY] (irq=136)
[    2.766061] mt7530-mdio mdio-bus:1f wan (uninitialized): PHY [mdio-bus:07] driver [Generic Clause 45 PHY] (irq=POLL)

ethtool wan provides:
Settings for wan:
        Supported ports: [ ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                2500baseT/Full 
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: No
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                2500baseT/Full 
        Advertised pause frame use: No
        Advertised auto-negotiation: No
        Advertised FEC modes: Not reported
        Speed: 2500Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 7
        Transceiver: external
        Auto-negotiation: off
        Supports Wake-on: d
        Wake-on: d
        Link detected: no

### OpenWrt version

r26272-39f595d1d8

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Netgear WAX206

### Image kind

Official downloaded image

### Steps to reproduce

Download the snapshot and proceed to a fresh install or a sysupgrade on a Netgear WAX206
Device is accessible through lan or uart/tty. But link is not going up on WAN interface (no message through logread or dmesg) when plugging an ethernet cable (tested with 2.5Gbps link or 100Mbps link)

### Actual behaviour

WAN interface not going up

### Expected behaviour

WAN interface going up !

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-05-10:
Duplicate: https://github.com/openwrt/openwrt/issues/15093

gca31, 2024-05-11:
Solved with https://github.com/openwrt/openwrt/pull/15453

closed 2024-05-11 by gca31