openwrt/openwrt #15455: [23.05.3|snapshot] Network chaos when multiple Wifi adapters or SSIDs are used?!

schuettecarsten, 2024-05-11:
### Describe the bug

I have a ath79 based router and installed the "default" package for this device. Now I create an SSID on the device, connect it to the internet and download kmod-mt76x0u. I attach a second wifi device to the USB port. Now I have a device that has two working Wifi adapters, one is used to connect to an AP and one to provide local AP. I use this setup in a camper.

However, when I disable the first Wifi adapter, OpenWRT also disables the second one. When I activate the first device, it does not activate the second one.  Activating and deactivating Wifi devices or single SSIDs causes big chaos on other adapters since about 3 or 4 month. Maybe it is related to some changes in netifd, no idea.

The issues are reproduceable with 23.05.3 and snapshot versions.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

ath79/generic

### Device

ALFA Network R36A

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

sages, 2024-05-11:
https://forum.openwrt.org/t/multiple-bsses-specified-in-bss-config/191177/1 might be related

ascendbeing, 2024-05-31:
i experienced something similar with kirkwood ea3500 on snapshot from a few mo ago (april or late mar) running an ath9k_htc addon card. It would work then stop working. then I'd have to restart it to get it working. and also, my 5ghz AP wasn't working at all. even after i removed the ath9k_htc. i wrote about this in some more depth on the 6.1 status issues thread.