openwrt/openwrt #15457: Kernel crash on __udp_gso_segment

zhaojh329, 2024-05-11:
### Describe the bug

```
Oops#1 Part1
<1>[  726.982328] Unable to handle kernel read from unreadable memory at virtual address 00000000000000c0
<1>[  726.991376] Mem abort info:
<1>[  726.994157]   ESR = 0x0000000096000005
<1>[  726.997891]   EC = 0x25: DABT (current EL), IL = 32 bits
<1>[  727.003191]   SET = 0, FnV = 0
<1>[  727.006231]   EA = 0, S1PTW = 0
<1>[  727.009357]   FSC = 0x05: level 1 translation fault
<1>[  727.014218] Data abort info:
<1>[  727.017084]   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
<1>[  727.022551]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
<1>[  727.027585]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
<1>[  727.032879] user pgtable: 4k pages, 39-bit VAs, pgdp=000000004a1b0000
<1>[  727.039302] [00000000000000c0] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
<0>[  727.047988] Internal error: Oops: 0000000096000005 [#1] SMP
<7>[  727.053540] Modules linked in: pppoe ppp_async option wireguard usb_wwan rndis_host qmi_wwan pppox ppp_generic nft_fib_inet nf_flow_table_inet mt7915e(O) mt76_connac_lib(O) mt76(O) mac80211(O) libchacha20poly1305 ipt_REJECT huawei_cdc_ncm chacha_neon cfg80211(O) cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_addrtype xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet ts_fsm ts_bm slhc poly1305_neon nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_compat nft_chain_nat nf_tables nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_amanda nf_log_syslog nf_flow_table
<7>[  727.053719]  nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount mdio_netlink(O) libcurve25519_generic libcrc32c libchacha iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables compat(O) cdc_wdm cdc_acm br_netfilter asn1_decoder arptable_filter arpt_mangle arp_tables crypto_safexcel fuse sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables
<7>[  727.142343]  nf_reject_ipv6 ifb ip6_udp_tunnel udp_tunnel tun ntfs nls_utf8 nls_iso8859_1 nls_cp437 sha512_arm64 sha1_ce sha1_generic seqiv md5 geniv des_generic libdes authencesn authenc uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd uhci_hcd ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl kmwan(O) ehci_hcd gpio_button_hotplug(O) gl_sdk4_tertf(O) gl_sdk4_black_white_list(O) vfat fat exfat usbcore usb_common aquantia mii gl_sdk4_hw_info(O)
<7>[  727.274331] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G           O       6.6.30 #0
<7>[  727.281705] Hardware name: GL.iNet GL-MT6000 (DT)
<7>[  727.286390] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
<7>[  727.293332] pc : __udp_gso_segment+0x36c/0x558
<7>[  727.297767] lr : __udp_gso_segment+0x33c/0x558
<7>[  727.302193] sp : ffffffc080c8b5b0
<7>[  727.305491] x29: ffffffc080c8b5b0 x28: 0000000000000000 x27: 0000000000040000
<7>[  727.312607] x26: 0000000000000000 x25: 0000000000000000 x24: ffffff800727fe00
<7>[  727.319723] x23: 0000000000000014 x22: 0000000000000000 x21: ffffff800727fe00
<7>[  727.326839] x20: 02020100001d4893 x19: 0000000000000000 x18: 0000000000000070
<7>[  727.333955] x17: ffffffbfbf26a000 x16: ffffffc080c88000 x15: 0000000000000080
<7>[  727.341072] x14: 0000000000000001 x13: 000502010b950104 x12: 01062b09060d300f
<7>[  727.348188] x11: 30000102000102de x10: 3a02021ba163696c x9 : 0000000000000004
<7>[  727.355304] x8 : 0000000000001369 x7 : 0000000000000004 x6 : 00000000000000a2
<7>[  727.362419] x5 : ffffff800235f000 x4 : ffffff800235fec0 x3 : 0000000000000000
<7>[  727.369535] x2 : ffffff800235f000 x1 : 0000000000000126 x0 : ffffff800235f126
<7>[  727.376651] Call trace:
<7>[  727.379083]  __udp_gso_segment+0x36c/0x558
<7>[  727.383164]  udp4_ufo_fragment+0x128/0x15c
<7>[  727.387246]  inet_gso_segment+0x104/0x364
<7>[  727.391248]  skb_mac_gso_segment+0x9c/0xec
<7>[  727.395338]  __skb_gso_segment+0x60/0x144
<7>[  727.399338]  validate_xmit_skb.isra.0+0x110/0x2d0
<7>[  727.404035]  validate_xmit_skb_list+0x34/0x7c
<7>[  727.408382]  sch_direct_xmit+0xf8/0x338
<7>[  727.412204]  __dev_queue_xmit+0x5a8/0xcc0
<7>[  727.416198]  ip_finish_output2+0x228/0x4ec
<7>[  727.420282]  __ip_finish_output+0xa0/0x154
<7>[  727.424364]  ip_finish_output+0x30/0x120
<7>[  727.428272]  ip_output+0x50/0xb8
<7>[  727.431486]  ip_forward_finish+0x6c/0xa4
<7>[  727.435395]  ip_forward+0x32c/0x45c
<7>[  727.438869]  ip_rcv+0xa8/0xcc
<7>[  727.441823]  __netif_receive_skb_one_core+0x44/0x50
<7>[  727.446683]  __netif_receive_skb+0x14/0x5c
<7>[  727.450763]  netif_receive_skb+0x20/0x8c
<7>[  727.454669]  br_pass_frame_up+0x100/0x1b0
<7>[  727.458665]  br_handle_frame_finish+0x474/0x58c
<7>[  727.463178]  br_handle_frame+0x374/0x424
<7>[  727.467086]  __netif_receive_skb_core.constprop.0+0x4b8/0xda4
<7>[  727.472812]  __netif_receive_skb_list_core+0xd4/0x1e4
<7>[  727.477845]  netif_receive_skb_list_internal+0x1bc/0x27c
<7>[  727.483138]  napi_complete_done+0x64/0x18c
<7>[  727.487217]  gro_cell_poll+0x80/0xa0
<7>[  727.490781]  __napi_poll+0x34/0x170
<7>[  727.494254]  net_rx_action+0x12c/0x260
<7>[  727.497986]  __do_softirq+0xf8/0x25c
<7>[  727.501547]  ____do_softirq+0xc/0x14
<7>[  727.505107]  call_on_irq_stack+0x24/0x34
<7>[  727.509014]  do_softirq_own_stack+0x18/0x24
<7>[  727.513180]  irq_exit_rcu+0xb4/0xcc
<7>[  727.516655]  el1_interrupt+0x34/0x4c
<7>[  727.520218]  el1h_64_irq_handler+0x14/0x1c
<7>[  727.524299]  el1h_64_irq+0x68/0x6c
<7>[  727.527685]  default_idle_call+0x24/0x34
<7>[  727.531592]  do_idle+0xa8/0xf4
<7>[  727.534636]  cpu_startup_entry+0x34/0x38
<7>[  727.538545]  cpu_die_early+0x0/0x84
<7>[  727.542019]  __secondary_switched+0x6c/0x70
<0>[  727.546191] Code: 79000813 3707fad6 f9400313 f9406302 (f9406260) 
<4>[  727.552264] ---[ end trace 0000000000000000 ]---
```

### OpenWrt version

r26272+26-39f595d1d8

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Ansuel, 2024-05-11:
@nbd168 maybe related to recent changes for gro ?

uosbox, 2024-06-03:
[image]
我也遇到和楼主相同的错误
内核版本也是6.6.30
内核崩溃的时间不固定，目前也不知道什么原因，之前6.1的内核没遇到这样的情况
我的硬件时rk3399 4G+64G

ErwanMAS, 2024-08-17:
look at the last version of Snapshot kernel ( https://github.com/openwrt/openwrt/commit/17f07040e7808a0c9cba2afc0e0b22e5589da133 ) .

easyteacher, 2024-09-26:
Latest snapshot at https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=23ac1ad9515588321970985859b2ee0cc8243f91. The crash still exists.

```
Thu Sep 26 11:15:20 2024 kern.warn kernel: [   51.685447] Ignoring NSS change in VHT Operating Mode Notification from 0c:ec:8d:a3:19:9b with invalid nss 2
Thu Sep 26 11:15:20 2024 kern.alert kernel: [  581.356974] CPU 2 Unable to handle kernel paging request at virtual address 000000a4, epc == 806a94c0, ra == 806a9490
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.377449] Oops[#1]:
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.379721] CPU: 2 PID: 1120 Comm: napi/phy0-7 Tainted: G           O       6.6.52 #0
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.387523] $ 0   : 00000000 00000001 84b4c074 00000074
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.392757] $ 4   : 86cf3cb0 00000000 001d4881 000000a2
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.397981] $ 8   : 84b4c700 ffffffde 060d300f 01062b09
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.403202] $12   : fefefeff 000000ee 8096fe18 00000001
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.408427] $16   : 00000000 0000004e 86cf3c00 84b4c000
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.413649] $20   : 86cf3c00 00000074 00000000 02020100
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.418874] $24   : 00000000 85e73000
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.424095] $28   : 84db6000 84db7978 00000011 806a9490
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.429320] Hi    : 00000003
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.432185] Lo    : a81de6e0
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.435051] epc   : 806a94c0 __udp_gso_segment+0x454/0x64c
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.440553] ra    : 806a9490 __udp_gso_segment+0x424/0x64c
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.446021] Status: 1100fc03      KERNEL EXL IE
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.450204] Cause : 40800008 (ExcCode 02)
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.454193] BadVA : 000000a4
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.457059] PrId  : 0001992f (MIPS 1004Kc)
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.461134] Modules linked in: pppoe ppp_async wireguard pppox ppp_generic nft_redir nft_nat nft_masq nft_flow_offload nft_fib_inet nft_ct nft_chain_nat nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table_inet nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_sane nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_netbios_ns nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge nf_conntrack_amanda nf_conntrack mt7615e(O) mt7615_common(O) mt76_connac_lib(O) mt76(O) mac80211(O) libchacha20poly1305 ipt_REJECT cfg80211(O) cdc_ether xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm tcp_bbr slhc sch_cake poly1305_mips nft_tproxy nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_quota
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.461589]  nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_dup_netdev nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcurve25519_generic iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat(O) chacha_mips asn1_decoder fuse sch_teql sch_sfq sch_multiq sch_gred sch_fq sch_codel em_text em_nbyte em_meta em_cmp act_skbmod act_simple act_pedit act_csum libcrc32c sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set x_tables ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6_gre ip_gre gre ifb ip6_udp_tunnel
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.551688]  udp_tunnel sit ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ipip ip6_tunnel tunnel6 tunnel4 ip_tunnel veth tun xfrm_user xfrm_ipcomp af_key xfrm_algo nls_utf8 nls_iso8859_1 nls_cp950 nls_cp936 nls_cp437 sha512_generic sha256_generic libsha256 sha1_generic seqiv sha3_generic jitterentropy_rng drbg md5 hmac echainiv geniv rng des_generic libdes cmac cbc authencesn authenc mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug(O) vfat fat f2fs ext4 mbcache jbd2 exfat usbcore nls_base usb_common mii crc32c_generic crc32_generic
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.701136] Process napi/phy0-7 (pid: 1120, threadinfo=71e5e2bc, task=ac38474a, tls=00000000)
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.709649] Stack : 84db7a98 84db79ec 84db7b28 84a90000 0000000e 84f69944 84db7b38 00000000
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.718009]         86cf3c00 001d4893 02020100 001d4893 00000052 00000000 00000000 02020100
[... cut ...]
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.892942] Code: 8e900000  8e9300a4  96950096 <8e0200a4> 96120096  0275a821  00529021  96850098  96a40002
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.902700]
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.904248] ---[ end trace 0000000000000000 ]---
Thu Sep 26 11:15:20 2024 kern.emerg kernel: [  581.908877] Kernel panic - not syncing: Fatal exception in interrupt
Thu Sep 26 11:15:20 2024 kern.warn kernel: [  581.915269] ------------[ cut here ]------------
```

nbd168, 2024-09-26:
Fixed in 774badd8a840be9888583266e12ba9d267201ee9

closed 2024-09-26 by nbd168

easyteacher, 2024-09-29:
No crash seen with the latest snapshot. Thanks!