openwrt/openwrt #15461: Nanopi R4S, LAN port dead after switching to Kernel 6.6

teqlee, 2024-05-11:
### Describe the bug

After switching to Kernel 6.6, the Nanopi R4S Lan led not light up, can't get ip from lan port. Tested with latest snapshot build.

### OpenWrt version

snapshot

### OpenWrt release

snapshot

### OpenWrt target/subtarget

Rockchip

### Device

Nanopi R4S

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

teqlee, 2024-05-11:
@1715173329 please notice.

1715173329, 2024-05-11:
"If there are no logs, no relevant screenshots that cannot be used, it is suggested to go find a fortune teller on the street."

teqlee, 2024-05-13:
It seems the problem is overclocking, in version 6.1 the cpu was overclocked to 2.0ghz (from 1.8ghz) without any problem, but version 6.6 seems to not like overclocking, 4/5 starts LAN port is not working. When not overclocked everything works normally.

closed 2024-05-13 by teqlee

smallprogram, 2024-11-21:
@teqlee Hi, how did you solve this problem? I have the same problem.