openwrt/openwrt #15462: Successfully installed the luci ipk file manually, but the service does not show up on the web page. luci-passwall. luci-openclass.

frozenfish0x, 2024-05-11:
### Describe the bug

Successfully installed the luci ipk file manually, but the service does not show up on the web page. luci-passwall. luci-openclass.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (i440FX + PIIX, 1996)

### Image kind

Official downloaded image

### Steps to reproduce

I was installing ipk installers such as luci-app-passwall,luci-app-openclash etc. on a newly installed device through a newly installed device and it installed successfully and still didn't show up after reboot (I know that at one time it wouldn't show up without rebooting for these newly installed services), and then later on I tested the services in the official sources and they would show up after refreshing the browser.

No matter how much I debugged I couldn't get these my ipk packages to display successfully, so I changed the theme, downloaded luci-theme-material and a few other official few themes and replaced him, and found that the following error would be reported:

Configuring luci-app-passwall. lua: /usr/lib/lua/luci/model/cbi/passwall/api/api.lua:6: module 'luci.cbi.datatypes' not found: no field package.preload['luci.cbi.datatypes'] no file '. /luci/cbi/datatypes.lua' no file '/usr/share/lua/luci/cbi/datatypes.lua' no file '/usr/share/lua/luci/cbi/datatypes/init.lua' no file '/usr /lib/lua/luci/cbi/datatypes.lua' no file '/usr/lib/lua/luci/cbi/datatypes/init.lua' no file '. /luci/cbi/datatypes.so' no file '/usr/lib/lua/luci/cbi/datatypes.so' no file '/usr/lib/lua/loadall.so' no file '. /luci.so' no file '/usr/lib/lua/luci.so' no file '/usr/lib/lua/loadall.so' stack traceback: [C]: in function 'require' /usr/lib/lua/luci/model/ cbi/passwall/api/api.lua:6: in main chunk [C]: in function 'require' .... .r/lib/lua/luci/model/cbi/passwall/server/api/app.lua:4: in main chunk [C]: ? uci: Entry not found

Then I install

opkg install luci-compat
and the previously installed ipk packages show up just fine!

### Actual behaviour

Web admin page does not show services installed by luci ipk

### Expected behaviour

I'm hoping this is how the problem arose and fixes the inability to directly display the installed ipk service

### Additional info

Original openwrt, in pve.
I'll expand the root partition after a successful install /
Here's what to do
Partition fdisk and format it as ext4 filesystem
Then mount it with block-mount to make it the root partition

mkdir -p /tmp/introot
mkdir -p /tmp/extroot
mount --bind / /tmp/introot
mount /dev/sda3 /tmp/extroot
tar -C /tmp/introot -cvf - . | tar -C /tmp/extroot -xf - .
umount /tmp/introot
umount /tmp/extroot

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

frozenfish0x, 2024-05-11:
This is the error after changing the theme

Runtime error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/passwall/api.lua:8: module 'luci.cbi.datatypes' not found:
no field package.preload['luci.cbi.datatypes']
no file './luci/cbi/datatypes.lua'
no file '/usr/share/lua/luci/cbi/datatypes.lua'
no file '/usr/share/lua/luci/cbi/datatypes/init.lua'
no file '/usr/lib/lua/luci/cbi/datatypes.lua'
no file '/usr/lib/lua/luci/cbi/datatypes/init.lua'
no file './luci/cbi/datatypes.so'
no file '/usr/lib/lua/luci/cbi/datatypes.so'
no file '/usr/lib/lua/loadall.so'
no file './luci.so'
no file '/usr/lib/lua/luci.so'
no file '/usr/lib/lua/loadall.so'

In error(), file [C]
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function ((tail call))
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 148, byte 45:
  called from function build_pagetree (/usr/share/ucode/luci/dispatcher.uc:382:73)
  called from function menu_json (/usr/share/ucode/luci/dispatcher.uc:439:26)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:898:24)
  called from anonymous function (/www/cgi-bin/luci:39:13)

 `        return lcall.call(modname, method, ...args);`
  Near here ----------------------------------------^

jow-, 2024-05-13:
You need to install luci-compat. Apart from that, neither package is maintained by OpeneWrt.

closed 2024-05-13 by jow-

Maxx69r, 2025-02-01:
tnx!  this is really useful