openwrt/openwrt #15466: [mvebu] wrt3200acm fails to compile with 6.6 kernel

johnjacob10, 2024-05-12:
### Describe the bug

I build my own images for wrt3200acm on the snapshot branch. Building with the 6.6 kernel version fails, and since 6.6 has been made the default kernel version, I can no longer build at all. The error is:
```
  gcc -L/var/local/openwrt/staging_dir/host/lib -o scripts/kconfig/conf scripts/kconfig/conf.o scripts/kconfig/confdata.o scripts/kconfig/expr.o scripts/kconfig/lexer.lex.o scripts/kconfig/menu.o scripts/kconfig/parser.tab.o scripts/kconfig/preprocess.o scripts/kconfig/symbol.o scripts/kconfig/util.o
scripts/kconfig/conf  --syncconfig Kconfig
*
* Restart config...
*
*
* Performance monitor support
*
ARM CCI PMU driver (ARM_CCI_PMU) [N/m/y/?] n
ARM CCN driver support (ARM_CCN) [N/m/y/?] n
ARM PMU framework (ARM_PMU) [Y/n/?] y
ARM PMUv3 support (ARM_PMUV3) [N/y/?] (NEW) make[8]: *** [scripts/kconfig/Makefile:77: syncconfig] Error 1
make[7]: *** [Makefile:697: syncconfig] Error 2
make[6]: *** [/var/local/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-6.6.30/Makefile:798: include/config/auto.conf.cmd] Error 2
make[5]: *** [Makefile:234: __sub-make] Error 2
make[5]: Leaving directory '/var/local/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-6.6.30'
make[4]: *** [Makefile:19: /var/local/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-6.6.30/.modules] Error 2
make[4]: Leaving directory '/var/local/openwrt/target/linux/mvebu'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/var/local/openwrt/target/linux'
time: target/linux/compile#28.21#13.48#46.64
```

I have attached the full log and the config used to build.

### OpenWrt version

latest snapshot

### OpenWrt release

latest snapshot

### OpenWrt target/subtarget

mvebu/wrt3200acm

### Device

wrt3200acm

### Image kind

Self-built image

### Steps to reproduce

Try to build openwrt from snapshot using attached config.
[mvebu_config.txt](https://github.com/openwrt/openwrt/files/15286017/mvebu_config.txt)

### Actual behaviour

Build fails with error as seen in attached log.
[compile.txt](https://github.com/openwrt/openwrt/files/15286019/compile.txt)

### Expected behaviour

Build should succeed.

### Diffconfig

```text
CONFIG_TARGET_mvebu=y
CONFIG_TARGET_mvebu_cortexa9=y
CONFIG_TARGET_mvebu_cortexa9_DEVICE_linksys_wrt3200acm=y
CONFIG_DEVEL=y
[... 247 lines cut ...]
# CONFIG_TARGET_ROOTFS_TARGZ is not set
CONFIG_VERSIONOPT=y
CONFIG_VERSION_BUG_URL=""
CONFIG_VERSION_CODE=""
CONFIG_VERSION_CODE_FILENAMES=y
CONFIG_VERSION_DIST="OpenWrt"
CONFIG_VERSION_FILENAMES=y
CONFIG_VERSION_HOME_URL=""
CONFIG_VERSION_HWREV=""
CONFIG_VERSION_MANUFACTURER="Linksys"
CONFIG_VERSION_MANUFACTURER_URL="https://www.linksys.com/us/c/wireless-routers/"
CONFIG_VERSION_NUMBER=""
CONFIG_VERSION_PRODUCT=""
CONFIG_VERSION_REPO=""
CONFIG_VERSION_SUPPORT_URL=""
CONFIG_ZLIB_OPTIMIZE_SPEED=y
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-05-12:
Add:
# CONFIG_ARM_PMUV3 is not set

robimarko, 2024-05-12:
@johnjacob10 Try picking this:
https://github.com/openwrt/openwrt/pull/15469

httpstorm, 2024-05-14:
I compiled and installed [main](https://github.com/openwrt/openwrt/commit/4f87a4d84f3d6d1625962413ea56cca1ce83db7c) today: `linux-6.6.30`. No issues. I have a few commits on top of main that fix the 5GHz radio, though they should not fix any build errors. https://github.com/httpstorm/openwrt/commits/2024-05-13/

robimarko, 2024-05-14:
Yeah, you need to set `CONFIG_KERNEL_PERF_EVENTS`

httpstorm, 2024-05-14:
@robimarko 
Yes, I reproduced it on macOS 14.4.1. I just finished building from an empty directory. Here are the logs before and after your [patch](https://github.com/openwrt/openwrt/pull/15469). Your patch fixes the build error in `make target/linux/compile`. Alas the third log shows that `make target/linux/install` fails. This is with @johnjacob10 's configuration file on the same branch which I compiled and installed on my router yesterday https://github.com/httpstorm/openwrt/commits/2024-05-13/
- [ ] [wrt3200acm.2024-05-14.01.arm-pmuv3.txt](https://httpstorm.com/share/.openwrt/test/2024-03-12_xcode-15.3/wrt3200acm.2024-05-14.01.arm-pmuv3.txt)
- [x] [wrt3200acm.2024-05-14.02.arm-pmuv3.txt](https://httpstorm.com/share/.openwrt/test/2024-03-12_xcode-15.3/wrt3200acm.2024-05-14.02.arm-pmuv3.txt)
- [ ] [wrt3200acm.2024-05-14.03.linux-install.txt](https://httpstorm.com/share/.openwrt/test/2024-03-12_xcode-15.3/wrt3200acm.2024-05-14.03.linux-install.txt)

```
find /Volumes/wrt3200/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-6.6.30 /Volumes/wrt3200/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/lib/modules -name \*.ko | xargs arm-openwrt-linux-muslgnueabi-nm | awk '$1 == "U" { print $2 } ' | sort -u > /Volumes/wrt3200/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mod_symtab.txt
arm-openwrt-linux-muslgnueabi-nm -n /Volumes/wrt3200/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-6.6.30/vmlinux.o | awk '/^[0-9a-f]+ [rR] __ksymtab_/ {print substr($3,11)}' > /Volumes/wrt3200/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/kernel_symtab.txt
grep -Ff /Volumes/wrt3200/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/mod_symtab.txt /Volumes/wrt3200/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/kernel_symtab.txt > /Volumes/wrt3200/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/sym_include.txt
make[3]: *** [Makefile:23: /Volumes/wrt3200/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/symtab.h] Error 1
```

I'll give it a try with my configuration, and `CONFIG_KERNEL_PERF_EVENTS` enabled.
Robi, can you please review the `frr` PR we were working on last week? [frr: merge frr-libfrr and frr-vtysh to frr #24083](https://github.com/openwrt/packages/pull/24083)

robimarko, 2024-05-14:
`CONFIG_STRIP_KERNEL_EXPORTS` is broken on 6.6 so that is why install fails

johnjacob10, 2024-05-14:
I tried @robimarko 's patch and I hit a subsequent error in the install. I'll wait for ```CONFIG_STRIP_KERNEL_EXPORTS``` to be fixed. Is there an issue raised for that?

robimarko, 2024-05-14:
Can you just disable it for now and test?

I dont think there is an issue for `CONFIG_STRIP_KERNEL_EXPORTS`

closed 2024-05-14 by openwrt-bot

httpstorm, 2024-05-14:
@robimarko @johnjacob10 
Confirmed: `target/linux/install` files only when `CONFIG_STRIP_KERNEL_EXPORTS` is enabled.

PR #15469 (2024-05-14): config: add ARM PMUv3 for kernel 6.6
Kernel 6.6 has moved the ARM PMUv3 driver to drivers/perf and now once KERNEL_ARM_PMU is selected trying to build the kernel will stop with: ARM PMUv3 support (ARM_PMUV3) [N/y/?] (NEW)

So, lets enable ARM_PMUV3 for ARMv7 and ARMv8 architectures if KERNEL_PERF_EVENTS is selected.

Fixes: #15466
commit 4341901f (2024-05-14): config: add ARM PMUv3 for kernel 6.6

Kernel 6.6 has moved the ARM PMUv3 driver to drivers/perf and now once
KERNEL_ARM_PMU is selected trying to build the kernel will stop with:
ARM PMUv3 support (ARM_PMUV3) [N/y/?] (NEW)

So, lets enable ARM_PMUV3 for ARMv7 and ARMv8 architectures if
KERNEL_PERF_EVENTS is selected.

Fixes: #15466
Link: https://github.com/openwrt/openwrt/pull/15469