openwrt/openwrt #15467: ERROR:   BL2: Failed to load image id 3 (-2)

QooGeek, 2024-05-12:
### Describe the bug

https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=mediatek%2Ffilogic&id=bananapi_bpi-r3-mini

openwrt-mediatek-filogic-bananapi_bpi-r3-mini-emmc-bl31-uboot.fip
openwrt-mediatek-filogic-bananapi_bpi-r3-mini-emmc-gpt.bin
openwrt-mediatek-filogic-bananapi_bpi-r3-mini-emmc-preloader.bin
openwrt-mediatek-filogic-bananapi_bpi-r3-mini-initramfs-recovery.itb
openwrt-mediatek-filogic-bananapi_bpi-r3-mini-squashfs-sysupgrade.itb

root@OpenWrt:/tmp# dd if=openwrt-mediatek-filogic-bananapi_bpi-r3-mini-emmc-gpt.
bin of=/dev/mmcblk0
34+0 records in
34+0 records out

root@OpenWrt:/tmp# dd if=/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-mini-emmc
-preloader.bin of=/dev/mmcblk0boot0
392+1 records in
392+1 records out

root@OpenWrt:/tmp# dd if=/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-mini-emmc
-bl31-uboot.fip of=/dev/mmcblk0p3
2240+1 records in
2240+1 records out

root@OpenWrt:/tmp# dd if=/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-mini-init
ramfs-recovery.itb of=/dev/mmcblk0p4
dd: error writing '/dev/mmcblk0p4': No space left on device
4097+0 records in
4096+0 records out

root@OpenWrt:/tmp# dd if=/tmp/openwrt-mediatek-filogic-bananapi_bpi-r3-mini-squa
shfs-sysupgrade.itb of=/dev/mmcblk0p5
22016+1 records in
22016+1 records out

root@OpenWrt:/tmp# sync

root@OpenWrt:/tmp# cu: Got hangup signal
Disconnected.

➜  ~ sudo cu -s 115200 -l /dev/tty.usbserial-110
Password:
Connected.
ERROR:   BL2: Failed to load image id 3 (-2)

### OpenWrt version

SNAPSHOT

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

SNAPSHOT

### Device

bananapi r3 mini

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-05-12:
Wrong partition layout, /dev/mmcblk0p4 is too small.

To start using the R3 download [openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz 270](https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz) and write the decompressed content of this file to a microSD card, e.g.
```
gzip -cd openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz | dd of=/dev/mmcblk0
```

QooGeek, 2024-05-13:
@namiltd  Thank you very much, I've got it done. In addition, this does not seem to support 5g modules. What do I need to do to support 5g modules?

```
root@OpenWrt:~# echo 443 > /sys/class/gpio/export
ash: write error: Invalid argument
root@OpenWrt:~#
root@OpenWrt:~# echo 431 > /sys/class/gpio/export
ash: write error: Invalid argument
root@OpenWrt:~#
root@OpenWrt:~# echo out > /sys/class/gpio/gpio431/direction
-ash: can't create /sys/class/gpio/gpio431/direction: nonexistent directory
root@OpenWrt:~#
root@OpenWrt:~# echo 0 > /sys/class/gpio/gpio431/value
-ash: can't create /sys/class/gpio/gpio431/value: nonexistent directory

```

namiltd, 2024-05-13:
I have no experience with this.

closed 2024-05-14 by QooGeek

Speechless22, 2024-08-11:
> @namiltd Thank you very much, I've got it done. In addition, this does not seem to support 5g modules. What do I need to do to support 5g modules?
>
> ```
> root@OpenWrt:~# echo 443 > [...]

```
dd: error writing '/dev/mmcblk0p4': No space left on device
```
Please tell me how you solved this problem?

starsuni, 2025-01-03:
> @namiltd Thank you very much, I've got it done. In addition, this does not seem to support 5g modules. What do I need to do to support 5g modules?
>
> ```
> root@OpenWrt:~# echo 443 > [...]

> @namiltd Thank you very much, I've got it done. In addition, this does not seem to support 5g modules. What do I need to do to support 5g modules?
>
> ```
> root@OpenWrt:~# echo 443 > [...]

May I know how you fix this? I have same issue, thanks